turtlebot/turtlebot4 #359: Turtlebot4 Does Not Move When Given /cmd_vel

kfu02, 2024-02-02:
### Robot Model

Turtlebot4 Standard

### ROS distro

Galactic

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Not sure, whatever was preinstalled on turtlebot when it arrived.

### Type of issue

Other

### Expected behaviour

I have followed the Basic Setup guide and successfully SSH'd into my turtlebot 4. When I try to move the turtlebot with ROS commands e.g. 
```
ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.1"
```
the turtlebot should spin in place.

### Actual behaviour

It does not move at all. Echoing the /cmd_vel shows:
```
---
linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.1
---
```
but the turtlebot remains idle.

### To Reproduce

1. Open a brand new Turtlebot
2. Follow Basic Setup to SSH into it
3. Publish a velocity to cmd_vel/
4. Echo cmd_vel/ to confirm it is receiving published messages

kfu02, 2024-02-02:
Might be related to #354 ?

xiaoshan-lin, 2024-02-13:
SSH into turtlebot and echo the /cmd_vel only verifies that raspberry pi is receiving the messages, but not the create3.

First of all, make sure the comm and battery LEDs are on. 

On your PC, open a terminal and run ros2 topic info /cmd_vel --verbose. There should be one node called motion_control that subscribes to the /cmd_vel topic. 
If subscription count is 0, that means the create3 is not listening to the topic. In that case, try running ros2 daemon stop; ros2 daemon start, and wait for some time (It usually takes me 30 seconds to see that /motion_control node is listening to /cmd_vel)

Once the motion_control node is listening to /cmd_vel, try publishing commands to /cmd_vel. For our turtlebots, it takes 20~40 seconds to eventually respond to velocity commands published on /cmd_vel. See https://github.com/turtlebot/turtlebot4/issues/358

If you have one turtlebot, discovery server might be a better option. We tried discovery server at some point and never had such issues.

kfu02, 2024-02-14:
Hi, I was actually able to resolve this issue, for some reason there is just a large delay (~30 s) before the turtlebot actually starts responding to cmd_vel/.

closed 2024-02-14 by kfu02

Esme1352, 2024-06-19:
What should I do if the communication and battery LEDs are not on?

I have a discovery server configuration and the topics appear on both the raspberry and the PC.

I have launched the topic /cmd_vel and I have waited more than 1 min and it does not move.