turtlebot/turtlebot4 #360: Issue with Raspberry Pi Wifi Basic Setup - Unable to Obtain IP Address

Alaeddine-Nasri, 2024-02-07:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

I think its not needed in my issue 

### Type of issue

Networking

### Expected behaviour

Show the IP address on the robot screen 

### Actual behaviour

Its shows Unknown
WiFi LED is off.

### To Reproduce

Hello everyone,

I hope this message finds you well. I'm encountering an issue during the Basic Setup of my Raspberry Pi as per the manual. Specifically, in the "**Connect the Raspberry Pi to your network**" section, I set up the Raspberry Pi in "client" mode, configured the network to "iPhone," and entered the correct password for my phone's network.

The problem arises after applying these changes and after the auto restart of the robot. Instead of displaying the expected IP address on the robot's screen, it shows "Unknown." This issue has occurred consistently with two standard robots, and despite following the manual, I am unable to access them.

I'm utilizing **VirtualBox** with Ubuntu 22.04 as the host system. Additionally, the WiFi LED after the restard went off.

As this is my first time working with Raspberry Pi, any assistance in identifying the issue and providing guidance on how to resolve it would be greatly appreciated.

Steps Taken:
Set up Raspberry Pi in "client" mode.
Configured the network to "iPhone" and entered the correct password.
Applied changes, and the robot restarted as expected.
The robot's screen displays "Unknown" instead of the IP address.
Encountered the same issue with two different standard robots.

**Environment:**
VirtualBox: 22.04 Linux
Standard robots

**Additional Information:**
WiFi LED is off now.

Thank you for your time and assistance.

Best regards,

jordanlv, 2024-02-12:
Hello,
Could you clarify what you mean by "Configured the network to 'iPhone'"? Are you setting up your iPhone as an access point ? 
If so, does your iPhone assign IP addresses ?

Alaeddine-Nasri, 2024-02-15:
Hello,

Thanks for your help! We fixed the initial issue by sharing the hotspot from a PC instead of an iPhone. Now, we have a new problem. We tried using the "turtlebot-ntpd-sync" command to sync "/scan" and "/odom" topics, but it didn't work.

Attached is a photo showing the problem:  
[image]

> When we try to execute "slam," it stops at a certain point, and the values aren't displayed as expected.

[image]

Any guidance on fixing this would be greatly appreciated!

Thanks again,

jordanlv, 2024-02-15:
I'm not sure I understand your problem but you may want to look at https://github.com/turtlebot/turtlebot4/issues/312#issuecomment-1829395307.

hilary-luo, 2024-02-15:
@Alaeddine-Nasri Can you please explain why you are using the "turtlebot-ntpd-sync" command and what are you trying to do to those particular two topics?

Also, can you expand on what you mean by the slam "stops at a certain point, and the values aren't displayed as expected"? The output you have shared has no indications that anything is wrong. Can you add photos or additional logs etc to explain what your problem is?

RustyCPR, 2024-03-04:
Hi @Alaeddine-Nasri  any updates on this? Did you resolve the issue? 
Thanks

RustyCPR, 2024-03-18:
Hello,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Rusty

closed 2024-03-18 by RustyCPR