turtlebot/turtlebot4 #364: all 4 led down, no SSH ethernet, no display

anantshukla8, 2024-02-14:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

desktop

### Type of issue

Other

### Expected behaviour

nothing, I wan doing initial setup for discovery server

### Actual behaviour

After I applied settings in Ros setup, the robot restarted with all 4 LEDs down, nothing on display, RPi is on, but cannot connect over SSH ethernet and wifi

Cannot connect to Ethernet it says: ssh: connect to host 192.168.185.3 port 22: No route to host

When I power on, 1st led like :img1 (attachment)
[image]

then led turn red: img2 (attachment)

[image]

then led turns green with chimes: image 3 (attachment)
[image]

### Error messages

```bash
all 4 leds off/down, nothing on display
```

### To Reproduce

was doing initial setup

smatarCPR, 2024-03-04:
Hello @anantshukla8 ,
Thank you for reaching on the official Turtlebot4 Github page. Apologies for the late reply on this. After powering on the Turtlebot using the dock, does the LCD screen change display's from the static like image to a text display? Does the IP address show as unknown if it reaches this state?

The Turtlebot 4 is known to take some time to power on.

The first item to check would be if the Raspberry Pi is turning on. At the rear of the Turtlebot you should be able to find the Raspberry Pi. Ensure you see a green LED light coming from the Pi indicating it is powered on.

If the Pi is not powered on - check the Create3's power adapter board. It can be found in the Create3's cargo bay (https://iroboteducation.github.io/create3_docs/hw/mechanical/) Ensure that the USB cable is connected securely to the power adapter board.

If the Pi is powered on, but the Turtlebot never exits this state, try reflashing the Raspberry Pi using the following method:

1. Find the latest TurtleBot 4 Raspberry Pi images at http://download.ros.org/downloads/turtlebot4/
2. Download the latest image and extract it.
3. Power off your robot and then remove the microSD card from the Raspberry Pi.
4. Insert the microSD card into your PC. You may need an adapter.
5. Install the imaging tool dcfldd: sudo apt install dcfldd
6. Identify the SD card using sudo fdisk -l
7. If you wish to back up the contents on the SD card: sudo dd if=/dev/<SD_NAME> of=<IMAGE_PATH> bs=1M
8. Get the SD flash script from turtlebot4_setup: wget https://raw.githubusercontent.com/turtlebot/turtlebot4_setup/humble/scripts/sd_flash.sh
9. Flash the SD Card: bash sd_flash.sh /path/to/downloaded/image
10. Be sure to flash the corresponding firmware to the Create3 Unit.

Let me know how these recommendations work for you. Looking forward to hearing from you soon.
Best Regards,
Saif

albertuswh, 2024-03-07:
Hi @smatarCPR  when i flash the sd card, it asked for the sd card name, so which one is the sd card name?

Best regards,
Albert

smatarCPR, 2024-03-25:
Hello @albertuswh ,
Any updates on this? Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2024-04-02 by smatarCPR