turtlebot/turtlebot4 #368: changing GUID of discovery server

mdxtinkernick, 2024-02-27:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

turtlebot-base is 1.02-1jammy.20240219.090534

### Type of issue

Networking

### Expected behaviour

I am trying to set up multiple t4s on the same isolated lan (no internet). I have got a system working for syncing time, and using ROS_DOMAIN_ID to isolate ros, but if I am using multiple robots, each with a discovery server on do the discovery servers need different GUID values to work - I have tried changing the 3rd value in the prefix for the discovery server in the config file to 01 on the pi and desktop, and restarting both machine to ensure everything is using that value - would hope that ros messages still visible on both machines

### Actual behaviour

both machines only show 

/parameter_events
/rosout

when running ros2 topic list, a publisher will run, but topic does not show up on either machine.

rosnode list is empty for both machines, but top on the robot shows all the expected processes are running for the nodes that have been automatically started at boot.

Is it necessary to try and do this for each robot to run independently, and if so what am I missing in the config process. I am aware that I will need to change config on create3 as well if it works, but for now without the create3 connected it is not working, so am not looking at that yet

### Error messages

```bash
no error messages displayed
```

### To Reproduce

change 3rd value to 01 in prefix in /etc/turtlebot4/fastdds_discovery_superclient.xml on pi and desktop

reboot both, reconnect on network.

changing it back to 00 and rebooting and topics show up again on both

mdxtinkernick, 2024-02-27:
Just to clarify - I am not trying to get communication between robots to work, I am trying to isolate each robot, with each one connecting to its own desktop device on the isolated lan

smatarCPR, 2024-03-11:
Hello @mdxtinkernick ,
Thank you for reaching out on the official Turtlebot4 Github page. Unfortunately at this time having multiple robots running a Discovery server on the same network is currently not supported. If you are working with multiple robots, it is required that you setup a Simple Discovery networking setup per: https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html

Let me know if I can help with anything else.
Best Regards,
Saif

hilary-luo, 2024-03-11:
@mdxtinkernick discovery servers need separate IDs only in order to operate together without causing conflicts. If you do not have any client in common for any of your various discovery servers then you should be able to run them all with the default ID. However, if you wanted to have one desktop computer communicate with more than one robot then this would not be sufficient.

In regards to why it didn't work, I will share my two cents but I am by no means an expert in Fast DDS xml profiles. What you changed was the client side profile only, not the server side. Since the client and server no longer match, the clients can no longer see the topics. I would not suggest modifying the xml profile given the complexity.

If you need to operate with multiple robots connected to a single computer then as Saif mentioned, please await the multi-robot support. You can follow along here for the latest updates: https://github.com/turtlebot/turtlebot4/issues/145

smatarCPR, 2024-04-02:
Hello @mdxtinkernick ,
Any updates on this matter? Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2024-04-09 by smatarCPR