turtlebot/turtlebot4 #369: Raspberry Pi IP: "Unknown"

Imani-Saba, 2024-02-29:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

sudo apt install ros-humble-desktop

### Type of issue

Networking

### Expected behaviour

Turtlebot4 connects to PC

### Actual behaviour

we encountered an issue in the beginning with connecting Ubuntu to the Raspberry Pi.

### Error messages

```bash
OS: Ubuntu 22.04
ROS Distro: Humble
Built from source or installed: Installed

I am working on the IEQ project at USC. We recently bought the Turtlebot4 to collect environmental data. However, we encountered an issue in the beginning with connecting Ubuntu to the Raspberry Pi. On the screen, the Raspberry Pi IP showed "Unknown," and I could not connect to the Turtlebot4 via WIFI.

I attempted to directly connect to the Raspberry Pi using an Ethernet cable as mentioned in the manual. After setting up IPv4 in wired and inputting IP: 192.168.185.3, it did not accept any password such as "turtlebot4" or others (default RPi password).

Someone mentioned on GitHub that after checking all avenues, we should flash and update the SD memory card of the Raspberry Pi and reimage it. However, access to the SD card requires disassembling the entire Turtlebot4, which is very challenging.

Is there any method to solve this problem?

I have called the company several times and sent emails, but nobody has been able to solve my problem. They only emailed me the following link on GitHub. I am truly shocked by this lack of support!
```

### To Reproduce

else

RustyCPR, 2024-03-01:
Hi there, thanks for replying here  - I replied to you in the other email. Regarding RPi displays "unknown" instead of an IP address.
Its best to reflash the RPi and I replied with step-by-step instructions and images on how to do so.
1. You need to remove the [top integration plate](https://turtlebot.github.io/turtlebot4-user-manual/mechanical/turtlebot4.html#removing-the-top-integration-plate)
2. Unscrew the 4 standoff posts.
3. The PCBA with camera and lidar can be tilted back to access the RPi and the micro SD card. 
This process can take under 5 minutes. 
[image]

We have some other troubleshooting steps, but all require you to access the RPi: https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#faq

After that please follow the [User Manual | Basic Setup](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html) or the [TurtleBot 4 | Unboxing & Getting Started](https://youtu.be/QN01AXjoLdQ?si=60ZI0okwnx6JrjnI)

Best regards,
Rusty

Imani-Saba, 2024-03-02:
Hi,

I Downloaded turtlebot4_standard_humble_1.0.0.zip then unzip it. I reimage SD card based on:
https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image

Now, IP shows up in Turtlebot4, but when put ssh ubuntu@10.42.0.1, could not find it.
I tried to use the static IP address for the ethernet: ssh ubuntu@192.168.185.3, but did not accept the password (turtlebot4).

Imani-Saba, 2024-03-03:
Hi,

I Downloaded turtlebot4_standard_humble_1.0.0.zip then unzip it. I reimage SD card based on:
https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image

Now, the IP shows up (10.42.0.1) on Turtlebot4, then connect to Raspberry Pi by ssh ubuntu@10.42.0.1, then try to Connect the Raspberry Pi to my network. after that using Humble instruction, in my SSH session, run the TurtleBot 4 setup tool:
-Apply setting, select wifi mode (client), put my home wifi and password in SSID and Password, and then save and exit.

In the manual says "Once the Wi-Fi settings are applied, the Raspberry Pi will reboot and connect to your network. DHCP will assign it a new IP address. On the TurtleBot 4, this IP address will be shown at the top of the display.", but I have not received any new IP and the IP address does not change on the TurtolBot display and it still shows the IP address 10.42.0.1

The controller also does not connect to the TurtleBot. I used the manual, after turning on the controller with the home button the light starts blinking, and then becomes blue, but after a few seconds it goes off and no other buttons work to move the Turtlebot.

RustyCPR, 2024-03-04:
Hi @Imani-Saba 
Can you please run through the setup processes again using the [Setup Tool](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#connect-the-raspberry-pi-to-your-network)
Are you able to confirm if you can see the RPi connected to the network? 

We do get support requests from university users who have issues with WiFi network connectivity issues with IT restrictions, user names, and additional enterprise/ certificate login requirements beyond the basic network requirements:  1. SSID, 2. network password requirement. 
Double check with your IT team. 

Regarding the Bluetooth issue: 
1. Run `sudo bluetoothctl `when the Bluetooth Control comes up, it will list all devices that have been connected.
Run: `bluetoothd`
2. Find all the devices named "`Wireless Controller`". 
Run the commands below with the controller's address for each controller in the list

`untrust XX:XX:XX:XX:XX:XX`
`remove XX:XX:XX:XX:XX:XX`

3. Each of these will confirm that the device has been untrusted or removed
4. With the Controller in pairing mode
5. Type `scan on` and press enter.
5. Find the Bluetooth device listed as Controller or Wireless Controller and Controller MAC address.
Then run the below commands with the listed controller address:
`trust XX:XX:XX:XX:XX:XX  `
`pair XX:XX:XX:XX:XX:XX  `     
`connect XX:XX:XX:XX:XX:XX  `    (_may need to enter this command two or three times for it to succeed, you will see a message controller has been connected and the light on the controller will turn blue)_

If the P4 controller is still not connecting test it can pair by pairing to your mobile phone? 
If not you should charge it using a phone USB charger.

albertuswh, 2024-03-07:
Hi @Imani-Saba, I also wanted to reimage the sd card. But I got confused when i did this command

    Get the SD flash script from turtlebot4_setup and flash the SD card:

wget https://raw.githubusercontent.com/turtlebot/turtlebot4_setup/humble/scripts/sd_flash.sh
bash sd_flash.sh /path/to/downloaded/image

it asked for sd card device name. So where do I get the SD card name? and if using the "sudo fdisk -l" are the names /dev/mmcblk0p1 and /dev/mmcblk0p2? 
Thank you,

Imani-Saba, 2024-03-10:
Hello
[image]
,

I followed all of your instructions to connect the controller, but I encountered this error:

"Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[ CHG ] Device A0:5A:5D:0C:05:FF Connected: no"

The light continues blinking and doesn't turn blue.

Following your suggestion, I attempted to pair it with my mobile phone, and it connected successfully.

However, despite several further attempts, I couldn't establish a connection between the controller and the Turtlebot.

Could you please help me with this issue?

Imani-Saba, 2024-03-10:
Hello,

I followed all of your instructions to connect the controller, but I encountered this error:

"Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[ CHG ] Device A0:5A:5D:0C:05:FF Connected: no"

[image]

The light continues blinking and doesn't turn blue.

Following your suggestion, I attempted to pair it with my mobile phone, and it connected successfully.

However, despite several further attempts, I couldn't establish a connection between the controller and the Turtlebot.

Could you please help me with this issue?

Imani-Saba, 2024-03-10:
> Hi @Imani-Saba Can you please run through the setup processes again using the [Setup Tool](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#connect-the-raspberry-pi-to-your-network) Are you able to confirm if you can see the RPi connected to the network?
> 
> We do get support requests from university users who have issues with WiFi network connectivity issues with IT restrictions, user names, and additional enterprise/ certificate login requirements beyond the basic network requirements: 1. SSID, 2. network password requirement. Double check with your IT team.
> 
> Regarding the Bluetooth issue:
> 
> 1. Run `sudo bluetoothctl `when the Bluetooth Control comes up, it will list all devices that have been connected.
>    Run: `bluetoothd`
> 2. Find all the devices named "`Wireless Controller`".
>    Run the commands below with the controller's address for each controller in the list
> 
> `untrust XX:XX:XX:XX:XX:XX` `remove XX:XX:XX:XX:XX:XX`
> 
> 3. Each of these will confirm that the device has been untrusted or removed
> 4. With the Controller in pairing mode
> 5. Type `scan on` and press enter.
> 6. Find the Bluetooth device listed as Controller or Wireless Controller and Controller MAC address.
>    Then run the below commands with the listed controller address:
>    `trust XX:XX:XX:XX:XX:XX  `
>    `pair XX:XX:XX:XX:XX:XX  `
>    `connect XX:XX:XX:XX:XX:XX  `    (_may need to enter this command two or three times for it to succeed, you will see a message controller has been connected and the light on the controller will turn blue)_
> 
> If the P4 controller is still not connecting test it can pair by pairing to your mobile phone? If not you should charge it using a phone USB charger.

Hello,

I followed all of your instructions to connect the controller, but I encountered this error:

"Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[ CHG ] Device A0:5A:5D:0C:05:FF Connected: no"

[image]

The light continues blinking and doesn't turn blue.

Following your suggestion, I attempted to pair it with my mobile phone, and it connected successfully.

However, despite several further attempts, I couldn't establish a connection between the controller and the Turtlebot.

Could you please help me with this issue?

Imani-Saba, 2024-03-10:
> Hi @Imani-Saba Can you please run through the setup processes again using the [Setup [...]

Hello,

I followed all of your instructions to connect the controller, but I encountered this error:

"Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[ CHG ] Device A0:5A:5D:0C:05:FF Connected: no"

The light continues blinking and doesn't turn blue.

Following your suggestion, I attempted to pair it with my mobile phone, and it connected successfully.

However, despite several further attempts, I couldn't establish a connection between the controller and the Turtlebot.

Could you please help me with this issue?

Imani-Saba, 2024-03-10:
> Hi @Imani-Saba Can you please run through the setup processes again using the [Setup [...]

Hello,

I followed all of your instructions to connect the controller, but I encountered this error:

"Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[ CHG ] Device A0:5A:5D:0C:05:FF Connected: no"

[image]

The light continues blinking and doesn't turn blue.

Following your suggestion, I attempted to pair it with my mobile phone, and it connected successfully.

However, despite several further attempts, I couldn't establish a connection between the controller and the Turtlebot.

Could you please help me with this issue?

RustyCPR, 2024-03-11:
Hi @Imani-Saba 
Can you try this: https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#3-no-default-controller-available
If it does not work - please share the full terminal outputs from the untrust / remove / trust / pair and connect commands being run in the terminal.

Imani-Saba, 2024-03-12:
Hello,

I have followed the process for connecting the controller with TurtleBot4 several times, following your instructions, but we are encountering the same issue.

First, I did the following steps:

1. disable hciuart
sudo systemctl disable hciuart
sudo reboot
2. restart hciuart
sudo systemctl restart hciuart
3. connect gamepad
sudo bluetoothctl
connect MAC_ADDRESS

After completing these steps, I proceeded with the following:

1. untrust MAC_ADDRESS
2. remove MAC_ADDRESS
3. scan on
4. trust MAC_ADDRESS  
5. pair MAC_ADDRESS  
6. connect MAC_ADDRESS

However, after proceeding with these commands, the terminal briefly displays "Wireless Controller" (as shown in screenshot 1) for about 5 seconds. Despite this, the light on the controller remains blinking and does not turn blue. After a few seconds, the terminal switches back to "Bluetooth" (as shown in screenshot 2).

[image]

[image]

RustyCPR, 2024-04-09:
Closing due to non-activity also duplicate ticket #371

closed 2024-04-09 by RustyCPR