turtlebot/turtlebot4 #370: ROS2 Comms Issues

CycleMark, 2024-03-02:
### Robot Model

Turtlebot4 Lite

### ROS distro

Galactic

### Networking Configuration

Discovery Server

### OS

Ubuntu 20.04

### Built from source or installed?

Installed

### Package version

galatic

### Type of issue

Networking

### Expected behaviour

Hi,

Having a lot of problems with the reliability of TB4 internal comms. To limits the issues this is all being done on the robot alone (i.e. no PC involved)

I get very intermitted service and/or topic communications. As an example regardless of if I bring ROS2 up as a service or from the command line (ros2 service call /stop_motor std_srvs/srv/Empty)

More often than not ROS2 can't communicate with, in this example, services. As you can see below it can "see" the service. But ask if to interact with it is doesn't know its there - just waits indefinitely for it to come up. Similar cases happen for topics. ROS2 can see them but doesn't always get data from them (ros2 topic echo <topic_name>) for example.

[image]

If this a known problem? Do you have any suggestions on how to improve/fix the situation. Sometimes it works, sometimes not. Power cycling the iCreate doesn't make a difference. I want to have the robot/system come up and just work.

Thanks

Mark

### Actual behaviour

Intermittent/Unreliable comms from services/topics either at command line or application/api

### To Reproduce

See command lines above

CycleMark, 2024-03-06:
Thanks - I'll give this a go and revert

smatarCPR, 2024-03-25:
Hello @CycleMark ,
Any update on this? Looking forward to hearing from you soon.
Best Regards,
Saif

CycleMark, 2024-03-26:
Hi Saif,

Sorry too time to reply. Successfully upgraded ROS version and the Create 3. Pleased to report it looks much more stable now. A little slow to respond sometime but the topics are at least there - consistently.

Thanks for the help

Mark

smatarCPR, 2024-04-02:
Hello @CycleMark ,
That is good to hear. I will go ahead and close this issue. Should you encounter any follow up issues, do not hesitate to reach out by simply replying to this thread or opening up a new issue.
Best Regards,
Saif

closed 2024-04-02 by smatarCPR