turtlebot/turtlebot4 #371: Troubleshooting Connectivity and Controller Issues with TurtleBot4 Setup

Imani-Saba, 2024-03-04:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

sudo apt install ros-humble-desktop

### Type of issue

Networking

### Expected behaviour

Turtlebot4 connects to PC and controller

### Actual behaviour

Hello,

I Downloaded turtlebot4_standard_humble_1.0.0.zip then unzip it. I reimage SD card based on:
https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image

Now, the IP shows up (10.42.0.1) on Turtlebot4, then connect to Raspberry Pi by ssh ubuntu@10.42.0.1, then try to Connect the Raspberry Pi to my network. after that using Humble instruction, in my SSH session, run the TurtleBot 4 setup tool:
-Apply setting, select wifi mode (client), put my home wifi and password in SSID and Password, and then save and exit.

In the manual says "Once the Wi-Fi settings are applied, the Raspberry Pi will reboot and connect to your network. DHCP will assign it a new IP address. On the TurtleBot 4, this IP address will be shown at the top of the display.", but I have not received any new IP and the IP address does not change on the TurtolBot display and it still shows the IP address 10.42.0.1

The controller also does not connect to the TurtleBot. I used the manual, after turning on the controller with the home button the light starts blinking, and then becomes blue, but after a few seconds it goes off and no other buttons work to move the Turtlebot.

Could you please help me with these issues?
Thank you.

### To Reproduce

else

azhar-maxtech, 2024-03-06:
Regarding your 1st issue, I believe you did some mistake while putting your network details. Better install fresh image of ubuntu and try again. (you can recover the wifi setting by other method given in user manual but I prefer installing new image as it is very simple). I was able to fix this issues same way. 

Regarding controller connectivity, I am also facing the same issue and waiting for github to help. 

good luck.

RustyCPR, 2024-03-18:
Hello,
This issue is being closed due to being a duplicate of ticket #369  

Best Regards,
Rusty

RustyCPR, 2024-04-22:
Closing due to inactivity.

closed 2024-04-22 by RustyCPR