turtlebot/turtlebot4 #372: Turtlebot4 Showing LiDAR data in model but not map

LiamCClarke, 2024-03-04:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

?
Command returned:
dpkg-query: package 'ros-humble-turtlebot4-package_with_issue' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.

### Type of issue

Networking

### Expected behaviour

When I launch the SLAM map in Rviz2 a map is displayed that is captured by the Turtlebot4 LiDAR

### Actual behaviour

A small white circle is displayed and the terminal complains that the data from 'rplidar' isn being used due to the queue being full

### To Reproduce

1. Have network type set to Discovery Server
2. Launch Rviz2 to view the LiDAR map with this command "ros2 launch turtlebot4_viz view_robot.launch.py"
3. Observe Rviz2 display

### Other notes

A similar issue that was closed due to inactivity can be found here: https://github.com/turtlebot/turtlebot4/issues/288

I am currently using the TurtleBot4 with a discovery server due to being on a university wifi. I tried to view the mapping software for the turtlebot4 but was instead met with a white circle. The terminal window is also filled with warnings for the queue being full for the 'rplidar' (I don't currently have access to robot so I am unable to give exact message). As seen in the linked post, my model window looks the exact same with the model being loaded and the room boundaries discovered by the LiDAR displayed as dots, but when I try to view the data as a map, I just have the white circle as seen in the post.

The package version not being available on my system concerns me but I am certain I have installed everything, but am happy to do so again.

smatarCPR, 2024-03-11:
Hello @LiamCClarke ,
Thank you for reaching out on the official Turtlebot 4 Github page. Out of curiosity what is the fixed frame set to in RVIZ? Double check to ensure it is set to base link, or if using a map, that the map file is readable by RVIZ. Would it be possible to share an excerpt from the terminal output, as well as a screenshot from RVIZ. Looking forward to hearing from you soon.
Best Regards,
Saif

LiamCClarke, 2024-03-12:
Hi Saif, thanks for responding. I have just checked RViz2 with the TurtleBot model and sensor data launched and seen that the **Fixed Frame is set to "base_link"**
[image]
The entire window looks like this
[image]
As you can see, I am receiving the correct model, sensor data, and camera data. Here is the terminal for the model view
[image]

Here is an attempt at viewing a map:
First I sourced my Ros bash file, then I attempted to launch synchronous SLAM via my PC but was met by the 'rplidar_link' queue is full messages
[image]
I then launched another terminal to open Rviz2 with this command
`ros2 launch turtlebot4_viz view_robot.launch.py`
again showing the same warning messages.
[image]
Finally, the Rviz2 window looks like this
[image]

hilary-luo, 2024-03-19:
@LiamCClarke Can you share a full topic list when you are running slam?

LiamCClarke, 2024-03-21:
@hilary-luo Hello, Here is the `ros2 topic list` while I am running synchronous SLAM in another terminal window
[image]

LiamCClarke, 2024-03-21:
From what I can gather, Rviz is complaining that the frame [map] doesn't exist. Using the "2D Pose Estimate" feature doesn't create my [map] frame. Setting the `Fixed Frame` to something like `base_link` or `rplidar_link` gives me LiDAR data. I am assuming that there needs to be some kind of publisher that completes the TF tree for map?? I dont quite understand what I have just typed but seen a lot of advice pointing to a solution like this. Do you have any experience with this? Thanks

hilary-luo, 2024-03-21:
@LiamCClarke are you able to echo the create3 topics on your PC? (dock_status, odom etc.)

Also, can you confirm that if you run `ip route`, do you have an entry along the lines of `192.168.186.0/24 via 10.0.0.121 dev wlp0s20f3` where the second value is the IP address of your raspberry pi and the last is your wifi interface across which you are communicating with the raspberry pi?

LiamCClarke, 2024-03-22:
@hilary-luo , Yes I currently have an IP route for my turtlebot.

hilary-luo, 2024-03-22:
@LiamCClarke and you can successfully echo the topics I mentioned above? I have seen this sort of presentation before when there was an issue with the ip routing from the PC to the create3 (it was entered incorrectly). In that situation Nav2 was receiving the lidar information from the robot but was unable to do anything with it and unable to produce the map because it was missing the odometry from the create3. Hence, none of the lidar data is being processed and you get the queue is full messages.

LiamCClarke, 2024-03-22:
@hilary-luo sorry I missed that part. I tried echoing the odom topic and received no return message. I have attached the command I used as I am not too familiar with bash and ros so you can correct me if I have called it incorrectly
[image]

hilary-luo, 2024-03-25:
@LiamCClarke That is the correct command. Be aware though it could take as long as a few seconds to receive the first message depending on your network. If you did wait for a significant amount of time with no response, it confirms that you are not getting the create3 data successfully. Since you can see the create3 topics that tells me that the ROS system is set up correctly (both the create3 and your computer are checking in with the Discovery Server on the Pi). This therefore leads me to believe that there is a problem with your IP routing. Either there is an error with the IP routing table entry or you have a more complex networking setup which requires more changes. Given my understanding of your system so far, you should not be able to ping the create3 currently and the issue will be solved when you are able to ping the create3 (ping 192.168.186.2).

The following steps assume that you are not using a VM or container etc. Those elements make the networking more complicated. If you are using anything like this then please provide detailed information about your networking setup. If not, then go ahead:

First I would confirm that the route aforementioned is accurate to your system. Is the IP Address listed for your raspberry pi accurate? (Ensure that it is the IP address that your wifi router has assigned to the pi and if you have not reserved the IP address on your router know that it may change over reboots. Make sure that you can ping the raspberry pi at this address and if you have more than one turtlebot4, turn the rest of them off and make sure that you are pinging the correct pi). Do not use any environment variables to define this IP address, ensure that it is hard coded with the IP address itself.

Next I would confirm that the network also listed in that same route is correct and matches the network that your computer will need to use to connect to the robot. For example, if my computer is connected to the same wifi as the turtelbot4 then I will be using the wifi connection to communicate with the robot. Therefore, the wireless network interface name should be listed after `dev`. This should be the same name as is listed for the network if I run `ip a`. However, if I have a wired connection to my router while the turtlebot4 is on the wifi network provided by that same router then my ip route should list my wired network interface connection.

The goal right now is to be able to identify the issue. Depending on what the issue is, you may have to rerun the discovery server configure script to make the change permanent.

LiamCClarke, 2024-03-26:
@hilary-luo Thanks for the detailed reply. I tried to echo the odom topic and didn't receive a reply after several minutes. I am not using any type of VM or container but instead use a dual boot system though I don't think this affects anything. As you said, I am unable to ping the Create3.
[image]
The IP route seems correct and whenever my network does change the IP, I rerun the script to recreate the iproute and test that it works. my network interface is the same as what is listed after the dev.

arkcom55, 2024-03-27:
@hilary-luo I have had similar problems.  You need to be receiving the /odom topic else the TF mapping from /map to /base_link cannot be accomplished.

A couple of troubleshooting things you can try (works for me, using Discovery Server):
1.  Completely reboot the turtlebot4 (power off (pull off dock and press large button on base until it chimes, about 8 seconds) and then power on by putting on dock, wait for "chime" (indicates create3 base has restarted).  If you do not see 5 green LEDs on turtlebot4, then you need to reboot the pi (ssh into pi and "sudo reboot")
2.  Make sure to see a "192.168.186.0/24 via <turtlebot4-ip> dev <wifi-device>" when you do an "ip r"
3.  "ros2 topic echo /tf_static --once | head", check that the "header | stamp | sec" indicates a current time (int >1711571993), can check by putting number into something like https://www.unixtimestamp.com/.  The pi does not have a real-time clock and so can  have an invalid "old" time after a reboot.
4. "ros2 topic echo /odom --once | head" to also check "header | stamp | sec" as this is coming from the create3 base and it may not have properly synced its timestamp
5. If all the above works, then slam "map" should work

Note that due to a delay in the "/odom" topic coming from the create3 base relative to the topics coming from the pi (I typically see on the order of 10s of milliseconds), this typically screws up Rviz if trying to vie the LIDAR using "/odom" versus "base-link".  I did a workaround by rebroadcasting the /odom tf by adding 0.1 s to the header timestamp, but that is a bit of a kluge and not necessary when using SLAM/NAV2.

smatarCPR, 2024-04-09:
Hello @LiamCClarke ,
Have you tried the suggestions mentioned by @arkcom55 ? Were they successful? If you have any updates please feel free to share them. Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2024-04-22 by smatarCPR