turtlebot/turtlebot4 #373: IP configuration problem

ValentinLeglise, 2024-03-06:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

*

### Type of issue

Networking

### Expected behaviour

The robot generates a new IP adress.

### Actual behaviour

When configuring my customer's robot network, it doesn't generate an ip address after the first use. My customer is able to connect via ssh to the address 192.168.185.3 with an Ethernet cable, but not via WiFi. After trying to flash Rpi's SD card with the humble lite image on this link http://download.ros.org/downloads/turtlebot4/, the result is still the same.

My client then tried typing the command "ubuntu@ubuntu:~$ ip addr" and this is what he got:
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet [127.0.0.1/8](http://127.0.0.1/8) scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether d8:3a:dd:20:7e:cc brd ff:ff:ff:ff:ff:ff
    inet [192.168.185.3/24](http://192.168.185.3/24) brd 192.168.185.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::da3a:ddff:fe20:7ecc/64 scope link
       valid_lft forever preferred_lft forever
3: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 56:75:35:e2:81:cf brd ff:ff:ff:ff:ff:ff
    inet [192.168.186.3/24](http://192.168.186.3/24) brd 192.168.186.255 scope global usb0
       valid_lft forever preferred_lft forever
    inet6 fe80::5475:35ff:fee2:81cf/64 scope link
       valid_lft forever preferred_lft forever
4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether d8:3a:dd:20:7e:ce brd ff:ff:ff:ff:ff:ff

He also notices that he doesn't have all the topics by typing ros2 topic list:
/battery_state
/cmd_vel
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/parameter_events
/robot_description
/rosout
/scan
/stereo/depth
/tf
/tf_static
/wheel_status

and running the command ros2 topic echo /ip, it gets "data=UNKOWN".

I've looked at other tickets with the same problem but haven't found any other solutions. Can you help me ?

### To Reproduce

*

ValentinLeglise, 2024-03-08:
Hello,

Is it possible to have an answer for this please ?

Thank you.

smatarCPR, 2024-03-11:
Hello @ValentinLeglise ,
Thank you for reaching out on the official Turtlebot 4 Github page. Regarding the networking of the Turtlebot 4 unit, the Turtlebot4 does not "generate" an IP address. The Turtlebot4 sends a connection request to the network and the dhcp server on the network (generally your router on a home network) assigns it an IP address.

Double check that the connection settings are correct (check the  ssid or password). There is also a chance that the network is rejecting the connection. Are you aware of any additional settings that are present on the network, such as the need for a username or for an allowlist?

Are you able to connect via ethernet? If so, please provide the output for the following command: `journalctl -b | grep wlan0`

Looking forward to hearing from you soon.
Best Regards,
Saif

ValentinLeglise, 2024-03-28:
Hello ClearPath,

Sorry for this late reply.

My customer is sure of the SSID and password because he can connect to it with other Raspberry Pi.
Next, here's the result of the requested command:

ubuntu@ubuntu:~$ journalctl -b | grep wlan0
Mar 02 21:26:33 ubuntu cloud-init[682]: ci-info: | wlan0  | False |     .     |     .     |   .   | d8:3a:dd:20:7e:ce |
Mar 02 21:26:35 ubuntu NetworkManager[688]: <info>  [1677792395.9335] device (wlan0): driver supports Access Point (AP) mode
Mar 02 21:26:35 ubuntu NetworkManager[688]: <info>  [1677792395.9367] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/4)
Mar 02 21:26:35 ubuntu NetworkManager[688]: <info>  [1677792395.9383] device (wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Mar 02 21:26:35 ubuntu systemd-networkd[675]: wlan0: Link UP
Mar 02 21:26:36 ubuntu NetworkManager[688]: <info>  [1677792396.3627] device (wlan0): supplicant interface state: internal-starting -> disconnected
Mar 02 21:26:36 ubuntu NetworkManager[688]: <info>  [1677792396.3632] Wi-Fi P2P device controlled by interface wlan0 created
Mar 02 21:26:36 ubuntu NetworkManager[688]: <info>  [1677792396.3640] manager: (p2p-dev-wlan0): new 802.11 Wi-Fi P2P device (/org/freedesktop/NetworkManager/Devices/5)
Mar 02 21:26:36 ubuntu NetworkManager[688]: <info>  [1677792396.3661] device (p2p-dev-wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Mar 02 21:26:36 ubuntu NetworkManager[688]: <info>  [1677792396.3697] device (wlan0): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
Mar 02 21:26:36 ubuntu NetworkManager[688]: <info>  [1677792396.3767] device (p2p-dev-wlan0): state change: unavailable -> disconnected (reason 'none', sys-iface-state: 'managed')
Mar 02 21:26:36 ubuntu networkd-dispatcher[697]: ERROR:Unable to retrieve ESSID for wireless interface wlan0: no supported wireless tool installed

He did this from a new card after downloading this version: http://download.ros.org/downloads/turtlebot4/turtlebot4_lite_humble_1.0.0.zip

He opened the TurtleBot and plugged a keyboard, monitor and mouse into it + an Ethernet cable.
He still has 'ros2 topic echo /ip' where it says UNKNOWN'.

Best regards.

smatarCPR, 2024-04-03:
Hello @ValentinLeglise ,
For how long were the logs collected for? We typically see `networkd-dispatcher` error shortly after startup, with everything coming before typical for initial setup. Would it be possible to have the client collect additional logs, and keep the command running for a few minutes after startup? The most critical pieces of information can be collected after `networkd-dispatcher`. Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-05-22:
Hello,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

ValentinLeglise, 2024-06-21:
Hi Clearpath, 
I'm taking the liberty of reopening this ticket because my customer still has his problem (which he had put aside for a while). Here's a summary of the things he's tried in the meantime to try and solve the problem.

- A Create3 reset with: https://iroboteducation.github.io/create3_docs/webserver/about/
- A reflash of the RPI using humble lite image there: http://download.ros.org/downloads/turtlebot4/
- setup with the WIFI settings for the SSID named DEVELOPPEMENT and the associated password, as well as the ROS settings to use the Discovery Network configuration instead of Simple Discovery (because in his case, that's what made it work, i.e. he could see all the topics).

He is now able to connect to the Turtlebot using ssh (ssh ubuntu@10.31.95.6) from his PC (which is also on Wifi on the DEVELOPPEMENT SSID).
On the TurtleBot4, the ros2 topic list command gives :

[image]

From his PC, the same command gives (even after sourcing his environment with "source /opt/ros/humble/setup.bash":

[image]

For information on the TurtleBot4, here are the results of some commands and file contents (that was requested before the ticket was set aside): 

[image]

[image]

It's clearly a network problem because my customer tried to use his robot from home with an unrestricted network, which worked fine, and now, at work with a new unrestricted network, it doesn't work either.

We're stuck. Thanks in advance

ValentinLeglise, 2024-06-25:
Hello, 

Thank you for reopening this ticket.

Would it be possible to have new instructions to try to solve the problem?

 Best regards, Valentin.

smatarCPR, 2024-06-25:
Hello @ValentinLeglise ,
Thank you for sending a detailed reply. Based on the information we have the following questions with regards to the current setup of the Turtlebot 4:
- Has the client fully switched to a discovery server setup or are they still on a simple discovery setup?
- On this new unrestricted network can they:
- Ping the Raspberry Pi IP address and SSH into the Pi using the address?
- If they can reach the Raspberry Pi, have them try restarting the ros2 daemon on their computer using the command `ros2 daemon stop; ros2 daemon start`
- Some items to check on the Raspberry Pi: after connecting to the new work network, can the Pi ping 8.8.8.8 and www.google.com so that they can verify it is connected to the internet
- Would it be possible to have the client provide the latest logs from the Turtlebot 4 using the command `journalctl -b > turtlebot4_log.txt` and send us the resulting txt file.
Looking forward to hearing from you soon.
Best Regards,
Saif

ValentinLeglise, 2024-06-27:
Hello ClearPath,

Yes my client is fully in Discovery Server.

My client made these tests:

- From his computer (koala), SSH to 10.31.95.6, which is the Turtlebot's IP address. 
[image]

- And "ip addr"
[image]

- When SSH into RPi, he tried "ros2 deamon stop" "ros2 daemon start", and he stills see only the two topics with "ros2 topic list"

-Tried the pings:
[image]

-Result of "journalctl -b > turtlebot4_log.txt:
[turtlebot4_log.txt](https://github.com/user-attachments/files/16010073/turtlebot4_log.txt)

Hopefully this will solve our problem.
Best regards, Valentin.

ValentinLeglise, 2024-07-01:
Hello ClearPath,

Is it possible to have an answer for this please ?

Best regards, Valentin.

ValentinLeglise, 2024-07-04:
Hello ClearPath,

I'd like to ask you again for an answer. This customer would like to resolve his problem fairly quickly.

Best

smatarCPR, 2024-07-08:
Hello @ValentinLeglise ,
With regards to this is the following evaluation correct: on the new unrestricted network  everything is working except the user computer is not showing the topics from the robot.

Assuming this is the correct evaluation, did the client reconfigure the user PC to contact the discovery server at the new IP address after changing routers? If not then they should re-run the script to update the IP address. If it still isn't working then they should share the file at /etc/turtlebot4_discovery/setup.bash and the output from the IP route confirmation step from the instructions.

Let me know how these recommendations work for you. Looking forward to hearing from you soon.
Best Regards,
Saif

ValentinLeglise, 2024-07-09:
Hello, thank you for your reply.

Not all of it. On the new network that he is using in his academy (without restrictions), he has connected his PC and his robot via Discovery Server. With his PC he tried two things: 1. in SSH inside the RPi, he obtained all the topics by doing "ros2 topic list". 2. Using a normal shell (so the PC he's using is connected to the same network as the robot), he only gets the /parameter_events and /rosout topics.
So no, the evaluation is not completely correct.

I'm still going to suggest what you've shared with me, but if you have any other suggestions in relation to the good rating I've given you, I'd like to hear them. Thank you for your help.

Best regards.

smatarCPR, 2024-07-10:
Hello @ValentinLeglise ,
With regards to the output from the IP route confirmation step from the instructions this is referring to this section in the user manual: https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html#user-pc

In that section, after setting up the IP route it is recommended to call the command `ip route` and the output should be 192.168.186.0/24 via 10.0.0.121 dev wlp0s20f3 with 10.0.0.121 would the Raspberry Pi IP, and wlp0s20f3 would be the wi-fi interface for the client's computer.

Let me know the results of the checks requested. Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2024-08-19 by smatarCPR