turtlebot/turtlebot4 #374: Controller not connecting with Turtlebot4

azhar-maxtech, 2024-03-06:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

[turtlebot4_standard_humble_1.0.0.zip] (http://download.ros.org/downloads/turtlebot4/turtlebot4_standard_humble_1.0.0.zip)

### Type of issue

Other

### Expected behaviour

I want to connect the controller with turtlebot4 via bluetooth after installing fresh image of Ubuntu in the Raspberry Pi 4. 

### Actual behaviour

Controller does not connect with turtlebot4 and shows blues error. 
[image]

### Error messages

```bash
Failed to connect: org.blues.Error.Failed br-connection-create-socket
```

### To Reproduce

[image]

### Other notes

I did “untrust”, “remove”, “scan on”, “trust”, “Pair”, “connect” but still the problem is there. I have tried connecting so many times but no success. Even tried resetting the controller. I talked to clearpath and they advised me to keep trying connecting until it connects but it doesn't.

RustyCPR, 2024-03-11:
Hi @azhar-maxtech 
Can you try this: https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#3-no-default-controller-available
If it does not work - please share the full terminal outputs from the untrust / remove / trust / pair and connect commands being run in the terminal.

RustyCPR, 2024-03-18:
Hi @azhar-maxtech - any update on your issue? 
Best regards,
Rusty

RustyCPR, 2024-04-02:
Closing due to inactivity.

closed 2024-04-02 by RustyCPR