turtlebot/turtlebot4 #376: Can't map or start a world with TurtleBot 4.

ajjaya18, 2024-03-07:
### Robot Model

Select One

### ROS distro

Select One

### Networking Configuration

Select One

### OS

Select One

### Built from source or installed?

Select One

### Package version

I am running the packages that come in the tutorial for ROS Galactic."

### Type of issue

Select One

### Expected behaviour

Mapping the depot world with all the present objects, running the robot in created worlds, and having an exact simulation and mapping.

### Actual behaviour

t's not possible to map in the depot world; it doesn't recognize any object and creates a completely empty map. It doesn't recognize the walls, boxes, or anything else. The ignition.launch.py file doesn't run with another world.

### Error messages

```bash
I run 3 terminal when I started to map, 
when I run ros2 launch turtlebot4_viz view_robot.launch.py I have this messages in the terminal. 
rviz2-1] [INFO] [1709828332.787346081] [rviz2]: Trying to create a map of size 1 x 1 using 1 swatches
[rviz2-1] [ERROR] [1709828332.822992388] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
[rviz2-1] active samplers with a different type refer to the same texture image unit
[rviz2-1] [INFO] [1709828340.595831080] [rviz2]: Trying to create a map of size 2 x 1 using 1 swatches
[rviz2-1] [INFO] [1709828344.915688166] [rviz2]: Trying to create a map of size 2 x 4 using 1 swatches
[rviz2-1] [INFO] [1709828346.292244643] [rviz2]: Trying to create a map of size 2 x 5 using 1 swatches
[rviz2-1] [INFO] [1709828347.666290991] [rviz2]: Trying to create a map of size 2 x 6 using 1 swatches
[rviz2-1] [INFO] [1709828351.606389143] [rviz2]: Trying to create a map of size 2 x 7 using 1 swatches
[rviz2-1] [INFO] [1709828352.913994110] [rviz2]: Trying to create a map of size 3 x 9 using 1 swatches
[rviz2-1] [INFO] [1709828354.292360510] [rviz2]: Trying to create a map of size 4 x 11 using 1 swatches
[rviz2-1] [INFO] [1709828355.604214327] [rviz2]: Trying to create a map of size 4 x 13 using 1 swatches
[rviz2-1] [INFO] [1709828356.948042847] [rviz2]: Trying to create a map of size 5 x 15 using 1 swatches
[rviz2-1] [INFO] [1709828358.292333768] [rviz2]: Trying to create a map of size 6 x 17 using 1 swatches
[rviz2-1] [INFO] [1709828359.699102871] [rviz2]: Trying to create a map of size 6 x 19 using 1 swatches
[rviz2-1] [INFO] [1709828361.012587453] [rviz2]: Trying to create a map of size 7 x 20 using 1 swatches
[rviz2-1] [INFO] [1709828362.323654823] [rviz2]: Trying to create a map of size 7 x 22 using 1 swatches
[rviz2-1] [INFO] [1709828363.637520799] [rviz2]: Trying to create a map of size 8 x 23 using 1 swatches
[rviz2-1] [INFO] [1709828364.632788972] [rviz2]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 0.064 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1709828364.709500043] [rviz2]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 0.080 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1709828364.789899371] [rviz2]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 0.096 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1709828364.884067898] [rviz2]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 0.112 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1709828364.983141419] [rviz2]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 0.128 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1709828365.011500578] [rviz2]: Trying to create a map of size 8 x 25 using 1 swatches
[rviz2-1] [INFO] [1709828365.065523395] [rviz2]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 0.144 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1709828365.141696490] [rviz2]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 0.160 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1709828365.236211862] [rviz2]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 0.176 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1709828365.310672948] [rviz2]: Message Filter dropping message: frame 'turtlebot4/rplidar_link/rplidar' at time 0.192 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1709828366.291351323] [rviz2]: Trying to create a map of size 9 x 27 using 1 swatches
[rviz2-1] [INFO] [1709828367.699225829] [rviz2]: Trying to create a map of size 9 x 28 using 1 swatches
[... 441 lines cut ...]

ros2 launch turtlebot4_navigation slam_sync.launch.py

2. run Launch Rviz2

To visualise the map, launch Rviz2 with the view_robot launch file.

ros2 launch turtlebot4_viz view_robot.launch.py
3. run the ignition gazebo 
Running the simulator with default settings:

ros2 launch turtlebot4_ignition_bringup ignition.launch.py

### Other notes

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ajjaya18, 2024-03-07:
@hilary-luo

ajjaya18, 2024-03-07:
I have this problem when I launch my own world 
andrius@andrius-HP-Pavilion-Laptop-15-eh0xxx:~/turtlebot4_ws$ ros2 launch turtlebot4_ignition_bringup ignition.launch.py world:=mi_mundo

[INFO] [launch]: All log files can be found below /home/andrius/.ros/log/2024-03-07-16-14-10-834014-andrius-HP-Pavilion-Laptop-15-eh0xxx-178044
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [178048]
[INFO] [parameter_bridge-2]: process started with pid [178051]
[INFO] [parameter_bridge-3]: process started with pid [178053]
[INFO] [parameter_bridge-4]: process started with pid [178055]
[INFO] [parameter_bridge-5]: process started with pid [178057]
[INFO] [parameter_bridge-6]: process started with pid [178059]
[INFO] [parameter_bridge-7]: process started with pid [178061]
[INFO] [parameter_bridge-8]: process started with pid [178063]
[INFO] [parameter_bridge-9]: process started with pid [178065]
[INFO] [parameter_bridge-10]: process started with pid [178067]
[INFO] [parameter_bridge-11]: process started with pid [178069]
[INFO] [parameter_bridge-12]: process started with pid [178071]
[INFO] [parameter_bridge-13]: process started with pid [178073]
[INFO] [parameter_bridge-14]: process started with pid [178075]
[INFO] [robot_state_publisher-15]: process started with pid [178077]
[INFO] [robot_state_publisher-16]: process started with pid [178079]
[INFO] [static_transform_publisher-17]: process started with pid [178081]
[INFO] [create-18]: process started with pid [178092]
[INFO] [create-19]: process started with pid [178137]
[INFO] [spawner-20]: process started with pid [178168]
[INFO] [hazards_vector_publisher-21]: process started with pid [178221]
[INFO] [ir_intensity_vector_publisher-22]: process started with pid [178291]
[INFO] [motion_control-23]: process started with pid [178293]
[INFO] [wheel_status_publisher-24]: process started with pid [178310]
[INFO] [mock_publisher-25]: process started with pid [178314]
[INFO] [robot_state-26]: process started with pid [178349]
[INFO] [kidnap_estimator_publisher-27]: process started with pid [178365]
[INFO] [ui_mgr-28]: process started with pid [178368]
[INFO] [pose_republisher_node-29]: process started with pid [178370]
[INFO] [sensors_node-30]: process started with pid [178401]
[INFO] [interface_buttons_node-31]: process started with pid [178422]
[INFO] [turtlebot4_node-32]: process started with pid [178424]
[INFO] [turtlebot4_ignition_hmi_node-33]: process started with pid [178426]
[INFO] [component_container-34]: process started with pid [178455]
[INFO] [static_transform_publisher-35]: process started with pid [178457]
[INFO] [static_transform_publisher-36]: process started with pid [178459]
[parameter_bridge-12] [INFO] [1709846054.804184357] [turtlebot4.hmi_led_msg_bridge]: Passing message from ROS std_msgs/msg/Int32 to Ignition ignition.msgs.Int32 (showing msg only once per type)
[robot_state_publisher-15] The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-15] Link bump_front_center had 0 children
[robot_state_publisher-15] Link bump_front_left had 0 children
[robot_state_publisher-15] Link bump_front_right had 0 children
[robot_state_publisher-15] Link bump_left had 0 children
[robot_state_publisher-15] Link bump_right had 0 children
[robot_state_publisher-15] Link button_1 had 0 children
[robot_state_publisher-15] Link button_2 had 0 children
[robot_state_publisher-15] Link button_power had 0 children
[robot_state_publisher-15] Link cliff_front_left had 0 children
[robot_state_publisher-15] Link cliff_front_right had 0 children
[robot_state_publisher-15] Link cliff_side_left had 0 children
[robot_state_publisher-15] Link cliff_side_right had 0 children
[robot_state_publisher-15] Link front_caster_link had 0 children
[robot_state_publisher-15] Link front_left_bottom_weight_block had 0 children
[robot_state_publisher-15] Link front_left_top_weight_block had 0 children
[robot_state_publisher-15] Link front_right_bottom_weight_block had 0 children
[robot_state_publisher-15] Link front_right_top_weight_block had 0 children
[robot_state_publisher-15] Link imu_link had 0 children
[robot_state_publisher-15] Link ir_intensity_front_center_left had 0 children
[robot_state_publisher-15] Link ir_intensity_front_center_right had 0 children
[robot_state_publisher-15] Link ir_intensity_front_left had 0 children
[robot_state_publisher-15] Link ir_intensity_front_right had 0 children
[robot_state_publisher-15] Link ir_intensity_left had 0 children
[robot_state_publisher-15] Link ir_intensity_right had 0 children
[robot_state_publisher-15] Link ir_intensity_side_left had 0 children
[robot_state_publisher-15] Link ir_omni had 0 children
[robot_state_publisher-15] Link mouse had 0 children
[... 109 lines cut ...]
[kidnap_estimator_publisher-27] [INFO] [1709846054.808510766] [kidnap_estimator_publisher]: Subscription to topic: /hazard_detection
[ui_mgr-28] [INFO] [1709846054.435703286] [ui_mgr]: Subscription to topic: /cmd_lightring
[ui_mgr-28] [INFO] [1709846054.438078505] [ui_mgr]: Subscription to topic: /cmd_audio
[turtlebot4_node-32] [INFO] [1709846054.595959122] [turtlebot4_node]: Init Turtlebot4 Node Main
[turtlebot4_node-32] [INFO] [1709846054.614528111] [turtlebot4_node]: Buttons Init
[turtlebot4_node-32] [INFO] [1709846054.615955952] [turtlebot4_node]: Init Display
[... cut ...]
[spawner-20] [INFO] [1709846059.490433451] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[create-18] [INFO] [1709846059.543992171] [ros_ign_gazebo]: Requesting list of world names.
[create-19] [INFO] [1709846059.836461021] [ros_ign_gazebo]: Requesting list of world names.
[spawner-20] [INFO] [1709846061.502772696] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)

@hilary-luo  check please

hilary-luo, 2024-03-11:
Please fill in the remaining requested information at the top of the ticket (model, ROS distro, Networking Configuration, OS, Built from source or installed?, Package version (with the package number of what you installed) and Type of issue). This information is important for us to be able to help you. 

Without that information listed above, the best suggestions I have for right now:
Do you get the same issue if you use the single command launch?
`ros2 launch turtlebot4_ignition_bringup ignition.launch.py slam:=sync nav2:=true rviz:=true`
This command will launch the simulation, slam, nav2 and rviz all at once.

If you are still having the same issue after doing this, can you confirm:
a) whether you can see and echo the topics from the simulation? and
b) what ROS related settings you are sourcing in your terminal (likely put in your .bashrc when you set up your networking)?

ajjaya18, 2024-03-11:
Well, I haven't done any network configuration; I'm just using simulation as I don't have a physical robot. Here is my .bashrc file:
# ~/.bashrc: executed by bash(1) for non-login shells.
# see /usr/share/doc/bash/examples/startup-files (in the package bash-doc)
# for examples

# If not running interactively, don't do anything
case $- in
    *i*) ;;
      *) return;;
esac

# don't put duplicate lines or lines starting with space in the history.
# See bash(1) for more options
HISTCONTROL=ignoreboth

# append to the history file, don't overwrite it
shopt -s histappend

# for setting history length see HISTSIZE and HISTFILESIZE in bash(1)
HISTSIZE=1000
HISTFILESIZE=2000

# check the window size after each command and, if necessary,
# update the values of LINES and COLUMNS.
shopt -s checkwinsize

# If set, the pattern "**" used in a pathname expansion context will
# match all files and zero or more directories and subdirectories.
#shopt -s globstar

# make less more friendly for non-text input files, see lesspipe(1)
[ -x /usr/bin/lesspipe ] && eval "$(SHELL=/bin/sh lesspipe)"

# set variable identifying the chroot you work in (used in the prompt below)
if [ -z "${debian_chroot:-}" ] && [ -r /etc/debian_chroot ]; then
    debian_chroot=$(cat /etc/debian_chroot)
fi

# set a fancy prompt (non-color, unless we know we "want" color)
case "$TERM" in
    xterm-color|*-256color) color_prompt=yes;;
esac

# uncomment for a colored prompt, if the terminal has the capability; turned
# off by default to not distract the user: the focus in a terminal window
# should be on the output of commands, not on the prompt
#force_color_prompt=yes

if [ -n "$force_color_prompt" ]; then
    if [ -x /usr/bin/tput ] && tput setaf 1 >&/dev/null; then
	# We have color support; assume it's compliant with Ecma-48
	# (ISO/IEC-6429). (Lack of such support is extremely rare, and such
	# a case would tend to support setf rather than setaf.)
	color_prompt=yes
    else
	color_prompt=
    fi
fi

if [ "$color_prompt" = yes ]; then
    PS1='${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
else
    PS1='${debian_chroot:+($debian_chroot)}\u@\h:\w\$ '
fi
unset color_prompt force_color_prompt

# If this is an xterm set the title to user@host:dir
case "$TERM" in
xterm*|rxvt*)
    PS1="\[\e]0;${debian_chroot:+($debian_chroot)}\u@\h: \w\a\]$PS1"
    ;;
*)
    ;;
esac

# enable color support of ls and also add handy aliases
if [ -x /usr/bin/dircolors ]; then
    test -r ~/.dircolors && eval "$(dircolors -b ~/.dircolors)" || eval "$(dircolors -b)"
    alias ls='ls --color=auto'
    #alias dir='dir --color=auto'
    #alias vdir='vdir --color=auto'

    alias grep='grep --color=auto'
    alias fgrep='fgrep --color=auto'
    alias egrep='egrep --color=auto'
fi

# colored GCC warnings and errors
#export GCC_COLORS='error=01;31:warning=01;35:note=01;36:caret=01;32:locus=01:quote=01'

# some more ls aliases
alias ll='ls -alF'
alias la='ls -A'
alias l='ls -CF'

# Add an "alert" alias for long running commands.  Use like so:
#   sleep 10; alert
alias alert='notify-send --urgency=low -i "$([ $? = 0 ] && echo terminal || echo error)" "$(history|tail -n1|sed -e '\''s/^\s*[0-9]\+\s*//;s/[;&|]\s*alert$//'\'')"'

# Alias definitions.
# You may want to put all your additions into a separate file like
# ~/.bash_aliases, instead of adding them here directly.
# See /usr/share/doc/bash-doc/examples in the bash-doc package.

if [ -f ~/.bash_aliases ]; then
    . ~/.bash_aliases
fi

# enable programmable completion features (you don't need to enable
# this, if it's already enabled in /etc/bash.bashrc and /etc/profile
# sources /etc/bash.bashrc).
if ! shopt -oq posix; then
  if [ -f /usr/share/bash-completion/bash_completion ]; then
    . /usr/share/bash-completion/bash_completion
  elif [ -f /etc/bash_completion ]; then
    . /etc/bash_completion
  fi
fi

ajjaya18, 2024-03-11:
[image]
I don't know if the configuration of the rplidar urdf.xacrp plugin is correct because I get strange values ​​when I do ros2 topic echo /scan that I showed you in a screenshot above, can you please check it
@hilary-luo help me please

hilary-luo, 2024-03-13:
Do you have a GPU? if you do have a GPU it may not actually be being used by the simulation and that could be the causing factor. If you do have a GPU please make sure that it is being used properly. If you don't have a GPU or it is already being used properly, continue with the following:

Can you visualize the lidar rays in gazebo and confirm if you are seeing the same issue as this ticket? https://github.com/turtlebot/turtlebot4_simulator/issues/20 If you are then try the solution that was found [there](https://github.com/turtlebot/turtlebot4_simulator/issues/20#issuecomment-1302831141).

Unfortunately you are using an end of life version of ROS so it no longer receives updates. Our support abilities are limited for these older distros, I would advise that you update to Humble when you can. I believe this issue is not present in Humble.

ajjaya18, 2024-03-13:
@hilary-luo  I don't have GPU

ajjaya18, 2024-03-13:
[image]
I managed to activate the mapping feature, but I can't see the rays because I don't have a GPU. Also, I can't move the TurtleBot4 much; I'm getting something about ignoring velocities in the terminal, as shown in the image. If you could help me, that would be great. @hilary-luo @roni-kreinin

hilary-luo, 2024-03-13:
@ajjaya18 Not having a GPU dopes not mean that you cannot see the rays. The lidar points are showing in rviz in the image you shared. If you want to visualize the rays in gazebo then you need to turn that on (vizualize lidar -> select the lidar).

That message that you are getting means that you are trying to command the robot to move while it is completing an autonomous behaviour. You need to wait until it is done the autonomous behaviour (such as undocking or backing up because it collided with something) before driving it.

ajjaya18, 2024-03-13:
@hilary-luo can you explain more about how to visualize the rays  in gazebo please.

hilary-luo, 2024-03-18:
Select the options at the top right
[image]
Start typing to find visualize lidar and select it
[image]
Scroll down until you find the Visualize Lidar section. Click on the refresh button and select the correct lidar that you want to visualize from the drop down.
[image]
You should now see the rays
[image]

If you don't see the lidar, ensure that your simulation is playing (should see a pause symbol in the bottom left).

ajjaya18, 2024-03-18:
[image]
Can't put anything in this section, please look at the image.

ajjaya18, 2024-03-18:
[image]
[image]
[image]
Do you know why I'm getting this error with tf or how can I solve it? Does it have to do with the wheels or does it not affect mapping and navigation? @hilary-luo

hilary-luo, 2024-03-18:
> Can't put anything in this section, please look at the image.

You need to first press the refresh button. 

> Do you know why I'm getting this error with tf or how can I solve it? Does it have to do with the wheels or does it not affect mapping and navigation?

If you are referring to the wheel / wheel drop tf errors in the simulation, that is fine and should not affect your mapping or navigation. I believe it is fixed in Humble.

ajjaya18, 2024-03-18:
> If you are referring to the wheel / wheel drop tf errors in the simulation, that is fine and should not affect your mapping or navigation. I believe it is fixed in Humble.
ok, thanks but you know the origin of this problem?

ajjaya18, 2024-03-18:
> >If you are referring to the wheel / wheel drop tf errors in the simulation, that is fine and should not affect your mapping or navigation. I believe it is fixed in Humble.

ok, thanks but you know the origin of this problem? because I have been reviewing all the URDF files and they seem to be fine. @hilary-luo

ajjaya18, 2024-03-19:
Hi, @hilary-luo  Are there any issues related to launching your own world?

hilary-luo, 2024-03-19:
> ok, thanks but you know the origin of this problem? because I have been reviewing all the URDF files and they seem to be fine

I don't recall off the top of my head and unfortunately don't have the time to chase this down. You can try to take a look at what has changed between the version that you are using and the latest Humble version but there are likely other changes in there as well.

> Are there any issues related to launching your own world?

No, you just need to follow the instructions on how to do so. There are some notes on the website in the documentation and if you look at some of the other issues I have walked people through launching their own worlds. In short, you have to install the turtlebot4_simulator package from source and place your new world beside the existing ones there. Make sure the file name matches the world name inside the file. Then you should be able to build and source it and then launch it with the name of your custom world.

hilary-luo, 2024-03-25:
Closing this issue because discussion has shifted to the new one.

closed 2024-03-25 by hilary-luo