turtlebot/turtlebot4 #377: oakd compressed topics not appearing

arkcom55, 2024-03-08:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Unknown

### Type of issue

Camera

### Expected behaviour

Should see ROS topics:
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora

### Actual behaviour

Actaully see ROS topics:
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw

### To Reproduce

Just start turtlebot4 and undock

### Other notes

Note this is NOT the same issue as mentioned in https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#4-oakd-topics-not-present which was related to NO oakd ROS topics is robot docked.

This issue is when undocked.

Have on Turtlebot4 displaying all the expected topics and a number only displaying the two topics.

arkcom55, 2024-03-08:
andrew.kostiuk@usask.ca

RustyCPR, 2024-03-11:
Hi Andrew - thank you for submitting the ticket. 
Can you please provide the outputs of the below from both TB4 (with and without the extra topics) 

`ros-humble-depthai-ros-driver`
`ros-humble-turtlebot4-robot`

RustyCPR, 2024-03-18:
Hi @arkcom55  - any updates? did you resolve the issue?  
Thanks,
Rusty

arkcom55, 2024-03-25:
Updated and now just:
oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw

Perhaps I am looking at this wrong.  Do I just want to do something with "ros2 run image_transport republish ..." to get compressed images?

RustyCPR, 2024-03-25:
Hi @arkcom55 
These are the standard ROS topics you should see out of the box. 

Please see this reply related to additional depth camera topics: https://github.com/turtlebot/turtlebot4/issues/277#issuecomment-1815024838

RustyCPR, 2024-04-09:
Closing due to no updates or activity.

closed 2024-04-09 by RustyCPR