turtlebot/turtlebot4 #379: messages synchronization

alain-lnpcrd, 2024-03-15:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

```
moi@rpi4b-u2204d-turtlebot4:~$ dpkg -s ros-$ROS_DISTRO-turtlebot4-python-tutorials
Package: ros-humble-turtlebot4-python-tutorials
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 80
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: arm64
Version: 1.0.1-1jammy.20240220.180339
Depends: ros-humble-irobot-create-msgs, ros-humble-rclpy, ros-humble-turtlebot4-navigation, ros-humble-ros-workspace
Description: TurtleBot 4 Python Tutorials

```

### Expected behaviour

create3 access to ntp on raspberry pi to synchronize

### Actual behaviour

extract logs create3
```
Mar 15 13:26:59 iRobot-9FBCB799E5E34EE0B1D045D8C1944997 user.notice ntpd: ntpd: reply from 192.168.186.3: delay 0.037659 is too high, ignoring
[...]
Mar 15 13:28:19 iRobot-9FBCB799E5E34EE0B1D045D8C1944997 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 33s
[...]
Mar 15 13:29:09 iRobot-9FBCB799E5E34EE0B1D045D8C1944997 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 66s
[...]
```

### Error messages

```bash
moi@rpi4b-u2204d-turtlebot4:~$ nmap 192.168.186.3
From Nmap 7.94 (https://nmap.org) on 03/14/2024 at 1:34 p.m. CET
Nmap scan report for rpi4b-u2204d-turtlebot4 (192.168.186.3)
The host is operational (0.00082 s latency).
Not shown: 996 TCP ports closed (conn-denied)
PORT STATE SERVICE
22/tcp open ssh
5900/tcp open vnc
5999/tcp opens ncd-conf
8080/tcp opens http proxy
Nmap completed: 1 IP address (1 host enabled) scanned in 0.75 seconds

moi@rpi4b-u2204d-turtlebot4:~$ nmap 192.168.186.1
From Nmap 7.94 (https://nmap.org) on 03/14/2024 at 1:35 p.m. CET
Note: The host appears to be down. If it's really active, but blocking our ping probes, try -Pn Nmap done: 1 IP address (0 hosts enabled) scanned in 3.21 seconds
me@rpi4b-u2204d-turtlebot4:~$

moi@rpi4b-u2204d-turtlebot4:~$ nmap -Pn 192.168.186.1
From Nmap 7.94 (https://nmap.org) on 03/14/2024 at 1:41 p.m. CET
Nmap scan report for 192.168.186.1
The host is operational (0.065 s latency).
All 1000 ports scanned on 192.168.186.1 are in ignored states.
Not shown: 990 TCP ports filtered (no response), 10 TCP ports filtered (host unreach)
Nmap completed: 1 IP address (1 host enabled) analyzed in 6.49 seconds
me@rpi4b-u2204d-turtlebot4:~$
```

### To Reproduce

same configuration
installation of chrony

### Other notes

https://github.com/iRobotEducation/create3_docs/discussions/572

hilary-luo, 2024-03-18:
@alain-lnpcrd You mention "[trying] to to configure an ntp server on the raspberrypi". Can you clarify what you mean by that? If you have manually created your own ntp server please try reverting to the [default one](https://github.com/turtlebot/turtlebot4_setup/blob/humble/etc/chrony/chrony.conf) that was created for the express purpose of syncing the create3. The recommended configuration on the create3 side is indicated here: https://github.com/turtlebot/turtlebot4/issues/216

If that does not resolve your issue, can you please clarify what the symptoms are of your issue? (are you getting errors when you try and operate?) You have shown here some logs where the ntp sync had issues but in the full log from your other discussion post there are several examples of it executing properly and gaining time sync.

alain-lnpcrd, 2024-03-18:
I use chrony as shown in the create3 manual
the errors are indicated in actual-behavior
The solution was to replace "server 192.168.186.186.186" and server "192.168.186.3 iburst" with the IP of the raspberrypi "server 192.168.1.88 iburst" in the net.conf of create3

closed 2024-03-18 by alain-lnpcrd