turtlebot/turtlebot4 #380: My TurtleBot 4 cannot run VIZ to simulation

Imani-Saba, 2024-03-18:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

git clone https://github.com/turtlebot/turtlebot4.git -b humble

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Hello,

I installed all packages in raspberry pi to run VIZ for simulating map according to:
https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_common.html

But, when run: ros2 launch turtlebot4_viz view_robot.launch.py
I got this error:
[ERROR] [rviz2-1]: process has died [pid 3929, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /opt/ros/humble/share/turtlebot4_viz/rviz/robot.rviz --ros-args -r __node:=rviz2 -r __ns:=/ --params-file /tmp/launch_params_1y8hb5wo -r /tf:=tf -r /tf_static:=tf_static'].

And, RVIZ app did not open.

Could you please help me with this issue?

Thanks.

### Actual behaviour

sudo apt update
sudo apt install ros-humble-turtlebot4-description \
ros-humble-turtlebot4-msgs \
ros-humble-turtlebot4-navigation \
ros-humble-turtlebot4-node

cd ~/turtlebot4_ws/src
git clone https://github.com/turtlebot/turtlebot4.git -b humble

cd ~/turtlebot4_ws
rosdep install --from-path src -yi

source /opt/ros/humble/setup.bash
colcon build --symlink-install

ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py

ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true

ros2 launch turtlebot4_viz view_robot.launch.py

### Error messages

```bash
ubuntu@ubuntu:~/turtlebot4_ws$ ros2 launch turtlebot4_viz view_robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-03-18-04-46-38-024641-ubuntu-4125
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [4126]
[rviz2-1] qt.qpa.xcb: could not connect to display 
[rviz2-1] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-1] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-1] 
[rviz2-1] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-1] 
[ERROR] [rviz2-1]: process has died [pid 4126, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /opt/ros/humble/share/turtlebot4_viz/rviz/robot.rviz --ros-args -r __node:=rviz2 -r __ns:=/ --params-file /tmp/launch_params_z_07rfkv -r /tf:=tf -r /tf_static:=tf_static'].
```

### To Reproduce

ubuntu@ubuntu:~$ ros2 topic list
/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/dock_status
/hazard_detection
/imu
/interface_buttons
/ir_intensity
/ir_opcode
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_state/transition_event
/rosout
/slip_status
/static_transform/transition_event
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

RustyCPR, 2024-03-18:
Hi @Imani-Saba 

Reviewing the [Rviz2 launch instructions](https://turtlebot.github.io/turtlebot4-user-manual/software/rviz.html#rviz2) are you running this on a laptop/ PC with Ubuntu 22.04 with the [Turtlebot4 Desktop package installed](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_desktop.html#turtlebot-4-desktop)? 
Best regards,
Rusty

Imani-Saba, 2024-03-19:
Hi,
I got two errors when I run the Source installation command.

The following is the process I followed and the error shows up when I run these commands in my PC with Ubuntu 22.4:
According to TurtleBot 4 Desktop guidance:

1- ssh ubuntu@MAC address
2- sudo apt update
3- sudo apt install ros-humble-turtlebot4-desktop
4- cd ~/turtlebot4_ws/src
git clone https://github.com/turtlebot/turtlebot4_desktop.git -b humble
6- cd ~/turtlebot4_ws
rosdep install --from-path src -yi
7- source /opt/ros/humble/setup.bash
colcon build --symlink-install

8- source /opt/ros/humble/setup.bash
colcon build --symlink-install

Starting >>> turtlebot4_description
Starting >>> turtlebot4_msgs
Starting >>> turtlebot4_ignition_gui_plugins                                               
Starting >>> turtlebot4_navigation
Finished <<< turtlebot4_description [2.22s]                                   
Starting >>> turtlebot4_viz                                              
Finished <<< turtlebot4_navigation [5.41s]                                
Finished <<< turtlebot4_msgs [6.87s]                                   
Starting >>> turtlebot4_ignition_toolbox
Starting >>> turtlebot4_node                                                         
Finished <<< turtlebot4_node [2.52s]                                                  
Finished <<< turtlebot4_viz [7.75s]                                                   
Starting >>> turtlebot4_desktop
Finished <<< turtlebot4_desktop [5.26s]                                                    
[Processing: turtlebot4_ignition_gui_plugins, turtlebot4_ignition_toolbox]                 
[Processing: turtlebot4_ignition_gui_plugins, turtlebot4_ignition_toolbox]                     
--- stderr: turtlebot4_ignition_gui_plugins 
In file included from /usr/include/ignition/plugin1/gz/plugin/Register.hh:22,
                 from /usr/include/ignition/plugin1/ignition/plugin/Register.hh:18,
                 from /home/ubuntu/turtlebot4_ws/src/turtlebot4_simulator/turtlebot4_ignition_gui_plugins/Turtlebot4Hmi/Turtlebot4Hmi.cc:26:
/usr/include/ignition/plugin1/gz/plugin/detail/Register.hh:28:10: fatal error: gz/utilities/SuppressWarning.hh: No such file or directory
   28 | #include <gz/utilities/SuppressWarning.hh>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [Turtlebot4Hmi/CMakeFiles/Turtlebot4Hmi.dir/build.make:99: Turtlebot4Hmi/CMakeFiles/Turtlebot4Hmi.dir/Turtlebot4Hmi.cc.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:153: Turtlebot4Hmi/CMakeFiles/Turtlebot4Hmi.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< turtlebot4_ignition_gui_plugins [1min 17s, exited with code 2]
Aborted  <<< turtlebot4_ignition_toolbox [1min 46s]                                          

Summary: 6 packages finished [1min 54s]
  1 package failed: turtlebot4_ignition_gui_plugins
  1 package aborted: turtlebot4_ignition_toolbox
  1 package had stderr output: turtlebot4_ignition_gui_plugins
  2 packages not processed
_________________________________________________

9- ros2 launch turtlebot4_viz view_robot.launch.py
ubuntu@ubuntu:~/turtlebot4_ws$ ros2 launch turtlebot4_viz view_robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-03-19-19-53-50-667918-ubuntu-3682
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [3683]
[rviz2-1] qt.qpa.xcb: could not connect to display 
[rviz2-1] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-1] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-1] 
[rviz2-1] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-1] 
[ERROR] [rviz2-1]: process has died [pid 3683, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /opt/ros/humble/share/turtlebot4_viz/rviz/robot.rviz --ros-args -r __node:=rviz2 -r __ns:=/ --params-file /tmp/launch_params_xq1nqao0 -r /tf:=tf -r /tf_static:=tf_static'].
_______________________________________________________

ubuntu@ubuntu:~/turtlebot4_ws$ ros2 launch turtlebot4_viz view_diagnostics.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-03-19-19-59-07-838503-ubuntu-3697
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rqt_robot_monitor-1]: process started with pid [3698]
[rqt_robot_monitor-1] could not connect to display 
[rqt_robot_monitor-1] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rqt_robot_monitor-1] 
[rqt_robot_monitor-1] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rqt_robot_monitor-1] 
[ERROR] [rqt_robot_monitor-1]: process has died [pid 3698, exit code 1, cmd '/opt/ros/humble/lib/rqt_robot_monitor/rqt_robot_monitor --ros-args -r __ns:=/ -r /diagnostics_agg:=diagnostics_agg'].

 Could you please help me to solve this issue?

Imani-Saba, 2024-03-19:
[image]

Imani-Saba, 2024-03-21:
[image]

Imani-Saba, 2024-03-21:
Hi,
I got two errors when I run the Source installation command.

The following is the process I followed and the error shows up when I run these commands in my PC with Ubuntu 22.4:
According to TurtleBot 4 Desktop guidance:

1- ssh ubuntu@MAC address
2- sudo apt update
3- sudo apt install ros-humble-turtlebot4-desktop
4- cd ~/turtlebot4_ws/src
git clone https://github.com/turtlebot/turtlebot4_desktop.git -b humble
6- cd ~/turtlebot4_ws
rosdep install --from-path src -yi
7- source /opt/ros/humble/setup.bash
colcon build --symlink-install

8- source /opt/ros/humble/setup.bash
colcon build --symlink-install

Starting >>> turtlebot4_description
Starting >>> turtlebot4_msgs
Starting >>> turtlebot4_ignition_gui_plugins
Starting >>> turtlebot4_navigation
Finished <<< turtlebot4_description [2.22s]
Starting >>> turtlebot4_viz
Finished <<< turtlebot4_navigation [5.41s]
Finished <<< turtlebot4_msgs [6.87s]
Starting >>> turtlebot4_ignition_toolbox
Starting >>> turtlebot4_node
Finished <<< turtlebot4_node [2.52s]
Finished <<< turtlebot4_viz [7.75s]
Starting >>> turtlebot4_desktop
Finished <<< turtlebot4_desktop [5.26s]
[Processing: turtlebot4_ignition_gui_plugins, turtlebot4_ignition_toolbox]
[Processing: turtlebot4_ignition_gui_plugins, turtlebot4_ignition_toolbox]
--- stderr: turtlebot4_ignition_gui_plugins
In file included from /usr/include/ignition/plugin1/gz/plugin/Register.hh:22,
from /usr/include/ignition/plugin1/ignition/plugin/Register.hh:18,
from /home/ubuntu/turtlebot4_ws/src/turtlebot4_simulator/turtlebot4_ignition_gui_plugins/Turtlebot4Hmi/Turtlebot4Hmi.cc:26:
/usr/include/ignition/plugin1/gz/plugin/detail/Register.hh:28:10: fatal error: gz/utilities/SuppressWarning.hh: No such file or directory
28 | #include <gz/utilities/SuppressWarning.hh>
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [Turtlebot4Hmi/CMakeFiles/Turtlebot4Hmi.dir/build.make:99: Turtlebot4Hmi/CMakeFiles/Turtlebot4Hmi.dir/Turtlebot4Hmi.cc.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:153: Turtlebot4Hmi/CMakeFiles/Turtlebot4Hmi.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
Failed <<< turtlebot4_ignition_gui_plugins [1min 17s, exited with code 2]
Aborted <<< turtlebot4_ignition_toolbox [1min 46s]

Summary: 6 packages finished [1min 54s]
1 package failed: turtlebot4_ignition_gui_plugins
1 package aborted: turtlebot4_ignition_toolbox
1 package had stderr output: turtlebot4_ignition_gui_plugins
2 packages not processed

9- ros2 launch turtlebot4_viz view_robot.launch.py
ubuntu@ubuntu:~/turtlebot4_ws$ ros2 launch turtlebot4_viz view_robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-03-19-19-53-50-667918-ubuntu-3682
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [3683]
[rviz2-1] qt.qpa.xcb: could not connect to display
[rviz2-1] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-1] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-1]
[rviz2-1] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-1]
[ERROR] [rviz2-1]: process has died [pid 3683, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /opt/ros/humble/share/turtlebot4_viz/rviz/robot.rviz --ros-args -r __node:=rviz2 -r __ns:=/ --params-file /tmp/launch_params_xq1nqao0 -r /tf:=tf -r /tf_static:=tf_static'].

ubuntu@ubuntu:~/turtlebot4_ws$ ros2 launch turtlebot4_viz view_diagnostics.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-03-19-19-59-07-838503-ubuntu-3697
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rqt_robot_monitor-1]: process started with pid [3698]
[rqt_robot_monitor-1] could not connect to display
[rqt_robot_monitor-1] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rqt_robot_monitor-1]
[rqt_robot_monitor-1] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rqt_robot_monitor-1]
[ERROR] [rqt_robot_monitor-1]: process has died [pid 3698, exit code 1, cmd '/opt/ros/humble/lib/rqt_robot_monitor/rqt_robot_monitor --ros-args -r __ns:=/ -r /diagnostics_agg:=diagnostics_agg'].

[image]

I reinstalled ROS 2 packages following the instructions at:

https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html

However, we encountered the same error.

[image]

Could you please help with this issue at your earliest convenience?

Thank you.

Imani-Saba, 2024-03-21:
> Hi @Imani-Saba
>
> Reviewing the [Rviz2 launch instructions](https://turtlebot.github.io/turtlebot4-user-manual/software/rviz.html#rviz2) are you running this on a laptop/ PC with Ubuntu 22.04 with the [...]

Hi,
I got two errors when I run the Source installation command.

The following is the process I followed and the error shows up when I run these commands in my PC with Ubuntu 22.4:
According to TurtleBot 4 Desktop guidance:

1- ssh ubuntu@MAC address
2- sudo apt update
3- sudo apt install ros-humble-turtlebot4-desktop
4- cd ~/turtlebot4_ws/src
git clone https://github.com/turtlebot/turtlebot4_desktop.git -b humble
6- cd ~/turtlebot4_ws
rosdep install --from-path src -yi
7- source /opt/ros/humble/setup.bash
colcon build --symlink-install

8- source /opt/ros/humble/setup.bash
colcon build --symlink-install

Starting >>> turtlebot4_description
Starting >>> turtlebot4_msgs
Starting >>> turtlebot4_ignition_gui_plugins
Starting >>> turtlebot4_navigation
Finished <<< turtlebot4_description [2.22s]
Starting >>> turtlebot4_viz
Finished <<< turtlebot4_navigation [5.41s]
Finished <<< turtlebot4_msgs [6.87s]
Starting >>> turtlebot4_ignition_toolbox
Starting >>> turtlebot4_node
Finished <<< turtlebot4_node [2.52s]
Finished <<< turtlebot4_viz [7.75s]
Starting >>> turtlebot4_desktop
Finished <<< turtlebot4_desktop [5.26s]
[Processing: turtlebot4_ignition_gui_plugins, turtlebot4_ignition_toolbox]
[Processing: turtlebot4_ignition_gui_plugins, turtlebot4_ignition_toolbox]
--- stderr: turtlebot4_ignition_gui_plugins
In file included from /usr/include/ignition/plugin1/gz/plugin/Register.hh:22,
from /usr/include/ignition/plugin1/ignition/plugin/Register.hh:18,
from /home/ubuntu/turtlebot4_ws/src/turtlebot4_simulator/turtlebot4_ignition_gui_plugins/Turtlebot4Hmi/Turtlebot4Hmi.cc:26:
/usr/include/ignition/plugin1/gz/plugin/detail/Register.hh:28:10: fatal error: gz/utilities/SuppressWarning.hh: No such file or directory
28 | #include <gz/utilities/SuppressWarning.hh>
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [Turtlebot4Hmi/CMakeFiles/Turtlebot4Hmi.dir/build.make:99: Turtlebot4Hmi/CMakeFiles/Turtlebot4Hmi.dir/Turtlebot4Hmi.cc.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:153: Turtlebot4Hmi/CMakeFiles/Turtlebot4Hmi.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
Failed <<< turtlebot4_ignition_gui_plugins [1min 17s, exited with code 2]
Aborted <<< turtlebot4_ignition_toolbox [1min 46s]

Summary: 6 packages finished [1min 54s]
1 package failed: turtlebot4_ignition_gui_plugins
1 package aborted: turtlebot4_ignition_toolbox
1 package had stderr output: turtlebot4_ignition_gui_plugins
2 packages not processed

9- ros2 launch turtlebot4_viz view_robot.launch.py
ubuntu@ubuntu:~/turtlebot4_ws$ ros2 launch turtlebot4_viz view_robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-03-19-19-53-50-667918-ubuntu-3682
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [3683]
[rviz2-1] qt.qpa.xcb: could not connect to display
[rviz2-1] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-1] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-1]
[rviz2-1] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-1]
[ERROR] [rviz2-1]: process has died [pid 3683, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /opt/ros/humble/share/turtlebot4_viz/rviz/robot.rviz --ros-args -r __node:=rviz2 -r __ns:=/ --params-file /tmp/launch_params_xq1nqao0 -r /tf:=tf -r /tf_static:=tf_static'].

ubuntu@ubuntu:~/turtlebot4_ws$ ros2 launch turtlebot4_viz view_diagnostics.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-03-19-19-59-07-838503-ubuntu-3697
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rqt_robot_monitor-1]: process started with pid [3698]
[rqt_robot_monitor-1] could not connect to display
[rqt_robot_monitor-1] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rqt_robot_monitor-1]
[rqt_robot_monitor-1] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rqt_robot_monitor-1]
[ERROR] [rqt_robot_monitor-1]: process has died [pid 3698, exit code 1, cmd '/opt/ros/humble/lib/rqt_robot_monitor/rqt_robot_monitor --ros-args -r __ns:=/ -r /diagnostics_agg:=diagnostics_agg'].

[image]

I reinstalled ROS 2 packages following the instructions at:

https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html

However, we encountered the same error.

[image]

Could you please help with this issue at your earliest convenience?

Thank you.

RustyCPR, 2024-04-02:
Hi @Imani-Saba 
My colleague Saif has provided some details on ticket 12793

> The Turtlebot 4 Desktop package is meant to be installed on the computer you plan to use to interface with your Turtlebot 4. If you notice in the output, the specific package that the Turtlebot 4 struggles with is turtlebot4_ignition_gui_plugins. The turtlebot 4 does not have a GUI installed hence why it cannot process GUI tools.
>  
> With regards to RVIZ similar to the earlier issue, the Turtlebot 4 does not have a GUI installed. As such you should be launching RVIZ on your computer. 

Let us know if this helps,
Rusty

RustyCPR, 2024-04-16:
Closing due to inactivity

closed 2024-04-16 by RustyCPR