turtlebot/turtlebot4 #381: Failed to subscribe Create 3 Topics

atinfinity, 2024-03-18:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

```shell
$ dpkg -l | grep turtlebot4
ii  ros-humble-turtlebot4-base                        1.0.2-1jammy.20240219.090534            arm64        Turtlebot4 Base Node
ii  ros-humble-turtlebot4-bringup                     1.0.2-1jammy.20240308.175126            arm64        Turtlebot4 Robot Bringup
ii  ros-humble-turtlebot4-description                 1.0.4-1jammy.20240220.151406            arm64        Turtlebot4 Description package
ii  ros-humble-turtlebot4-diagnostics                 1.0.2-1jammy.20240219.092405            arm64        Turtlebot4 Diagnostics
ii  ros-humble-turtlebot4-msgs                        1.0.4-1jammy.20240217.101439            arm64        Turtlebot4 Messages
ii  ros-humble-turtlebot4-navigation                  1.0.4-1jammy.20240220.175556            arm64        Turtlebot4 Navigation
ii  ros-humble-turtlebot4-node                        1.0.4-1jammy.20240219.090001            arm64        Turtlebot4 Node
ii  ros-humble-turtlebot4-robot                       1.0.2-1jammy.20240308.175711            arm64        Turtlebot4 Robot Metapackage
ii  ros-humble-turtlebot4-setup                       1.0.3-1jammy.20240217.102329            arm64        Turtlebot4 setup scripts
ii  ros-humble-turtlebot4-tests                       1.0.2-1jammy.20240217.102249            arm64        Turtlebot4 System Tests
```

### Type of issue

Networking

### Expected behaviour

RasPi4 could subscribe Create 3 Topics.

### Actual behaviour

RasPi4 could NOT subscribe Create 3 Topics.

### Error messages

```bash
$ ros2 topic echo /battery_state
```

There is no response.

$ ros2 topic echo /odom
WARNING: topic [/odom] does not appear to be published yet
Could not determine the type for the passed topic
```

RasPi4 could NOT subscribe Create 3 Topics.
```

### To Reproduce

1. Change to Simple discovery and Cyclone DDS(RasPi4)
2. Change to Simple discovery and Cyclone DDS(Create 3)
3. Create 3 connect Wi-Fi AP via Create 3 webserver
4. Shutdown RasPi4 and Create 3
5. Power on RasPi4 and Create 3

### Other notes

I checked the following environment variables.

```shell
$ echo $ROS_DOMAIN_ID
0

$ echo $CYCLONEDDS_URI
/etc/turtlebot4/cyclonedds_rpi.xml

$ echo $RMW_IMPLEMENTATION
rmw_cyclonedds_cpp
```

I checked the configuration via Create 3 webserver.

[image]

I checked the connection from RasPi4 to Create 3. As a result, there is no problem.

```shell
ping 10.5.244.4
```

smatarCPR, 2024-03-18:
Hello @atinfinity ,
Thank you for reaching out on the official Turtlebot 4 Github page. I am very sorry to hear that you are having issues with on our products. Would it be possible to provide the following:
1. Are you able to view the topics from the Create3 on your remote computer? What is the output of `ros2 topic list` on the Raspberry Pi and the remote computer?
2. Is the Create3 connected to a 5 GHz network? The Create3 cannot handle a 5 GHz network and must be connected to a 2.4 GHz network.
3. Would it be possible to share the Create3 logs: https://iroboteducation.github.io/create3_docs/webserver/logs/
Looking forward to hearing from you soon.
Best Regards,
Saif

atinfinity, 2024-03-19:
@smatarCPR 

> Are you able to view the topics from the Create3 on your remote computer?

No.

I read <https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html#additional-cyclonedds-configuration> to set up DDS on remote computer.

```shell
wget https://raw.githubusercontent.com/turtlebot/turtlebot4_setup/galactic/conf/cyclonedds_pc.xml
sudo mv cyclonedds_pc.xml /etc/turtlebot4/
export CYCLONEDDS_URI=/etc/turtlebot4/cyclonedds_pc.xml
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
export ROS_DOMAIN_ID=0
```

As a result, remote computer could not subscribe RasPi4 & Create3 topics.

```shell
$ ros2 topic list
/parameter_events
/rosout
```

> What is the output of ros2 topic list on the Raspberry Pi and the remote computer?

<https://github.com/turtlebot/turtlebot4/issues/381#issue-2191363120> is the result of RasPi4.

```shell
$ ros2 topic list
/battery_state
/cmd_vel
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/parameter_events
/robot_description
/rosout
/scan
/stereo/depth
/tf
/tf_static
/wheel_status
```

- There is `battery_state` topic. But RasPi4 could not subscribe this topic.
- There is no `odom` topic.

> Is the Create3 connected to a 5 GHz network? The Create3 cannot handle a 5 GHz network and must be connected to a 2.4 GHz network.

Create3 and RasPi4 connected 2.4GHz network.

> Would it be possible to share the Create3 logs: https://iroboteducation.github.io/create3_docs/webserver/logs/
Looking forward to hearing from you soon.

I downloaded Create3 logs via webserver. [messages.txt](https://github.com/turtlebot/turtlebot4/files/14644018/messages.txt)

> tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -1

It seems that DDS communication failed.

atinfinity, 2024-03-19:
And I found strange behavior.

### Create 3

I set to `Cyclone DDS` via Create 3 webserver. And I restart Create 3.

[image]

### Raspi4

I set to `Fast DDS` on RasPi4.

```shell
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
export ROS_DOMAIN_ID=0
```

In this case, RasPi4 could subscribe `odom` topic.
But I don't understand this behavior.

smatarCPR, 2024-03-25:
Hello @atinfinity ,
Would it be possible to try factory resetting the Create3: https://iroboteducation.github.io/create3_docs/webserver/about/#:~:text=set%20to%20USB.-,Factory%20Reset,-A%20hyperlink%20to

Following this flash the latest firmware, based on your initial opening, you are running ROS Humble, meaning the latest firmware for the Create3 would be H.2.6 which can be found here: https://iroboteducation.github.io/create3_docs/releases/overview/

After updating, reconfigure the Create3 and see if the issue persists. Looking forward to hearing from you soon.
Best Regards,
Saif

atinfinity, 2024-03-26:
@smatarCPR I tried to flash [H.2.6](https://iroboteducation.github.io/create3_docs/releases/h_2_6) firmware and try factory resetting.
But, this issue factory persist.

atinfinity, 2024-03-26:
@smatarCPR I used Dual Band Wi-Fi Router(2.4GHz + 5GHz).
And, I select `2.4GHz` as band in `turtlebot4-setup(Wi-Fi setup)`
<https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#connect-the-raspberry-pi-to-your-network>

Is this a problem?

smatarCPR, 2024-04-03:
Hello @atinfinity ,
Does the dual band network have separate SSIDs for the different frequency bands or are they found on the same SSID? If so, then this is likely what is causing the issue seen. If possible separate the two frequency bands into two uniquely named networks. Connect the Raspberry Pi to the 5.0 GHz network and set the band as 5.0 GHz in the turtlebot4-setup(Wi-Fi setup). Then for the Create3, connect it to the 2.4 GHz network. The two networks should be joined in the backend allowing for communication across all devices.

Alternatively, you can switch to a discovery server setup, if the option is available to you, this will have the Create3 communicate to the Raspberry Pi using the USB-C connection, and the Pi will forward the data using the 5 GHz network. We typically see faster speeds and a more stable connection when this setup is used.

Let me know how these recommendations work for you. Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-04-29:
Hello @atinfinity ,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-04-29 by smatarCPR