turtlebot/turtlebot4 #383: Simulation world depot in Turtlebot4

ajjaya18, 2024-03-21:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

the last one 

### Type of issue

Other

### Expected behaviour

I hope the simulated world 'depot' goes well.

### Actual behaviour

I'm getting errors with a controller manager and the robot never loads in the simulator. It also says it's waiting for the world list.

### Error messages

```bash
@andrius-Inspiron-16-5630:~$ ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py world:=depot
[INFO] [launch]: All log files can be found below /home/andrius/.ros/log/2024-03-21-15-56-04-968416-andrius-Inspiron-16-5630-12764
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [12769]
[INFO] [parameter_bridge-2]: process started with pid [12771]
[INFO] [robot_state_publisher-3]: process started with pid [12774]
[INFO] [joint_state_publisher-4]: process started with pid [12777]
[INFO] [robot_state_publisher-5]: process started with pid [12779]
[INFO] [static_transform_publisher-6]: process started with pid [12781]
[INFO] [create-7]: process started with pid [12783]
[INFO] [create-8]: process started with pid [12785]
[INFO] [parameter_bridge-9]: process started with pid [12787]
[INFO] [parameter_bridge-10]: process started with pid [12789]
[INFO] [parameter_bridge-11]: process started with pid [12795]
[INFO] [parameter_bridge-12]: process started with pid [12797]
[INFO] [parameter_bridge-13]: process started with pid [12799]
[INFO] [parameter_bridge-14]: process started with pid [12801]
[INFO] [parameter_bridge-15]: process started with pid [12803]
[INFO] [parameter_bridge-16]: process started with pid [12805]
[INFO] [parameter_bridge-17]: process started with pid [12835]
[INFO] [parameter_bridge-18]: process started with pid [12838]
[INFO] [parameter_bridge-19]: process started with pid [12867]
[INFO] [parameter_bridge-20]: process started with pid [12887]
[INFO] [parameter_bridge-21]: process started with pid [12897]
[INFO] [parameter_bridge-22]: process started with pid [12899]
[INFO] [parameter_bridge-23]: process started with pid [12955]
[INFO] [parameter_bridge-24]: process started with pid [12958]
[INFO] [parameter_bridge-25]: process started with pid [12962]
[INFO] [parameter_bridge-26]: process started with pid [12971]
[INFO] [parameter_bridge-27]: process started with pid [13022]
[... 180 lines cut ...]
[create-7] [INFO] [1711054571.620958801] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1711054571.622085652] [ros_gz_sim]: Requesting list of world names.
[spawner-32] [INFO] [1711054573.198512850] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[spawner-32] [INFO] [1711054575.214588278] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[create-7] [INFO] [1711054576.621835335] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1711054576.622909267] [ros_gz_sim]: Requesting list of world names.
[spawner-32] [ERROR] [1711054577.231285135] [spawner_joint_state_broadcaster]: Controller manager not available
[ERROR] [spawner-32]: process has died [pid 13123, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster -c controller_manager --ros-args -r __ns:=/'].
[INFO] [spawner-46]: process started with pid [13496]
[spawner-46] [INFO] [1711054579.630099110] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[create-7] [INFO] [1711054581.622671603] [ros_gz_sim]: Requesting list of world names.
[create-8] [INFO] [1711054581.623515430] [ros_gz_sim]: Requesting list of world names.
[spawner-46] [INFO] [1711054581.647656418] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-46] [INFO] [1711054583.658717986] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
[spawner-46] [INFO] [1711054585.675005018] [spawner_diffdrive_controller]: Waiting for '/controller_manager' node to exist
```

### To Reproduce

I run this command.
ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py world:=depot

### Other notes

@hilary-luo, Hi can you help me with this?

hilary-luo, 2024-03-25:
@ajjaya18 Please go back and update your package version properly. If you are running things from source this means indicating the commit hash that you are running. One way to get this commit hash is running the following command in your local repo: `git log -1 --abbrev-commit`

hilary-luo, 2024-03-25:
@ajjaya18 If you are running the debian install then you need to provide the version of the package that you have installed. The command to get this is given in the issue template, but for simplicity I will also provide it here for you: `dpkg -s ros-$ROS_DISTRO-turtlebot4-PACKAGE_WITH_ISSUE` so in your case if you are creating a ticket about the simulator package then you would run `dpkg -s ros-$ROS_DISTRO-turtlebot4-simulator`

hilary-luo, 2024-03-25:
@ajjaya18 Thank you for updating the version. Can you please clarify what you're core issue is. You have reported one output here, a different one with different errors on your previous issue https://github.com/turtlebot/turtlebot4/issues/376#issuecomment-2007808087 and no errors launching in the latest issue https://github.com/turtlebot/turtlebot4/issues/384#issuecomment-2018690667. Which one accurately represents your current operational status so that we can focus our responses on the correct issue - or are these simultaneous issues on separate systems?

ajjaya18, 2024-03-25:
In issue 376, it was in ROS Galactic, but you advised me to switch to Humble, and in issue 384, it's a problem in Humble. Also, issue 383 is a problem in ROS Humble as well, and I also created another issue because I'm trying to do the simulation with GPU and without GPU, which is the one in the next screenshot. So, I'm doing it on two systems, both Ubuntu with ROS Humble
[image]

ajjaya18, 2024-03-25:
I want you to help me see the rplidar thing with a gpu and simulation in the depot world and help in the first simulation in ignition gazebo because I am having problems as I told you I need help with everything

ajjaya18, 2024-03-25:
andrius@andrius-HP-Pavilion-Laptop-15-eh0xxx:~/turtlebot4_ws$ ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
[INFO] [launch]: All log files can be found below /home/andrius/.ros/log/2024-03-25-14-38-06-009209-andrius-HP-Pavilion-Laptop-15-eh0xxx-19844
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [19847]
[INFO] [parameter_bridge-2]: process started with pid [19849]
[INFO] [robot_state_publisher-3]: process started with pid [19852]
[INFO] [joint_state_publisher-4]: process started with pid [19854]
[INFO] [robot_state_publisher-5]: process started with pid [19857]
[INFO] [static_transform_publisher-6]: process started with pid [19859]
[INFO] [create-7]: process started with pid [19873]
[INFO] [create-8]: process started with pid [19891]
[INFO] [parameter_bridge-9]: process started with pid [19895]
[INFO] [parameter_bridge-10]: process started with pid [19908]
[INFO] [parameter_bridge-11]: process started with pid [19927]
[INFO] [parameter_bridge-12]: process started with pid [19943]
[INFO] [parameter_bridge-13]: process started with pid [19951]
[INFO] [parameter_bridge-14]: process started with pid [19953]
[INFO] [parameter_bridge-15]: process started with pid [19955]
[INFO] [parameter_bridge-16]: process started with pid [19975]
[INFO] [parameter_bridge-17]: process started with pid [19991]
[INFO] [parameter_bridge-18]: process started with pid [20002]
[INFO] [parameter_bridge-19]: process started with pid [20016]
[INFO] [parameter_bridge-20]: process started with pid [20023]
[INFO] [parameter_bridge-21]: process started with pid [20038]
[INFO] [parameter_bridge-22]: process started with pid [20053]
[INFO] [parameter_bridge-23]: process started with pid [20062]
[INFO] [parameter_bridge-24]: process started with pid [20072]
[INFO] [parameter_bridge-25]: process started with pid [20082]
[INFO] [parameter_bridge-26]: process started with pid [20099]
[INFO] [parameter_bridge-27]: process started with pid [20118]
[INFO] [parameter_bridge-28]: process started with pid [20131]
[INFO] [parameter_bridge-29]: process started with pid [20149]
[INFO] [turtlebot4_node-30]: process started with pid [20158]
[INFO] [turtlebot4_ignition_hmi_node-31]: process started with pid [20166]
[INFO] [spawner-32]: process started with pid [20175]
[INFO] [hazards_vector_publisher-33]: process started with pid [20189]
[INFO] [ir_intensity_vector_publisher-34]: process started with pid [20247]
[INFO] [motion_control-35]: process started with pid [20256]
[INFO] [wheel_status_publisher-36]: process started with pid [20298]
[INFO] [mock_publisher-37]: process started with pid [20326]
[INFO] [robot_state-38]: process started with pid [20340]
[INFO] [kidnap_estimator_publisher-39]: process started with pid [20350]
[INFO] [ui_mgr-40]: process started with pid [20355]
[INFO] [pose_republisher_node-41]: process started with pid [20368]
[INFO] [sensors_node-42]: process started with pid [20377]
[INFO] [interface_buttons_node-43]: process started with pid [20384]
[INFO] [static_transform_publisher-44]: process started with pid [20395]
[INFO] [static_transform_publisher-45]: process started with pid [20417]
[ruby $(which ign) gazebo-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ruby $(which ign) gazebo-1] [Wrn] [FuelClient.cc:1978] The fuel.ignitionrobotics.org URL is deprecrated. Pleasse change https://fuel.ignitionrobotics.org to https://fuel.gazebosim.org
[parameter_bridge-2] [INFO] [1711395490.118055936] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[robot_state_publisher-3] [WARN] [1711395490.070198176] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-3] [INFO] [1711395490.070418598] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1711395490.070498288] [robot_state_publisher]: got segment bump_front_center
[robot_state_publisher-3] [INFO] [1711395490.070512536] [robot_state_publisher]: got segment bump_front_left
[robot_state_publisher-3] [INFO] [1711395490.070525037] [robot_state_publisher]: got segment bump_front_right
[robot_state_publisher-3] [INFO] [1711395490.070538168] [robot_state_publisher]: got segment bump_left
[robot_state_publisher-3] [INFO] [1711395490.070550530] [robot_state_publisher]: got segment bump_right
[robot_state_publisher-3] [INFO] [1711395490.070563311] [robot_state_publisher]: got segment bumper
[robot_state_publisher-3] [INFO] [1711395490.070576511] [robot_state_publisher]: got segment button_1
[robot_state_publisher-3] [INFO] [1711395490.070589013] [robot_state_publisher]: got segment button_2
[robot_state_publisher-3] [INFO] [1711395490.070600397] [robot_state_publisher]: got segment button_power
[robot_state_publisher-3] [INFO] [1711395490.070613737] [robot_state_publisher]: got segment cliff_front_left
[robot_state_publisher-3] [INFO] [1711395490.070626169] [robot_state_publisher]: got segment cliff_front_right
[robot_state_publisher-3] [INFO] [1711395490.070638950] [robot_state_publisher]: got segment cliff_side_left
[robot_state_publisher-3] [INFO] [1711395490.070651522] [robot_state_publisher]: got segment cliff_side_right
[... cut ...]
[INFO] [create-8]: sending signal 'SIGTERM' to process[create-8]
[INFO] [create-7]: sending signal 'SIGTERM' to process[create-7]
[ERROR] [create-8]: process has died [pid 19891, exit code -15, cmd '/opt/ros/humble/lib/ros_ign_gazebo/create -name standard_dock -x 0.157 -y 0.0 -z 0.0 -Y 3.1416 -topic standard_dock_description --ros-args -r __ns:=/'].
[ERROR] [create-7]: process has died [pid 19873, exit code -15, cmd '/opt/ros/humble/lib/ros_ign_gazebo/create -name turtlebot4 -x 0.0 -y 0.0 -z -0.0025 -Y 0.0 -topic robot_description --ros-args -r __ns:=/'].

hilary-luo, 2024-03-27:
@ajjaya18 Please try running it again. That particular error should be resolved now. If it still is not working please provide a new terminal output.

ajjaya18, 2024-03-27:
Hi, I have questions, I hope you can answer me.
1. Which navigation algorithm is used in TurtleBot4? I've been searching for this in the launch files and other resources, but I couldn't find much information about it. I would like to know if it uses A* or any other algorithm, but I can't seem to find this.
2. Is it posible implement RRT algorithm? If it's possible, how should I start?
3. If I have a laptop with a GPU, how should I configure it or something so that the RPLIDAR scans the environment?look the issues #384

ajjaya18, 2024-03-27:
@hilary-luo

ajjaya18, 2024-03-27:
I try to launch my own wolrd but I get mistakes, the following:

andrius@andrius-Inspiron-16-5630:~/turtlebot4_ws$ ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py world:=mundo1 x:=2.0 y:=5.0
[INFO] [launch]: All log files can be found below /home/andrius/.ros/log/2024-03-27-12-07-18-335503-andrius-Inspiron-16-5630-13419
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [13422]
[INFO] [parameter_bridge-2]: process started with pid [13425]
[INFO] [robot_state_publisher-3]: process started with pid [13427]
[INFO] [joint_state_publisher-4]: process started with pid [13430]
[INFO] [robot_state_publisher-5]: process started with pid [13432]
[INFO] [static_transform_publisher-6]: process started with pid [13434]
[INFO] [create-7]: process started with pid [13436]
[INFO] [create-8]: process started with pid [13438]
[INFO] [parameter_bridge-9]: process started with pid [13440]
[INFO] [parameter_bridge-10]: process started with pid [13442]
[INFO] [parameter_bridge-11]: process started with pid [13447]
[INFO] [parameter_bridge-12]: process started with pid [13450]
[INFO] [parameter_bridge-13]: process started with pid [13452]
[INFO] [parameter_bridge-14]: process started with pid [13454]
[INFO] [parameter_bridge-15]: process started with pid [13456]
[INFO] [parameter_bridge-16]: process started with pid [13458]
[INFO] [parameter_bridge-17]: process started with pid [13488]
[INFO] [parameter_bridge-18]: process started with pid [13500]
[INFO] [parameter_bridge-19]: process started with pid [13510]
[INFO] [parameter_bridge-20]: process started with pid [13541]
[INFO] [parameter_bridge-21]: process started with pid [13547]
[INFO] [parameter_bridge-22]: process started with pid [13553]
[INFO] [parameter_bridge-23]: process started with pid [13574]
[INFO] [parameter_bridge-24]: process started with pid [13628]
[INFO] [parameter_bridge-25]: process started with pid [13638]
[INFO] [parameter_bridge-26]: process started with pid [13648]
[INFO] [parameter_bridge-27]: process started with pid [13651]
[INFO] [parameter_bridge-28]: process started with pid [13665]
[INFO] [parameter_bridge-29]: process started with pid [13714]
[INFO] [turtlebot4_node-30]: process started with pid [13725]
[INFO] [turtlebot4_ignition_hmi_node-31]: process started with pid [13781]
[INFO] [spawner-32]: process started with pid [13784]
[INFO] [hazards_vector_publisher-33]: process started with pid [13787]
[INFO] [ir_intensity_vector_publisher-34]: process started with pid [13812]
[INFO] [motion_control-35]: process started with pid [13818]
[INFO] [wheel_status_publisher-36]: process started with pid [13820]
[INFO] [mock_publisher-37]: process started with pid [13822]
[INFO] [robot_state-38]: process started with pid [13824]
[INFO] [kidnap_estimator_publisher-39]: process started with pid [13828]
[INFO] [ui_mgr-40]: process started with pid [13861]
[INFO] [pose_republisher_node-41]: process started with pid [13886]
[INFO] [sensors_node-42]: process started with pid [13903]
[INFO] [interface_buttons_node-43]: process started with pid [13908]
[INFO] [static_transform_publisher-44]: process started with pid [13911]
[INFO] [static_transform_publisher-45]: process started with pid [13913]
[parameter_bridge-2] [INFO] [1711559240.028041801] [clock_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[robot_state_publisher-3] [WARN] [1711559239.992624242] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-3] [INFO] [1711559239.992789700] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1711559239.992848340] [robot_state_publisher]: got segment bump_front_center
[robot_state_publisher-3] [INFO] [1711559239.992854312] [robot_state_publisher]: got segment bump_front_left
[robot_state_publisher-3] [INFO] [1711559239.992858878] [robot_state_publisher]: got segment bump_front_right
[robot_state_publisher-3] [INFO] [1711559239.992863228] [robot_state_publisher]: got segment bump_left
[robot_state_publisher-3] [INFO] [1711559239.992867562] [robot_state_publisher]: got segment bump_right
[robot_state_publisher-3] [INFO] [1711559239.992871900] [robot_state_publisher]: got segment bumper
[robot_state_publisher-3] [INFO] [1711559239.992876259] [robot_state_publisher]: got segment button_1
[robot_state_publisher-3] [INFO] [1711559239.992880561] [robot_state_publisher]: got segment button_2
[robot_state_publisher-3] [INFO] [1711559239.992884766] [robot_state_publisher]: got segment button_power
[robot_state_publisher-3] [INFO] [1711559239.992889204] [robot_state_publisher]: got segment cliff_front_left
[robot_state_publisher-3] [INFO] [1711559239.992893492] [robot_state_publisher]: got segment cliff_front_right
[robot_state_publisher-3] [INFO] [1711559239.992897790] [robot_state_publisher]: got segment cliff_side_left
[robot_state_publisher-3] [INFO] [1711559239.992901934] [robot_state_publisher]: got segment cliff_side_right
[robot_state_publisher-3] [INFO] [1711559239.992905763] [robot_state_publisher]: got segment front_caster_link
[robot_state_publisher-3] [INFO] [1711559239.992909839] [robot_state_publisher]: got segment front_left_bottom_weight_block
[... cut ...] ms_1q3_w3vg'].
[ERROR] [motion_control-35]: process[motion_control-35] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [motion_control-35]: sending signal 'SIGKILL' to process[motion_control-35]
[ERROR] [motion_control-35]: process has died [pid 13818, exit code -9, cmd '/opt/ros/humble/lib/irobot_create_nodes/motion_control --ros-args -r __node:=motion_control -r __ns:=/ --params-file /tmp/launch_params_oxemix00 -r /tf:=tf -r /tf_static:=tf_static'].

hilary-luo, 2024-03-27:
> Which navigation algorithm is used in TurtleBot4? I've been searching for this in the launch files and other resources, but I couldn't find much information about it. I would like to know if it uses A* or any other algorithm, but I can't seem to find this.

The turtlebot4 uses the [nav2 packages](https://navigation.ros.org/). Within the Turtlebot4 package you should be able to see [launch files](https://github.com/turtlebot/turtlebot4/tree/humble/turtlebot4_navigation/launch) that launch nav2 functionality (slam, navigation and localization) and corresponding [config files](https://github.com/turtlebot/turtlebot4/tree/humble/turtlebot4_navigation/config) for each of them. Those config files contain all of the parameters that will be sent to the Nav2 packages and that shape the Nav2 operation. All this to say, if you want to understand how the navigation algorithms work, you will need to look at the nav2 documentation. Further questions about nav2 are best handled on the nav2 repos, [robotics stack exchange](https://robotics.stackexchange.com/) or the nav2 slack. When you run the turtlebot4 navigation launch files, you are able to pass through custom config files to modify the default operation.

> Is it posible implement RRT algorithm? If it's possible, how should I start?

See the Nav2 documentation to see if it is already implemented and if not, about how to build your own plugins.

> If I have a laptop with a GPU, how should I configure it or something so that the RPLIDAR scans the environment?look the issues https://github.com/turtlebot/turtlebot4/issues/384

As I told you before, since your issues are all about running simulations I am only going to be helping you with one issue at a time. I gave the option of issue 383 or 384 and you have continued to communicate on this ticket. If you want to switch to 384 then we can, just provide full detailed terminal output on that ticket and I can switch to responding to that ticket instead of this one.

> I try to launch my own wolrd but I get mistakes, the following

If you would like support on a custom world then please provide the world sdf file. If you are running any other additional custom code or custom nodes please provide a link to the relevant repos with the code.

ajjaya18, 2024-03-27:
<?xml version="1.0"?>
<sdf version='1.8'>
    <world name='maze'>
        <physics name='1ms' type='ignored'>
            <max_step_size>0.003</max_step_size>
            <real_time_factor>1</real_time_factor>
            <real_time_update_rate>1000</real_time_update_rate>
        </physics>
        <plugin name='ignition::gazebo::systems::Physics' filename='ignition-gazebo-physics-system' />
        <plugin name='ignition::gazebo::systems::UserCommands' filename='ignition-gazebo-user-commands-system' />
        <plugin name='ignition::gazebo::systems::SceneBroadcaster' filename='ignition-gazebo-scene-broadcaster-system' />
        <plugin name='ignition::gazebo::systems::Contact' filename='ignition-gazebo-contact-system' />
        <light name='sun' type='directional'>
            <cast_shadows>1</cast_shadows>
            <pose>0 0 10 0 -0 0</pose>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <specular>0.2 0.2 0.2 1</specular>
            <attenuation>
                <range>1000</range>
                <constant>0.90000000000000002</constant>
                <linear>0.01</linear>
                <quadratic>0.001</quadratic>
            </attenuation>
            <direction>-0.5 0.1 -0.9</direction>
            <spot>
                <inner_angle>0</inner_angle>
                <outer_angle>0</outer_angle>
                <falloff>0</falloff>
            </spot>
        </light>
        <gravity>0 0 -9.8</gravity>
        <magnetic_field>6e-06 2.3e-05 -4.2e-05</magnetic_field>
        <atmosphere type='adiabatic' />
        <scene>
            <ambient>0.4 0.4 0.4 1</ambient>
            <background>0.7 0.7 0.7 1</background>
            <shadows>1</shadows>
        </scene>
        <model name='ground_plane'>
            <static>1</static>
            <link name='link'>
                <collision name='collision'>
                    <geometry>
                        <plane>
                            <normal>0 0 1</normal>
                            <size>100 100</size>
                        </plane>
                    </geometry>
                    <surface>
                        <friction>
                            <ode />
                        </friction>
                        <bounce />
                        <contact />
                    </surface>
                </collision>
                <visual name='visual'>
                    <geometry>
                        <plane>
                            <normal>0 0 1</normal>
                            <size>100 100</size>
                        </plane>
                    </geometry>
                    <material>
                        <ambient>0.8 0.8 0.8 1</ambient>
                        <diffuse>0.8 0.8 0.8 1</diffuse>
                        <specular>0.8 0.8 0.8 1</specular>
                    </material>
                    <plugin name='__default__' filename='__default__' />
                </visual>
[... 570 lines cut ...]
                <static>true</static>
                <collision name='wall36_collision'>
                    <geometry>
                        <box>
                            <size>1 2 1</size>
                        </box>
                    </geometry>
                </collision>
                <visual name='wall36_visual'>
                    <geometry>
                        <box>
                            <size>1 2 1</size>
                        </box>
                    </geometry>
                    <material>
                        <ambient>0.1 0.1 0.1 1</ambient>
                        <diffuse>0 0.1 0.2 1</diffuse>
                        <specular>0 0.01 0.05 1</specular>
                    </material>
                </visual>
            </link>
        </model>
    </world>
</sdf> this is my sdf is similar to world maze 
@hilary-luo

ajjaya18, 2024-03-27:
This is my sdf is similar to world maze
<?xml version="1.0"?>
<sdf version='1.8'>
    <world name='maze'>
        <physics name='1ms' type='ignored'>
            <max_step_size>0.003</max_step_size>
            <real_time_factor>1</real_time_factor>
            <real_time_update_rate>1000</real_time_update_rate>
        </physics>
        <plugin name='ignition::gazebo::systems::Physics' filename='ignition-gazebo-physics-system' />
        <plugin name='ignition::gazebo::systems::UserCommands' filename='ignition-gazebo-user-commands-system' />
        <plugin name='ignition::gazebo::systems::SceneBroadcaster' filename='ignition-gazebo-scene-broadcaster-system' />
        <plugin name='ignition::gazebo::systems::Contact' filename='ignition-gazebo-contact-system' />
        <light name='sun' type='directional'>
            <cast_shadows>1</cast_shadows>
            <pose>0 0 10 0 -0 0</pose>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <specular>0.2 0.2 0.2 1</specular>
            <attenuation>
                <range>1000</range>
                <constant>0.90000000000000002</constant>
                <linear>0.01</linear>
                <quadratic>0.001</quadratic>
            </attenuation>
            <direction>-0.5 0.1 -0.9</direction>
            <spot>
                <inner_angle>0</inner_angle>
                <outer_angle>0</outer_angle>
                <falloff>0</falloff>
            </spot>
        </light>
        <gravity>0 0 -9.8</gravity>
        <magnetic_field>6e-06 2.3e-05 -4.2e-05</magnetic_field>
        <atmosphere type='adiabatic' />
        <scene>
            <ambient>0.4 0.4 0.4 1</ambient>
            <background>0.7 0.7 0.7 1</background>
            <shadows>1</shadows>
        </scene>
        <model name='ground_plane'>
            <static>1</static>
            <link name='link'>
                <collision name='collision'>
                    <geometry>
                        <plane>
                            <normal>0 0 1</normal>
                            <size>100 100</size>
                        </plane>
                    </geometry>
                    <surface>
                        <friction>
                            <ode />
                        </friction>
                        <bounce />
                        <contact />
                    </surface>
                </collision>
                <visual name='visual'>
                    <geometry>
                        <plane>
                            <normal>0 0 1</normal>
                            <size>100 100</size>
                        </plane>
                    </geometry>
                    <material>
                        <ambient>0.8 0.8 0.8 1</ambient>
                        <diffuse>0.8 0.8 0.8 1</diffuse>
                        <specular>0.8 0.8 0.8 1</specular>
                    </material>
                    <plugin name='__default__' filename='__default__' />
[... 570 lines cut ...]
                <pose>9.5 8 0 0 0 0</pose>
                <static>true</static>
                <collision name='wall36_collision'>
                    <geometry>
                        <box>
                            <size>1 2 1</size>
                        </box>
                    </geometry>
                </collision>
                <visual name='wall36_visual'>
                    <geometry>
                        <box>
                            <size>1 2 1</size>
                        </box>
                    </geometry>
                    <material>
                        <ambient>0.1 0.1 0.1 1</ambient>
                        <diffuse>0 0.1 0.2 1</diffuse>
                        <specular>0 0.01 0.05 1</specular>
                    </material>
                </visual>
            </link>
        </model>
    </world>
</sdf>

ajjaya18, 2024-03-27:
```bash
<?xml version="1.0"?>
<sdf version='1.8'>
    <world name='maze'>
        <physics name='1ms' type='ignored'>
            <max_step_size>0.003</max_step_size>
            <real_time_factor>1</real_time_factor>
            <real_time_update_rate>1000</real_time_update_rate>
        </physics>
        <plugin name='ignition::gazebo::systems::Physics' filename='ignition-gazebo-physics-system' />
        <plugin name='ignition::gazebo::systems::UserCommands' filename='ignition-gazebo-user-commands-system' />
        <plugin name='ignition::gazebo::systems::SceneBroadcaster' filename='ignition-gazebo-scene-broadcaster-system' />
        <plugin name='ignition::gazebo::systems::Contact' filename='ignition-gazebo-contact-system' />
        <light name='sun' type='directional'>
            <cast_shadows>1</cast_shadows>
            <pose>0 0 10 0 -0 0</pose>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <specular>0.2 0.2 0.2 1</specular>
            <attenuation>
                <range>1000</range>
                <constant>0.90000000000000002</constant>
                <linear>0.01</linear>
                <quadratic>0.001</quadratic>
            </attenuation>
            <direction>-0.5 0.1 -0.9</direction>
            <spot>
                <inner_angle>0</inner_angle>
                <outer_angle>0</outer_angle>
                <falloff>0</falloff>
            </spot>
        </light>
        <gravity>0 0 -9.8</gravity>
        <magnetic_field>6e-06 2.3e-05 -4.2e-05</magnetic_field>
        <atmosphere type='adiabatic' />
        <scene>
            <ambient>0.4 0.4 0.4 1</ambient>
            <background>0.7 0.7 0.7 1</background>
            <shadows>1</shadows>
        </scene>
        <model name='ground_plane'>
            <static>1</static>
            <link name='link'>
                <collision name='collision'>
                    <geometry>
                        <plane>
                            <normal>0 0 1</normal>
                            <size>100 100</size>
                        </plane>
                    </geometry>
                    <surface>
                        <friction>
                            <ode />
                        </friction>
                        <bounce />
                        <contact />
                    </surface>
                </collision>
                <visual name='visual'>
                    <geometry>
                        <plane>
                            <normal>0 0 1</normal>
                            <size>100 100</size>
                        </plane>
                    </geometry>
                    <material>
                        <ambient>0.8 0.8 0.8 1</ambient>
                        <diffuse>0.8 0.8 0.8 1</diffuse>
                        <specular>0.8 0.8 0.8 1</specular>
                    </material>
                    <plugin name='__default__' filename='__default__' />
[... 571 lines cut ...]
                <static>true</static>
                <collision name='wall36_collision'>
                    <geometry>
                        <box>
                            <size>1 2 1</size>
                        </box>
                    </geometry>
                </collision>
                <visual name='wall36_visual'>
                    <geometry>
                        <box>
                            <size>1 2 1</size>
                        </box>
                    </geometry>
                    <material>
                        <ambient>0.1 0.1 0.1 1</ambient>
                        <diffuse>0 0.1 0.2 1</diffuse>
                        <specular>0 0.01 0.05 1</specular>
                    </material>
                </visual>
            </link>
        </model>
    </world>
</sdf>
``

ajjaya18, 2024-03-27:
this is my sdf, is similar to world maze
@hilary-luo

ajjaya18, 2024-03-27:
this one is the other one 
```bash
<sdf version='1.9'>
  <world name='actors'>
    <plugin name='gz::sim::systems::Physics' filename='ignition-gazebo-physics-system'/>
    <plugin name='gz::sim::systems::Sensors' filename='ignition-gazebo-sensors-system'>
      <render_engine>ogre2</render_engine>
    </plugin>
    <plugin name='gz::sim::systems::UserCommands' filename='ignition-gazebo-user-commands-system'/>
    <plugin name='gz::sim::systems::SceneBroadcaster' filename='ignition-gazebo-scene-broadcaster-system'/>
    <actor name='demo_actor'>
      <pose>0 0 0 0 0 0</pose>
      <skin>
        <filename>https://fuel.gazebosim.org/1.0/mingfei/models/actor/1/files/meshes/walk.dae</filename>
        <scale>1</scale>
      </skin>
      <animation name='gesture'>
        <filename>https://fuel.gazebosim.org/1.0/mingfei/models/actor/1/files/meshes/gesture.bvh</filename>
        <scale>0.055</scale>
        <interpolate_x>true</interpolate_x>
      </animation>
      <animation name='walk'>
        <filename>https://fuel.gazebosim.org/1.0/mingfei/models/actor/1/files/meshes/walk.dae</filename>
        <scale>1</scale>
        <interpolate_x>true</interpolate_x>
      </animation>
      <animation name='moonwalk'>
        <filename>https://fuel.gazebosim.org/1.0/mingfei/models/actor/1/files/meshes/moonwalk.dae</filename>
        <scale>1</scale>
        <interpolate_x>true</interpolate_x>
      </animation>
      <animation name='run'>
        <filename>https://fuel.gazebosim.org/1.0/mingfei/models/actor/1/files/meshes/run.dae</filename>
        <scale>1</scale>
        <interpolate_x>true</interpolate_x>
      </animation>
      <animation name='stand'>
        <filename>https://fuel.gazebosim.org/1.0/mingfei/models/actor/1/files/meshes/stand.dae</filename>
        <scale>1</scale>
        <interpolate_x>true</interpolate_x>
      </animation>
      <animation name='sit_down'>
        <filename>https://fuel.gazebosim.org/1.0/mingfei/models/actor/1/files/meshes/sit_down.dae</filename>
        <scale>1</scale>
        <interpolate_x>true</interpolate_x>
      </animation>
      <animation name='sitting'>
        <filename>https://fuel.gazebosim.org/1.0/mingfei/models/actor/1/files/meshes/sitting.dae</filename>
        <scale>1</scale>
        <interpolate_x>true</interpolate_x>
      </animation>
      <animation name='stand_up'>
        <filename>https://fuel.gazebosim.org/1.0/mingfei/models/actor/1/files/meshes/stand_up.dae</filename>
        <scale>1</scale>
        <interpolate_x>true</interpolate_x>
      </animation>
      <script>
        <loop>true</loop>
        <delay_start>1</delay_start>
        <auto_start>true</auto_start>
        <trajectory id='0' type='stand'>
          <waypoint>
            <time>0</time>
            <pose>0 0 1.0 0 0 0</pose>
          </waypoint>
          <waypoint>
            <time>3</time>
            <pose>0 0 1.0 0 0 0</pose>
          </waypoint>
        </trajectory>
[... 524 lines cut ...]
    </model>
    <light name='sun' type='directional'>
      <pose>0 0 10 0 -0 0</pose>
      <cast_shadows>true</cast_shadows>
      <intensity>1</intensity>
      <direction>-0.5 0.1 -0.9</direction>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>1000</range>
        <linear>0.01</linear>
        <constant>0.90000000000000002</constant>
        <quadratic>0.001</quadratic>
      </attenuation>
      <spot>
        <inner_angle>0</inner_angle>
        <outer_angle>0</outer_angle>
        <falloff>0</falloff>
      </spot>
    </light>
  </world>
</sdf>
``` 
when I try to launch my robot with own world I have the mistakes
@hilary-luo

smatarCPR, 2024-04-09:
Hello @ajjaya18 ,
This issue is being closed per your comments on #389. If you believe this to be a mistake, feel free to reopen this thread by replying to this message.
Best Regards,
Saif

closed 2024-04-09 by smatarCPR