turtlebot/turtlebot4 #384: Rplidar is not working

ajjaya18, 2024-03-21:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

the last one 

### Type of issue

LIDAR

### Expected behaviour

I hope that the robot detects the objects around it.

### Actual behaviour

The robot doesn't detect anything around it.

### To Reproduce

I run this command  when I am generating a map. 
1. ros2 launch turtlebot4_navigation slam.launch.py
2. in other terminal is ros2 launch turtlebot4_viz view_robot.launch.py
3. in other terminal is ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
[image]

ajjaya18, 2024-03-21:
[image]

ajjaya18, 2024-03-25:
@hilary-luo

smatarCPR, 2024-04-09:
Hello @ajjaya18 ,
This issue is being closed per your comments on #389. If you believe this to be a mistake, feel free to reopen this thread by replying to this message.
Best Regards,
Saif

closed 2024-04-09 by smatarCPR