turtlebot/turtlebot4 #387: turtlebot_navigation localization.launch.py not working

Mikef220, 2024-03-25:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

latest version

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

After entering the command
ros2 launch turtlebot4_navigation localization.launch.py map:=fullmap4.yaml
The node should run, and the robot should localize it's position on the map.

### Actual behaviour

The program gets stuck on this line and does not proceed

[lifecycle_manager-3] [INFO] [<TimeStamp>] [life_cycle_manager_localization]: Creating bond timer...

### To Reproduce

Run the command while connected to the turtlebot

ros2 launch turtlebot4_navigation localization.launch.py map:=fullmap4.yaml

smatarCPR, 2024-04-03:
Hello @Mikef220 ,
Thank you for reaching out on the Turtlebot 4 github page. I am very sorry to hear that you are having issues with our product. Out of curiosity are you running the `ros2 launch turtlebot4_navigation localization.launch.py map:=fullmap4.yaml` command when SSH'ed into the turtlebot? It is recommended that this command be run on the remote computer interfacing with the Turtlebot 4 and not when SSH'ed. 

How large is the fullmap4.yaml file?

Is the option to switch to a Discovery server setup available to you? If so, please do so and see if the program encounters the same hangup regarding the bond timer.

Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-05-06:
Hello,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-05-06 by smatarCPR

albertuswh, 2024-06-24:
Hi im facing the same problem, have you resolved this issue?

@smatarCPR I run the launch file in remote PC than the yaml size is 124 bytes
and im using simple discovery