turtlebot/turtlebot4 #388: Error when trying to simulate the turtlebot4 in Gazebo

francescodonofrio, 2024-03-29:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

1.0.1-1jammy.20240321.134526

### Type of issue

Other

### Expected behaviour

It is expected that the turtlebot4 simulation in Gazebo is started with the empty world.

### Actual behaviour

An error occurs when parsing the robot's urdf.

### Error messages

```bash
[ruby $(which ign) gazebo-1] [INFO] [1711708564.116039117] [gz_ros2_control]: connected to service!! robot_state_publisher asking for robot_description
[ruby $(which ign) gazebo-1] [INFO] [1711708564.116853038] [gz_ros2_control]: Received URDF from param server
[ruby $(which ign) gazebo-1] [INFO] [1711708564.121486048] [gz_ros2_control]: The position_proportional_gain has been set to: 0.1
[ruby $(which ign) gazebo-1] [INFO] [1711708564.121898665] [gz_ros2_control]: Loading joint: left_wheel_joint
[ruby $(which ign) gazebo-1] [INFO] [1711708564.121914659] [gz_ros2_control]: 	State:
[ruby $(which ign) gazebo-1] [INFO] [1711708564.121927020] [gz_ros2_control]: 		 velocity
[ruby $(which ign) gazebo-1] [INFO] [1711708564.121940011] [gz_ros2_control]: 		 position
[ruby $(which ign) gazebo-1] [INFO] [1711708564.121950836] [gz_ros2_control]: 	Command:
[ruby $(which ign) gazebo-1] [INFO] [1711708564.121961941] [gz_ros2_control]: 		 velocity
[ruby $(which ign) gazebo-1] [INFO] [1711708564.122097361] [gz_ros2_control]: Loading sensor: imu
[ruby $(which ign) gazebo-1] [INFO] [1711708564.122111120] [gz_ros2_control]: 	State:
[ruby $(which ign) gazebo-1] [INFO] [1711708564.122146250] [resource_manager]: Initialize hardware 'left_wheel_controller' 
[ruby $(which ign) gazebo-1] [WARN] [1711708564.122167551] [gz_ros2_control]: On init...
[ruby $(which ign) gazebo-1] [INFO] [1711708564.122229500] [resource_manager]: Successful initialization of hardware 'left_wheel_controller'
[ruby $(which ign) gazebo-1] [INFO] [1711708564.122270496] [resource_manager]: 'configure' hardware 'left_wheel_controller' 
[ruby $(which ign) gazebo-1] [INFO] [1711708564.122276712] [gz_ros2_control]: System Successfully configured!
[ruby $(which ign) gazebo-1] [INFO] [1711708564.122288166] [resource_manager]: Successful 'configure' of hardware 'left_wheel_controller'
[ruby $(which ign) gazebo-1] [INFO] [1711708564.122294800] [resource_manager]: 'activate' hardware 'left_wheel_controller' 
[ruby $(which ign) gazebo-1] [INFO] [1711708564.122301016] [resource_manager]: Successful 'activate' of hardware 'left_wheel_controller'
[ruby $(which ign) gazebo-1] terminate called after throwing an instance of 'rclcpp::exceptions::ParameterAlreadyDeclaredException'
[ruby $(which ign) gazebo-1]   what():  parameter 'position_proportional_gain' has already been declared
[ruby $(which ign) gazebo-1] Stack trace (most recent call last):
[ruby $(which ign) gazebo-1] #31   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f30634c96, in 
[ruby $(which ign) gazebo-1] #30   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f30631fc5, in 
[ruby $(which ign) gazebo-1] #29   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f3062fc34, in 
[ruby $(which ign) gazebo-1] #28   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7a4f3079144b, in 
[ruby $(which ign) gazebo-1] #27   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f305fd088, in rb_nogvl
[ruby $(which ign) gazebo-1] #26   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7a4f30790d6b, in 
[ruby $(which ign) gazebo-1] #25   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7a4f30782492, in 
[ruby $(which ign) gazebo-1] #24   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7a4f30785e2d, in 
[ruby $(which ign) gazebo-1] #23   Object "/usr/lib/x86_64-linux-gnu/libignition-gazebo6-ign.so.6.16.0", at 0x7a4f2b9aee20, in runServer
[... 66 lines cut ...]
[ruby $(which ign) gazebo-1] #16   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7a4f30029d8f, in 
[ruby $(which ign) gazebo-1] #15   Object "ign gazebo gui", at 0x55e66f2f117e, in 
[ruby $(which ign) gazebo-1] #14   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f304a8e19, in ruby_run_node
[ruby $(which ign) gazebo-1] #13   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f304a5317, in 
[ruby $(which ign) gazebo-1] #12   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f3063a30c, in rb_vm_exec
[ruby $(which ign) gazebo-1] #11   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f30634c96, in 
[ruby $(which ign) gazebo-1] #10   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f30631fc5, in 
[ruby $(which ign) gazebo-1] #9    Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f3062fc34, in 
[ruby $(which ign) gazebo-1] #8    Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f3057ba27, in 
[ruby $(which ign) gazebo-1] #7    Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7a4f304a8db5, in ruby_stop
[ruby $(which ign) gazebo-1] #6    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7a4f3004560f, in exit
[ruby $(which ign) gazebo-1] #5    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7a4f30045494, in 
[ruby $(which ign) gazebo-1] #4    Object "/lib/x86_64-linux-gnu/libQt5Qml.so.5", at 0x7a4f28e9c64c, in 
[ruby $(which ign) gazebo-1] #3    Object "/lib/x86_64-linux-gnu/libQt5Qml.so.5", at 0x7a4f28eaa611, in 
[ruby $(which ign) gazebo-1] #2    Object "/lib/x86_64-linux-gnu/libQt5Qml.so.5", at 0x7a4f28ec544c, in QQmlPropertyCache::~QQmlPropertyCache()
[ruby $(which ign) gazebo-1] #1    Object "/lib/x86_64-linux-gnu/libQt5Qml.so.5", at 0x7a4f28ec5108, in QQmlPropertyCache::~QQmlPropertyCache()
[ruby $(which ign) gazebo-1] #0    Object "/lib/x86_64-linux-gnu/libQt5Qml.so.5", at 0x7a4f28e60ff0, in 
[ruby $(which ign) gazebo-1] Segmentation fault (Address not mapped to object [0x7a4ed00f8f80])
[INFO] [ruby $(which ign) gazebo-1]: process has finished cleanly [pid 45930]
```

### To Reproduce

1. Install all the turtlebot4 packages as indicated in [Turtlebot4 software manual section](https://turtlebot.github.io/turtlebot4-user-manual/software/) until the simulator section (simulator section included).
2. ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py world:=empty

IbrayevRamil, 2024-04-01:
I have exactly same issue with command ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py.
Tried both from deb packages and sources

hilary-luo, 2024-04-01:
This issue has been answered on [Robotics Stack Exchange](https://robotics.stackexchange.com/questions/110301/turtlebot4-problems-when-trying-to-simulate). @francescodonofrio please close this issue with your accepted answer if your issue has been resolved by that answer.

francescodonofrio, 2024-04-02:
Thank you for your help @hilary-luo

francescodonofrio, 2024-04-02:
> This issue has been answered on [Robotics Stack Exchange](https://robotics.stackexchange.com/questions/110301/turtlebot4-problems-when-trying-to-simulate). @francescodonofrio please close this issue with your accepted answer if your issue has been resolved by that answer.

closed 2024-04-02 by francescodonofrio