turtlebot/turtlebot4 #389: Error when I'm trying navigation

ajjaya18, 2024-04-01:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Package: ros-humble-turtlebot4-simulator
Version: 1.0.1-1jammy.20240304.161058
Priority: optional
Section: misc
Maintainer: rkreinin [rkreinin@clearpathrobotics.com](mailto:rkreinin@clearpathrobotics.com)
Installed-Size: 42.0 kB
Depends: ros-humble-turtlebot4-ignition-bringup, ros-humble-turtlebot4-ignition-gui-plugins, ros-humble-turtlebot4-ignition-toolbox, ros-humble-ros-workspace
Download-Size: 5,112 B
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: TODO: Package description

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I hope that my turtlebot can navigate to its goal without any problem.

### Actual behaviour

I am having problems browsing and I don't understand why I will attach the screenshots to have a better understanding.
[image]
[image]
[image]
I run this command in the terminal 
```bash
andrius@andrius-Inspiron-16-5630:~$ ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py nav2:=true slam:=false localization:=true rviz:=true
```  

### To Reproduce

1. run 
 ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py nav2:=true slam:=false localization:=true rviz:=true

ajjaya18, 2024-04-01:
@hilary-luo

ajjaya18, 2024-04-02:
@hilary-luo

hilary-luo, 2024-04-02:
@ajjaya18 Please stop tagging me repeatedly. I see when you tag something but this is a small portion of my job. I am very busy doing software development and will respond to you when I can. Also, as I said, I will only respond to 1 ticket at a time for any given person when the topics among the tickets are similar. Issues are often related to the same core problem and it can make things more confusing when the troubleshooting is broken up.

ajjaya18, 2024-04-03:
It is ok, sorry for that, you can close the others issues the #383 and #384 because I solve the problems. although the #384 I only used my cpu and can't use my gpu.

hilary-luo, 2024-04-11:
@ajjaya18 I have some time now to look into some issues with you, I appreciate your patience. As an FYI, there are a couple of open issues with the turtlebot4 simulations at the moment so I wouldn't suggest updating your packages at the present if you have things mostly working. 

Assuming that your current primary issue is that listed in this ticket, can you please copy and paste the text from the full terminal output? and can you add a more detailed description of the steps that you have taken and where the operation deviates from what you expect?

I can't tell much so far from the screenshots that you have attached but I would suggest undocking the robot first before starting to navigate.

closed 2025-04-14 by hilary-luo