turtlebot/turtlebot4 #390: LiDAR ranges do not start from angle 0 relative to the robot's heading

myamo1, 2024-04-02:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

ros-humble-rplidar-ros

### Type of issue

LIDAR

### Expected behaviour

The first scanner range should be at angle 0 (the heading of turtlebot).

### Actual behaviour

The scanner ranges are not calibrated to start from the angle 0 relative to the x-axis of the robot (its heading) and keep changing.

### To Reproduce

ros2 launch turtlebot4_bringup rplidar.launch.py

RustyCPR, 2024-04-09:
Hi @myamo1 
Can you please share a screen shot of the concern you are seeing? 
Thanks,
Rusty

myamo1, 2024-04-09:
[data_scan_pb_tbt4.odt](https://github.com/turtlebot/turtlebot4/files/14924102/data_scan_pb_tbt4.odt)
The attached file is an example where I put a box facing the turtlebot and collect the scanner ranges. The red ranges in the file are the ones associated with the box but are shifted. The first range (i.e., scanner.ranges[0]) should reflect the box. Moreover, at every scan, the order changes; it keeps shifting.

RustyCPR, 2024-04-16:
Hi @myamo1 
There should be no expectation that the first angle would be at 0 degrees according to the robot's direction. The laser scan message contains the angle min, angle max and angle increment required to be able to properly use the range data from that message.
See details here: https://docs.ros.org/en/api/sensor_msgs/html/msg/LaserScan.html 

Regards,
Rusty

myamo1, 2024-04-16:
Thanks @RustyCPR
But which frame are the angles given relative to? It should be given relative to the frame attached to the robot. At least, that is how it works in Gazebo. I tried my code there, and it worked. But if the scans are not given in a specific frame, how can we know each scan corresponds to which angle in the global frame or the attached frame to the robot?

RustyCPR, 2024-04-17:
Hi @myamo1 
The message contains all of the information needed: 
```
header:
  stamp:
    sec: 1713366206
    nanosec: 502688652
  frame_id: rplidar_link
angle_min: -3.1241390705108643
angle_max: 3.1415927410125732
angle_increment: 0.005806980188935995
time_increment: 0.001498553203418851
scan_time: 1.6169389486312866
range_min: 0.15000000596046448
range_max: 12.0
ranges:
- 0.6639999747276306
```
The frame_id is in the header, all angles are relevant to that frame. That is the first part of the message when echo'ed. 
Reviewing the [link](https://docs.ros.org/en/api/sensor_msgs/html/msg/LaserScan.html) I sent yesterday, it also specifies:
 **_angles are measured around the positive Z axis (counterclockwise, if Z is up) with zero angle being forward along the x axis_**

myamo1, 2024-04-18:
Thanks again @RustyCPR 
Yes, that what I was thinking that the angles are giving relative the x axis but the ranges that I receive do not reflect this principle.  In the example that I sent you, the red ranges are from a box that is in x-axis of the robot. 
Here is my message:
header:
  stamp:
    sec: 1713404631
    nanosec: 734668360
  frame_id: rplidar_link
angle_min: -3.1241390705108643
angle_max: 3.1415927410125732
angle_increment: 0.005806980188935995
time_increment: 0.00011300243204459548
scan_time: 0.12192963063716888
range_min: 0.15000000596046448
range_max: 12.0
ranges:
- 2.2880001068115234
- 2.2880001068115234

The only difference that I can see from your message is the scan_time, mine is faster than yours.

myamo1, 2024-04-28:
Hi @RustyCPR,

Do you have any suggestion as to why the Lidar is giving ranges in an uncalibrated way?

RustyCPR, 2024-05-13:
HI @myamo1 
Sorry for the delayed reply. I was working with our TB4 Lead Eng to get you the below details: 

Our TB4 Engieering lead made the below diagram which hopefully will clarify your understanding of the data and how to interpret it. 
Notice that there are 2 frames shown, the base frame and the rplidar link. The x axis of the base frame is indeed towards the front of the robot but the lidar data is relative to the rplidar link which is rotated by 90 degrees as well as being offset.The lidar ranges will start from angle min (on the right of the robot) and then continue counterclockwise to the angle max. If you had a box in front of the robot, that means that it would appear approximately 1/4 of the way down the ranges list. This expectation matched the data that you shared. Further proof that this is in place and that the scan is properly calibrated is that you can view the lidar data in rviz and it is always in the correct location around the robot. I would suggest that you take some time to explore the tf package and how to translate data between frames. There is common functionality that you can use to transform this data into the baselink frame if that is how you want to use it.

[image]

myamo1, 2024-06-08:
Thanks for the clarification. Now It makes sense. 
1- Is there any sources that can explain how to exploit the tf package to make the transformation from the RPlidar link to the basel ink? 
2- Sometimes, the Lidar gives random 'inf' where it should be an obstacle. Is there a way (filter ...) to deal with this ?

myamo1, 2024-07-23:
> Thanks for the clarification. Now It makes sense. 1- Is there any sources that can explain how to exploit the tf package to make the transformation from the RPlidar link to the basel ink? 2- [...]

- Hi, is there any suggestions for the 2 problems or should I open a new issue?

RustyCPR, 2024-08-19:
Hi @myamo1 did you find an answer to your 2 questions?

RustyCPR, 2024-09-09:
Closing due to no activity.

closed 2024-09-09 by RustyCPR

myamo1, 2024-10-22:
Hello @RustyCPR,

I found the solution to transform the detected points from the Lidar frame (rplidar_link) to the robot's frame (base_link).
In the Lidar_callback, I first get the transformation:
transform = self.tf_buffer.lookup_transform(
                'base_link',  # Target frame
                'rplidar_link',  # Source frame
                rclpy.time.Time()  # Use the most recent transformation
            )
then, after filtering the values that I do not desire (within a desired range), I write the detected points in the cartesian coordinates of the rplidar_link. After that I transform the obtained points:
point_in_base = tf2_geometry_msgs.do_transform_point(point_in_lidar, transform).

I am using the tf2 package.
Thank you.