turtlebot/turtlebot4 #392: Message filter dropping messages

mbergman257, 2024-04-05:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

$ dpkg -s ros-humble-turtlebot4-navigation
Package: ros-humble-turtlebot4-navigation
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 2470
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: arm64
Version: 1.0.4-1jammy.20240220.175556
Depends: ros-humble-nav2-bringup, ros-humble-nav2-simple-commander, ros-humble-slam-toolbox, ros-humble-ros-workspace
Description: Turtlebot4 Navigation

dpkg -l | grep turtlebot4
ii  ros-humble-turtlebot4-base                        1.0.2-1jammy.20240219.090534            arm64        Turtlebot4 Base Node
ii  ros-humble-turtlebot4-bringup                     1.0.2-1jammy.20240308.175126            arm64        Turtlebot4 Robot Bringup
ii  ros-humble-turtlebot4-description                 1.0.4-1jammy.20240220.151406            arm64        Turtlebot4 Description package
ii  ros-humble-turtlebot4-diagnostics                 1.0.2-1jammy.20240219.092405            arm64        Turtlebot4 Diagnostics
ii  ros-humble-turtlebot4-msgs                        1.0.4-1jammy.20240217.101439            arm64        Turtlebot4 Messages
ii  ros-humble-turtlebot4-navigation                  1.0.4-1jammy.20240220.175556            arm64        Turtlebot4 Navigation
ii  ros-humble-turtlebot4-node                        1.0.4-1jammy.20240219.090001            arm64        Turtlebot4 Node
ii  ros-humble-turtlebot4-robot                       1.0.2-1jammy.20240308.175711            arm64        Turtlebot4 Robot Metapackage
ii  ros-humble-turtlebot4-setup                       1.0.3-1jammy.20240217.102329            arm64        Turtlebot4 setup scripts
ii  ros-humble-turtlebot4-tests                       1.0.2-1jammy.20240217.102249            arm64        Turtlebot4 System Tests

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I'm trying to simply follow the tutorial "generate a map" https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html

### Actual behaviour

Running SLAM launch file or view_robot.launch.py according to the tutorials gives message filter dropping message error. No map or robot is shown on rviz. Teleop works just fine.

### Error messages

```bash
ubuntu@pi5:~$ ros2 launch turtlebot4_navigation slam.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-04-05-03-10-49-883262-pi5-1475
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [1476]
[sync_slam_toolbox_node-1] [INFO] [1712286650.437134643] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1712286650.621515132] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1712286650.622153777] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[sync_slam_toolbox_node-1] [INFO] [1712286651.981918291] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1712286651.712 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1712286652.104942154] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1712286651.849 for reason 'discarding message because the queue is full'
[... 124 lines cut ...]
[INFO] [sync_slam_toolbox_node-1]: process has finished cleanly [pid 1476]
```

### To Reproduce

Fresh install of humble & turtlebot4 setup on RPI, updated H.2.6 firmware. 
Follow discovery server steps. https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html
Edit ntp.conf as follows & restart ntpd.
```
# Use RPi4 as preferred server
# Minpoll 2^4 = 16s
# Maxpoll 2^6 = 64s
server 192.168.186.3 prefer iburst minpoll 4 maxpoll 6
```
Both PC and RPi system clocks synchronized according to `timedatectl status` command

[create3_log_messages.txt](https://github.com/turtlebot/turtlebot4/files/14879463/messages.txt)
[logs_terminal.txt](https://github.com/turtlebot/turtlebot4/files/14879473/logs_terminal.txt)

### Other notes

My /etc/chrony/chrony.conf file is different from the default file mentioned in this comment: https://github.com/turtlebot/turtlebot4/issues/379#issuecomment-2004064383

Changing it didn't seem to fix anything though, so I reverted back to how it was.
[chrony.conf.txt](https://github.com/turtlebot/turtlebot4/files/14879527/chrony.conf.txt)

mbergman257, 2024-04-05:
I think I solved part of it.

> My /etc/chrony/chrony.conf file is different from the default file... 

I set up this pi (with incorrect chrony.conf file) using https://github.com/turtlebot/turtlebot4_setup/
in which I installed Ubuntu 22.04 then ran the new image setup script with this command: 
`wget -qO - https://raw.githubusercontent.com/turtlebot/turtlebot4_setup/humble/scripts/turtlebot4_setup.sh | bash`
This procedure seems to not have set up the /etc/chrony/chrony.conf file. It could also be missing other files, I'm not sure.

(Partial fix) I have another pi that I set up by downloading and installing the latest available humble image (1.0.0) from http://download.ros.org/downloads/turtlebot4/, then updating those packages to the actual latest humble version with `sudo apt update && sudo apt upgrade`. This second pi has the correct /etc/chrony/chrony.conf file in it.
After giving pi#2 some time to sync with the create, it successfully maps for a little while with only a few "message filter dropping messages" (still following the same tutorial). 

(New Issue)
After a few minutes it gives messages like these:

```[sync_slam_toolbox_node-1] [WARN] [1712336405.868042655] [slam_toolbox]: Queue size has grown to: 109. Recommend stopping until message is gone if online mapping.```

Then a few minutes later it gave messages like these:
```[sync_slam_toolbox_node-1] [ERROR] [1712336406.776605582] [slam_toolbox]: Transform from base_link to odom failed: Lookup would require extrapolation into the past.  Requested time 1712336376.494601 but the earliest data is at time 1712336376.833029, when looking up transform from frame [base_link] to frame [odom]```

When these are popping up, the mapping seems less reliable.
[image]

smatarCPR, 2024-04-09:
Hello @mbergman257 
Thank you for reaching out on the Turtlebot 4 Github. Glad to hear that you were able to make progress on this matter. From the two error messages you have shared it appears that there is a time sync issue. For a discovery server setup, as long as your network is connected to the internet then all you should need to do is to make sure that your laptop is also syncing time with the internet ((for default ubuntu you can check using `timedatectl status` - this also works on the raspberry pi) AND redirect the create3 drive base to sync time with the raspberry pi (instructions https://github.com/turtlebot/turtlebot4/issues/216)

If your system is not connected to the internet then you will need to modify the timesyncing on both the create3 and raspberry pi. Since you are using discovery server then still point the create3 at the raspberry pi (instructions https://github.com/turtlebot/turtlebot4/issues/216) and then point the raspberry pi at an NTP server on your PC.

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif

mbergman257, 2024-04-09:
Hi @smatarCPR,
I appreciate your response. I had followed those recommended steps already, as noted in my issue description and comment above. I'm looking for more help on the new issue I ran into after resolving the ntp sync issue, as noted in my comment. Do you have recommendations for that issue?
Thank you!

mbergman257, 2024-04-09:
Additionally, if there is any extra information I can provide to help debug, or any extra clarification I can provide, please let me know!
Thank you!

hilary-luo, 2024-04-10:
This is information that hopefully will help you understand much of what is happening (80% information, 20% directions on what you can do to help improve operation) as well as helping anyone else who might come across this ticket and not completely understand what is going on here:

1. `Message Filter dropping message: frame 'rplidar_link' at time 1712286666.008 for reason 'discarding message because the queue is full'` This is not a time sync error in itself. This is just a notification that the queue is full and seeing this message is normal while bringing up the system. This message will always show up when nothing is taking the messages out of the queue, so they are just being accumulated. Examples of when you will see this is after you launch localization before you set up your initial position, essentially any in between time before the system is fully set up / fully communicating. Seeing this message can tell you something about your system's current operating state but it in itself is not the error nor would it alone be enough for us to tell what the actual error is. With SLAM, this would mean that slam is either not up and running yet, crashed or is not able to use those messages for some reason. 

2. I am glad you were able to find the chrony issue. This shows us that the issue was that SLAM was not able to use the messages because of the time sync issue. That being said, this issue would have been that the odom information from the create3 was not time synced. So even though the data from the raspberry pi was likely properly time synced and thus the lidar data was being queued properly, SLAM was not able to run properly. We will take a look at the setup instructions to see where the gap is that this chrony file was not set up properly.

4. `[sync_slam_toolbox_node-1] [WARN] [1712336405.868042655] [slam_toolbox]: Queue size has grown to: 109. Recommend stopping until message is gone if online mapping.` This message means that slam is not able to process the messages quickly enough and the queue is growing in length. Messages are being added quicker than they are being removed. This will happen when mapping large environments, especially if you have lower computational power on your PC that is running the SLAM (or if you are attempting to run SLAM onboard). By default SLAM will run continuously irrelevant of if the robot is moving or not. This tends to be the most intuitive for absolute beginners. This is generally only a problem when you try to map larger facilities. You will need to adjust how often the map is updating to fit your application. To do so, you will need to launch slam with a custom config file. This will allow you to update the map less often for a lighter map and/or allow you to stop the mapping when you stop the robot to allow for the system to catch up. Make a copy of the [config file and update the minimum travel distance and heading](https://github.com/turtlebot/turtlebot4/blob/d6881758863264ca32cf2a965c2a0ddfb2cc3094/turtlebot4_navigation/config/slam.yaml#L34-L35) for what makes sense for your use case and then pass it when you launch slam.
`ros2 launch turtlebot4_navigation slam.launch.py params:=/full/path/to/slam.yaml`
Try different numbers and see how it impacts your map.

6. `[sync_slam_toolbox_node-1] [ERROR] [1712336406.776605582] [slam_toolbox]: Transform from base_link to odom failed: Lookup would require extrapolation into the past. Requested time 1712336376.494601 but the earliest data is at time 1712336376.833029, when looking up transform from frame [base_link] to frame [odom]` This is a result of not heeding the previous warning. SLAM is so far behind in processing the messages that by the time it processes a given message, it can no longer access the transform information so far in the past from when the lidar message was received. As a result, it will be unable to add that scan into the map.  This is an error that can definitely cause issues with your map. 

Hopefully this clears things up and gives you the information to move forward. Feel free to ask any more question.

mbergman257, 2024-04-11:
Thank you! This was very helpful.

closed 2024-04-11 by mbergman257

tharukakk1, 2024-08-06:
Hi, I'm having a similar issue on my Turtlebot4 Standard. I have Simple Discovery set up because I eventually want to connect a second Turtlebot. How would I fix the time-sync issue with Simple Discovery? I was able to correct the time on the Create3, but not on the Pi.

mbergman257, 2024-08-06:
> Hi, I'm having a similar issue on my Turtlebot4 Standard. I have Simple Discovery set up because I eventually want to connect a second Turtlebot. How would I fix the time-sync issue with Simple Discovery? I was able to correct the time on the Create3, but not on the Pi.

I believe a new version of the turtlebot packages is out now or will be out soon, and they address the ntp issue. You can see which ros packages will be available here: https://discourse.ros.org/t/preparing-for-humble-sync-and-patch-release-2024-08-06/38909
So update those when theyre available. If you have any other issues you should make a new issue thread with all the requested information.

tharukakk1, 2024-08-06:
Got it, I'll look into updating that thank you. In the meantime, I followed this article's instructions to sync the time: https://robofoundry.medium.com/how-to-sync-time-between-robot-and-host-machine-for-ros2-ecbcff8aadc4
For some reason, the time is around 45 seconds off (as observed when I run timedatectl on the Pi and desktop computer within 2 seconds of each other). I'll start a new issue because I was not able to resolve the issue of "Message Filter dropping message: frame 'rplidar_link'..." when I try to run $ros2 launch turtlebot4_viz view_robot.launch.py

elanchezhiyan-official, 2024-08-07:
for people facing this in turtlebot3, this can be sorted by,
open opt/ros/humble/share/turtlebot3_navigation2/param/{turtlebot3_model}.yaml

in the yaml file, change the robot_model_type in the line 29;

from: 
robot_model_type: "differential"

change that line to : 
robot_model_type: "nav2_amcl::DifferentialMotionModel"

after this try cartographer slam with sim time argument set to true,
ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True