turtlebot/turtlebot4 #394: Turtlebot4's Rasberry PI power supply problem

kanimedes, 2024-04-08:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Any package version.
Power supply problem

### Type of issue

Battery

### Expected behaviour

When charging via docking, power must be supplied to the PI board.
(PI board on + LED on + display on)

### Actual behaviour

Turtlebot4's Rasberry PI power supply problem

When TurtleBot4 supplies power through docking, the PI Board turns on for 3 to 5 seconds and then turns off.

At the same time, there is no power response for LED and PI Display.

for testing
It has been confirmed that it operates normally when a separately verified power source is connected to the PI Board. (USB-C)

Where should I start checking?

### To Reproduce

Start charging through the docking station.

RustyCPR, 2024-04-09:
Hi @kanimedes 
Please check the Power distribution board - it supplies power from the Create 3 to the RPi and TB4 PCB board. 
You can remove the power board by pulling it straight out, ensure both USB and power connections are securely seated. 

[image]

RustyCPR, 2024-04-16:
Hi @kanimedes - any updates?

RustyCPR, 2024-04-29:
Closing due to inactivity.

closed 2024-04-29 by RustyCPR