turtlebot/turtlebot4 #395: Simulation crashes on startup

mbergman257, 2024-04-11:
### Robot Model

Select One

### ROS distro

Select One

### Networking Configuration

Select One

### OS

Select One

### Built from source or installed?

Select One

### Package version

ii  ros-humble-turtlebot4-description                  1.0.4-1jammy.20240217.092344            amd64        Turtlebot4 Description package
ii  ros-humble-turtlebot4-desktop                      1.0.0-1jammy.20240217.105800            amd64        Turtlebot4 Desktop Metapackage
ii  ros-humble-turtlebot4-diagnostics                  1.0.2-1jammy.20240217.081500            amd64        Turtlebot4 Diagnostics
ii  ros-humble-turtlebot4-ignition-bringup             1.0.1-1jammy.20240321.133710            amd64        TurtleBot 4 Ignition Simulator bringup
ii  ros-humble-turtlebot4-ignition-gui-plugins         1.0.1-1jammy.20240217.040505            amd64        Turtlebot4 Ignition Simulator GUI Plugins
ii  ros-humble-turtlebot4-ignition-toolbox             1.0.1-1jammy.20240217.075945            amd64        Turtlebot4 Ignition Toolbox
ii  ros-humble-turtlebot4-msgs                         1.0.4-1jammy.20240217.061823            amd64        Turtlebot4 Messages
ii  ros-humble-turtlebot4-navigation                   1.0.4-1jammy.20240217.111144            amd64        Turtlebot4 Navigation
ii  ros-humble-turtlebot4-node                         1.0.4-1jammy.20240217.075843            amd64        Turtlebot4 Node
ii  ros-humble-turtlebot4-simulator                    1.0.1-1jammy.20240321.134526            amd64        TODO: Package description
ii  ros-humble-turtlebot4-viz                          1.0.0-1jammy.20240217.105404            amd64        Visualization launchers and helpers for Turtlebot4

### Type of issue

Select One

### Expected behaviour

Follow simulator steps up to "Running the simulator with default settings"
https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html

### Actual behaviour

When attempting to run the simulator, gazebo opens briefly with a black screen, error messages appear, then gazebo crashes.

### Error messages

```bash
Error messages too long for github, adding .txt file to 'other notes' section.
```

### To Reproduce

Follow simulator steps up to "Running the simulator with default settings"
https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html
Run `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py` once and download resources.
Run `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py` again and get these errors and gazebo crashing.
Tried uninstalling and reinstalling ignition and turtlebot4-simulator.
Tried rebooting PC.

### Other notes

I have my environment set up for discovery server with a real turtlebot4, however, I don't expect it to be causing issues in the simulator.

Error log:
[simulation_error_log.txt](https://github.com/turtlebot/turtlebot4/files/14950305/simulation_error_log.txt)

hilary-luo, 2024-04-11:
Functionally a duplicate of https://github.com/turtlebot/turtlebot4_simulator/issues/73 and https://github.com/clearpathrobotics/clearpath_simulator/issues/30

Addressed by https://github.com/turtlebot/turtlebot4_simulator/pull/70

mbergman257, 2024-04-16:
Thanks! The simulator works now. I also ran into and resolved issues addressed by: https://github.com/turtlebot/turtlebot4_simulator/issues/72 and https://github.com/turtlebot/turtlebot4_simulator/issues/20#issuecomment-1302831141.

closed 2024-04-16 by mbergman257