turtlebot/turtlebot4 #396: Unable to use turtlebot4 simulator

alexdo105, 2024-04-11:
### Robot Model

Select One

### ROS distro

Select One

### Networking Configuration

Select One

### OS

Select One

### Built from source or installed?

Select One

### Package version

1.0.1-1jammy.20240321.134526

### Type of issue

Select One

### Expected behaviour

Simulator running

### Actual behaviour

The simulator starts but stays with a black screen.

### Error messages

```bash
os2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
[INFO] [launch]: All log files can be found below /home/alexdo105/.ros/log/2024-04-11-16-58-09-730934-alexdo105-Yoga-6-13ALC6-27554
[INFO] [launch]: Default logging verbosity is set to INFO
ign_args is deprecated, migrate to gz_args!
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [27557]
[INFO] [parameter_bridge-2]: process started with pid [27560]
[INFO] [robot_state_publisher-3]: process started with pid [27562]
[INFO] [joint_state_publisher-4]: process started with pid [27564]
[INFO] [robot_state_publisher-5]: process started with pid [27567]
[INFO] [static_transform_publisher-6]: process started with pid [27569]
[INFO] [create-7]: process started with pid [27571]
[INFO] [create-8]: process started with pid [27573]
[INFO] [parameter_bridge-9]: process started with pid [27575]
[INFO] [parameter_bridge-10]: process started with pid [27579]
[INFO] [parameter_bridge-11]: process started with pid [27583]
[INFO] [parameter_bridge-12]: process started with pid [27585]
[INFO] [parameter_bridge-13]: process started with pid [27587]
[INFO] [parameter_bridge-14]: process started with pid [27589]
[INFO] [parameter_bridge-15]: process started with pid [27592]
[INFO] [parameter_bridge-16]: process started with pid [27619]
[INFO] [parameter_bridge-17]: process started with pid [27624]
[INFO] [parameter_bridge-18]: process started with pid [27643]
[INFO] [parameter_bridge-19]: process started with pid [27669]
[INFO] [parameter_bridge-20]: process started with pid [27671]
[INFO] [parameter_bridge-21]: process started with pid [27700]
[INFO] [parameter_bridge-22]: process started with pid [27731]
[INFO] [parameter_bridge-23]: process started with pid [27765]
[INFO] [parameter_bridge-24]: process started with pid [27767]
[INFO] [parameter_bridge-25]: process started with pid [27770]
[INFO] [parameter_bridge-26]: process started with pid [27773]
[INFO] [parameter_bridge-27]: process started with pid [27793]
[... 506 lines cut ...]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[9]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L113]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[10]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L123]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[11]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L130]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf_big]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[12]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L137]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[13]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L144]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[14]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L151]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/MovAi/models/shelf]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[19]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L192]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Chair]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[20]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L199]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Chair]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[21]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L206]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/foldable_chair]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[22]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L213]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/foldable_chair]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/world[@name="warehouse"]/include[23]/uri:/opt/ros/humble/share/turtlebot4_ignition_bringup/worlds/warehouse.sdf:L223]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Table]
[ruby $(which ign) gazebo-1] [Err] [Server.cc:139] Error Code 13: [/sdf/model[@name="rescue_randy_sitting"]/include[0]/uri:/home/alexdo105/.ignition/fuel/fuel.gazebosim.org/openrobotics/models/rescue randy sitting/8/model.sdf:L6]: Msg: Unable to find uri[https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Artifact Proximity Detector]
[ruby $(which ign) gazebo-1] [Dbg] [gz.cc:161] Subscribing to [/gazebo/starting_world].
[ruby $(which ign) gazebo-1] [Dbg] [gz.cc:163] Waiting for a world to be set from the GUI...
[ruby $(which ign) gazebo-1] [Msg] Received world [warehouse.sdf] from the GUI.
```

### To Reproduce

Launch ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
See the error.

### Other notes

I have tried to install the turtlebot4 simulation packages through the source and through apt, and none of those are working.

hilary-luo, 2024-04-12:
@alexdo105 If you are running this from source with the latest version then please indicate the commit hash that you are running and ensure that you are building and sourcing your local packages correctly. Based on your output, you are not actually running the up to date source version.

RustyCPR, 2024-09-09:
Closing due to no activity

closed 2024-09-09 by RustyCPR