turtlebot/turtlebot4 #397: Top and bottom of the robot have different ROS_DOMAIN_IDs

hamidosooli, 2024-04-18:
### Robot Model

Turtlebot4 Standard

### ROS distro

Galactic

### Networking Configuration

Simple Discovery

### OS

Ubuntu 20.04

### Built from source or installed?

Installed

### Package version

0.1.3-1focal.20220916.172555

### Type of issue

Networking

### Expected behaviour

When we set the ROS_DOMAIN_ID from the browser it should be set for both top and the bottom.

### Actual behaviour

Modified ROS_DOMAIN_ID for the top is different from the bottom

### To Reproduce

We connect a monitor and a keyboard to the Raspberry-pi board, and check the ROS_DOMAIN_ID and its different from what we set__we set it to 5 and it goes back to 0 when we restart the robot.

MikeDegany, 2024-04-19:
On the *webserver*, you switch up the Domain ID for ***Create3*** (the thing you're likely talking about as 'bottom'.) To change it for RPi you should dive into ```turtlebot4-setup``` and change ROS_DOMAIN_ID in _bash_ settings.

Mikef220, 2024-04-20:
turtlebot4-setup is for ros2 Humble, not Galactic
Where would the startup bash file be located for Galactic?

MikeDegany, 2024-04-20:
My apologies! I did not know that Galactic doesn't have ```turtlebot4-setup``` tool. By the way, you could try doing ```export ROS_DOMAIN_ID=0```, it might sort out your issue.

Mikef220, 2024-04-20:
We've tried doing `export ROS_DOMAIN_ID=5` however that then sets the domain ID to an empty string (as in `echo $ROS_DOMAIN_ID` then yields blank space). We've tried just `ROS_DOMAIN_ID=5` without the export which does set the domain ID to 5, however with both of these it then reverts back to 0 upon restarting the Turtlebot, causing all of the functions that begin on start up to run on ID 0.
Thank you for your suggestions though

RustyCPR, 2024-04-22:
Hi @hamidosooli 
We recommend you update to ROS2 Humble - it has the latest updates and features. 

- `export ROS_DOMAIN_ID=#` in ROS2 Galactic will only update the domain ID in the active terminal. 

- Review this: https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html#configuration-tools ROSconfig.sh should be used to update ROS_Domain in Galactic. 

Regards,
Rusty

closed 2024-04-29 by hamidosooli