turtlebot/turtlebot4 #401: Set persistent DNS nameservers on RPi4

aharshac, 2024-04-24:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

```bash
$ dpkg -s netplan.io
Package: netplan.io
Status: install ok installed
Priority: optional
Section: net
Installed-Size: 481
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: arm64
Multi-Arch: foreign
Version: 0.106.1-7ubuntu0.22.04.2
Replaces: nplan (<< 0.34~)
Provides: nplan
Depends: libc6 (>= 2.34), libglib2.0-0 (>= 2.70.0), libnetplan0 (= 0.106.1-7ubuntu0.22.04.2), libsystemd0 (>= 243), iproute2, python3, python3-dbus, python3-yaml, python3-netifaces, systemd (>= 248~)
Suggests: network-manager | wpasupplicant, openvswitch-switch, python3-rich, iw
Breaks: nplan (<< 0.34~)
Conflicts: netplan
Description: YAML network configuration abstraction for various backends
 netplan reads YAML network configuration files which are written
 by administrators, installers, cloud image instantiations, or other OS
 deployments. During early boot it then generates backend specific
 configuration files in /run to hand off control of devices to a particular
 networking daemon.
 .
 Currently supported backends are networkd and NetworkManager.
Homepage: https://netplan.io/
Original-Maintainer: Debian Netplan Maintainers <team+netplan@tracker.debian.org>
```

### Type of issue

Networking

### Expected behaviour

Added a nameserver to `/etc/resolv.conf`, but it disappeared after a reboot.

As suggested in that file, I then added it to `/run/systemd/resolve/stub-resolv.conf` and it did not show up in `/etc/resolv.conf` after running `sudo systemctl restart systemd-resolved`.

```sh
$ readlink -f /etc/resolv.conf 
/run/systemd/resolve/stub-resolv.conf
```

Tried adding it to `/etc/netplan/40-ethernets.yaml` and then ran `sudo netplan apply`, but it still did not show up in `/etc/resolv.conf`.                       
```yaml
network:
    ethernets:
        eth0:
            addresses:
            - 192.168.185.3/24
            dhcp4: true
            optional: true
            nameservers:
                addresses:
                - 10.42.0.25
                - 8.8.8.8
                - 8.8.4.4
```

### Actual behaviour

DNS nameservers shall be added to /etc/resolv.conf at boot

### To Reproduce

1. Add DNS nameserver to `/etc/resolv.conf`, `/run/systemd/resolve/stub-resolv.conf` or `/etc/netplan/40-ethernets.yaml`
2. Reboot, run `sudo systemctl restart systemd-resolved`, or run `sudo netplan apply` respectively
3. `cat /etc/resolv.conf`

DNS nameserver is missing in `/etc/resolv.conf`

hilary-luo, 2024-05-06:
Adding your nameservers into the netplan configuration is not supposed to update `/etc/resolv.conf`. This file should be a symlink to `/run/systemd/resolve/stub-resolv.conf` which in turn should not be edited. If you have made changes to these two files then please reset them by:
1. remove the text that you added to `/run/systemd/resolve/stub-resolv.conf`
2. ensure that `/etc/resolv.conf` is still a symlink to `/run/systemd/resolve/stub-resolv.conf` (you can use `ls -l /etc | grep resolv` on the directory to verify this)

To add DNS nameservers you will have to know which network backend the network interface is being managed by. 
1. **networkd**: This is the network backend that is managing eth0 (by default) based on the netplan configuration file that you shared. You have used the correct method / syntax for adding the nameservers in the netplan file. To ensure that they were applied properly you need only to run `resolvectl status`. This command will show the DNS Servers for each network interface. 
2. **NetworkManager**: If you want to add nameservers for the wlan0 connection, it is being managed using NetworkManager and thus you would have to add the nameservers using the nmcli tool instead of netplan. 

**Additional Note**: If you happen to have connected a device with a DHCP server (such as a router) to eth0 then you may also need to use [DHCP Overrides](https://netplan.readthedocs.io/en/latest/netplan-yaml/#dhcp-overrides) to set `use-dns` to false in the netplan configuration file so that the router does not override the nameservers. (These instructions are specific to networkd, and can be done for NetworkManager using the nmcli tool).

aharshac, 2024-05-08:
@hilary-luo I'm using the wlan0 interface and set the nameservers in `/etc/netplan/50-wifis.yaml`

```yaml
network:
    version: 2
    wifis:
        renderer: NetworkManager
        wlan0:
            access-points:
                ABC-network:
                    band: 2.4GHz
                    password: 32caaeea
            addresses:
            - 192.168.1.103/24
            dhcp4: true
            optional: true
            nameservers:
                addresses:
                - 8.8.8.8
                - 8.8.4.4

```

`sudo netplan apply`
```sh
resolvectl status
Global
       Protocols: -LLMNR -mDNS -DNSOverTLS DNSSEC=no/unsupported
resolv.conf mode: stub

Link 2 (eth0)
Current Scopes: none
     Protocols: -DefaultRoute +LLMNR -mDNS -DNSOverTLS DNSSEC=no/unsupported

Link 3 (usb0)
Current Scopes: none
     Protocols: -DefaultRoute +LLMNR -mDNS -DNSOverTLS DNSSEC=no/unsupported

Link 4 (wlan0)
Current Scopes: DNS
     Protocols: +DefaultRoute +LLMNR -mDNS -DNSOverTLS DNSSEC=no/unsupported
   DNS Servers: 8.8.8.8 8.8.4.4 192.168.1.1
```

Thanks for the help!

closed 2024-05-08 by aharshac

hilary-luo, 2024-05-09:
Given that this issue is now closed I assume you have figured it out but I would like to follow up for anyone who comes across this. The wifi connection shown is managed by the NetworkManager renderer and as such you should be using the nmcli tool to set the nameservers, not the netplan method.

aharshac, 2024-05-09:
@hilary-luo Will the DNS changes made with nmcli persist after a reboot?