turtlebot/turtlebot4 #402: /tf_static queue is full, Slam can't be done

coco-fr, 2024-04-25:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

commit 6b7331978c5a959e63284b528122f3bf691cbda8
Author: Hilary Luo <103377417+hilary-luo@users.noreply.github.com>
Date:   Wed Dec 20 10:43:47 2023 -0500

    Issue forms (#315)
    
    * Update issue templates to use form interface and changed assignees
    
    * Force issues to be created using the templates
    
    * Updating CI package versions
    
    * Added new troubleshooting form to differentiate between requests for troubleshooting help and actual bugs
    

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I am trying to start the Slam.launch.py algortihm. I want to generate a map of the Room where I am working. 

### Actual behaviour

I am trying to start the Slam.launch.py algortihm, and each time a get an error msg which states the the transformation from the Lidar_link is being discarded due to the Queue being full. 
I visualised the tf_static data with Rviz2 and Foxglove Studio and the data is 42 min old even though the Robot is sending the data. 

### Error messages

```bash
[sync_slam_toolbox_node-1] [INFO] [1714053987.454483701] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714053986.383 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714053987.654841661] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714053986.484 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1714053987.829329237] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714053986.584 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714053987.913326816] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714053986.686 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714053988.008697979] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714053986.790 for reason 'discarding message because the queue is full'
```

### To Reproduce

1. run Lidar script
2. start slam 
3. Wait

### Other notes

I am using zenoh as communication between the Turtlebot4 and my Remote Pc, as well as a second lidar (velodyne). Is there a possibility to reduce the time the tf data is saved in the Queue or to increase the queue size ? 
With kind regards.

smatarCPR, 2024-04-30:
Hello @coco-fr 
Thank you for reaching out on the Turtlebot 4 Github. At the moment we do not support the use of Zenoh for communication as it has not yet been fully supported by ROS.

In terms of the messages you are seeing, these are normal to see initially when the system is getting started and not yet fully connected.

Would it be possible to share the full terminal outputs and exact commands being used to run SLAM?

Looking forward to hearing from you soon.
Best Regards,
Saif

coco-fr, 2024-05-02:
Hello @smatarCPR 

Thank you for the Answer. 
Regarding the Terminal Output and the commands, I Use the ros2 launch turtlebot4_navigation slam.launch.py 
and i get 
ros2 launch turtlebot4_navigation slam.launch.py
[INFO] [launch]: All log files can be found below /home/piekoszadm/.ros/log/2024-05-02-11-18-58-314750-innb03005-43484
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [43496]
[sync_slam_toolbox_node-1] [INFO] [1714641538.609338262] [slam_toolbox]: Node using stack size -294967296
[sync_slam_toolbox_node-1] [INFO] [1714641538.868370902] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1714641538.868600272] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[sync_slam_toolbox_node-1] [INFO] [1714641539.080550741] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641538.942 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641539.186438922] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641539.043 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641539.286783221] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641539.144 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641539.388591732] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641539.245 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641539.487688058] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641539.346 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641539.586462815] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641539.447 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641539.688419921] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641539.548 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641539.788500895] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641539.648 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641539.890717527] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641539.749 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641539.992301034] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641539.850 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641540.095164557] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641539.952 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641540.193706030] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641540.058 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641540.290900885] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641540.153 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641540.398402363] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641540.254 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641540.492071501] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641540.355 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641540.598207733] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641540.457 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641540.697934272] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641540.556 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641540.797130236] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641540.657 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641540.894803669] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641540.758 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641540.997646459] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641540.859 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1714641541.102915342] [slam_toolbox]: Message Filter dropping message: frame 'velodyne_link' at time 1714641540.960 for reason 'discarding message because the queue is full'
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[sync_slam_toolbox_node-1] [INFO] [1714641541.197394607] [rclcpp]: signal_handler(signum=2)
[INFO] [sync_slam_toolbox_node-1]: process has finished cleanly [pid 43496]

The Lidar is always running when the robot is turned on.

coco-fr, 2024-05-02:
tI tried with the RPlidar and with the Velodyne Lidar (which has been added later on) and in both scenarios i get the same error (this one is for the velodyne_link, but there is the same with the Rplidar:link)

smatarCPR, 2024-05-08:
Hello @coco-fr ,
For the time being let's focus on getting the Navigation stack working with the RPLidar. To begin, what modifications have you done to slam.launch.py and/or config?

Currently the terminal output showing no errors, with the full queue message, seems to indicate that the the system gets stuck in the startup sequence, hence the queue of messages filling quickly and not being cleared.

Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-05-27:
Hello,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-05-27 by smatarCPR

SengulDemirkol02, 2024-09-01:
Hello @coco-fr 
 I am getting the same errors. Did you fix this error? If so, could you explain how you did it?