turtlebot/turtlebot4 #403: A lot of issue with TB4

ValentinLeglise, 2024-04-26:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

***

### Type of issue

Other

### Expected behaviour

Hello, my customer is having numerous problems with his TB4. All the details of what he wants to do and what he's tried are in the “Actual Behavior” section.

### Actual behaviour

My customer has drawn up a document summarizing all these tests and the problems he is encountering. Here it is:
[Turtlebot4 Issue Report_2024-04-04_KM.pdf](https://github.com/turtlebot/turtlebot4/files/15128454/Turtlebot4.Issue.Report_2024-04-04_KM.pdf)

To add further detail to what he did, he tried reflashing the RPi many times but it didn't solve these problems.
He also tried to reset the battery by taking it out and putting it back in after a while, but without success.

If necessary, you can also retrieve the logs from the Create3 database: 
[Log_Create_2024-04-04.pdf](https://github.com/turtlebot/turtlebot4/files/15128515/Log_Create_2024-04-04.pdf)

We looked for solutions but all the ones found on your user guide or on other git tickets didn't work.
Thanks for your help.

### To Reproduce

Go to "Actual Behavior"

RustyCPR, 2024-04-29:
Hi good afternoon @ValentinLeglise 
Can you and your customers please keep these tickets one issue per TB4? 
As you know we have a larger team looking at these and its difficult to support multiple TB4 and multiple issues in one thread.
Its not easy to read and organize information.  

**We can address the first Bluetooth issue you have listed in your PDF, please open separate tickets for the other issues:**

Issue Description: 
> We have observed that both controllers function properly when connected to other
> devices such as laptops, iPads, and phones, but we are experiencing significant issues when
> attempting to connect to the robot. The maximum connection we have been able to achieve is
> controlling the options on the display of the robot, and even that is a very rare occurrence. We have
> attempted multiple troubleshooting steps, but the issue persists.
> Resolution: Investigate the Bluetooth connectivity issue with the robots and provide a solution to
> establish a stable connection.
>  For Controller 1: 
 > [image]
> The images above show our initial attempts to connectivity issues with Turtlebot4.
> We followed the instructions provided by support:
> 1. disable hciuart
> sudo systemctl disable hciuart
> sudo reboot
> 2. restart hciuart
> sudo systemctl restart hciuart
> 3. connect gamepad
> sudo bluetoothctl
> connect MAC_ADDRESS
> [image]
> Despite these efforts, we continue to face challenges in establishing a successful connection, as it
> consistently fails. We are currently troubleshooting and exploring further solutions to address this
> issue professionally and efficiently.
>

RustyCPR, 2024-04-29:
Hi @ValentinLeglise
Can the customer please try this solution [here ](https://github.com/turtlebot/turtlebot4/issues/342#issuecomment-2007741779)
Let us know if it helps,? 

Ensure the Bluetooth controller is charged with another Mini USB cable and phone charger - I believe the supplied mini USB cable that comes with the controller can be used to pair the controller with whatever hardware it is connected to. 
Let us know, 
Rusty

ValentinLeglise, 2024-05-02:
Hello, Rusty.

Thank you for your reply. Okay for the other problems, I'll open new tickets.

For this one, my customer has already tried the instructions you sent me, but with his second controller. Here's the answer:
 
[image]

Nevertheless, I asked him to carry out these instructions with the first controller, and to give me the result.

Do you have any other instructions to provide to solve this Bluetooth problem? Best regards.

ValentinLeglise, 2024-05-13:
Hello ClearPath, 

Is it possible to have a response for this ?

Thank you.

RustyCPR, 2024-05-13:
Hi @ValentinLeglise 
The customer did not follow the[ instructions in the linked ticket:](https://github.com/turtlebot/turtlebot4/issues/342#issuecomment-2007741779)  they did not download the file. 

To run this script: 
1. Extract the below attached file **_p4ppair.zip_** and run the following set of commands: 
2. File:  [p4pair.zip](https://github.com/turtlebot/turtlebot4/files/15299273/p4pair.zip)
3. Run the following  commands: 
`sudo apt install expect`
`chmod 755 p4pair `
`./p4pair <MAC of P4 Wireless Controller>`

4. Ensure that the controller is in pairing mode when the script is run. 
Let me know how this recommendation works for you. Looking forward to hearing from you soon.

RustyCPR, 2024-09-09:
Closing due to no activity

closed 2024-09-09 by RustyCPR

christopherJet, 2025-04-08:
Hello, i try too connect my turtlebot4 controller but it do not work:

ubuntu@ubuntu:~$ sudo bluetoothctl
Agent registered
[CHG] Controller D8:3A:DD:36:83:9D Pairable: yes
[bluetooth]# connect D8:3A:DD:36:83:9D
Device D8:3A:DD:36:83:9D not available
[bluetooth]# scan on
Discovery started
[CHG] Controller D8:3A:DD:36:83:9D Discovering: yes
[NEW] Device 70:DF:F7:8E:EE:DB Fibe TV Box
[NEW] Device 24:94:CB:53:1F:37 Fibe TV Box
[CHG] Device 24:94:CB:53:1F:37 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Device 24:94:CB:53:1F:37 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device 24:94:CB:53:1F:37 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 24:94:CB:53:1F:37 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 24:94:CB:53:1F:37 UUIDs: 00000000-0000-0000-0000-000000000000
[CHG] Device 70:DF:F7:8E:EE:DB UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Device 70:DF:F7:8E:EE:DB UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device 70:DF:F7:8E:EE:DB UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 70:DF:F7:8E:EE:DB UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 70:DF:F7:8E:EE:DB UUIDs: 00000000-0000-0000-0000-000000000000
[CHG] Device 70:DF:F7:8E:EE:DB RSSI: -76
[CHG] Device 24:94:CB:53:1F:37 RSSI: -79
[CHG] Device 24:94:CB:53:1F:37 RSSI: -59
[CHG] Device 24:94:CB:53:1F:37 RSSI: -73
[bluetooth]# exit
ubuntu@ubuntu:~$ ls
ubuntu@ubuntu:~$ nano p4pair
ubuntu@ubuntu:~$ nano p4pair
ubuntu@ubuntu:~$ sudo apt install expect
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
expect is already the newest version (5.45.4-2build1).
The following package was automatically installed and is no longer required:
  libncurses5-dev
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 316 not upgraded.
ubuntu@ubuntu:~$ chmod 755 p4pair
ubuntu@ubuntu:~$ ./p4pair D8:3A:DD:36:83:9D
Device D8:3A:DD:36:83:9D
spawn bluetoothctl
Agent registered
[CHG] Controller D8:3A:DD:36:83:9D Pairable: yes
[bluetooth]# remove D8:3A:DD:36:83:9D
Device D8:3A:DD:36:83:9D not available
[bluetooth]# 

I follow your step but it not working, what can i do now?

Best regard