turtlebot/turtlebot4 #405: power_saver_mode not working

alexdo105, 2024-04-29:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Last

### Type of issue

LIDAR

### Expected behaviour

When calling the service stop_motor "ros2 service call /stop_motor std_srvs/srv/Empty {}", the lidar should stop spinning. 

### Actual behaviour

When callind the service "ros2 service call /stop_motor std_srvs/srv/Empty {}", the lidar does not stop, although it says that the service has been completed.

### To Reproduce

1. Turn on the robot.
2. Run "ros2 service call /stop_motor std_srvs/srv/Empty {}"

### Other notes

I had the turtlebot4 with galactic before, and this did not happened. I changed to humble succesfully and now I cannot turn off the LiDAR with the service. Also, I changed some things such as disabling diagnostics and changing the "auto-standby" from true to false in the rplidar.launch.py but I still dont have success.

RustyCPR, 2024-04-29:
Hi @alexdo105 
Can you try manually installing it again by following these instructions: 
https://turtlebot.github.io/turtlebot4-user-manual/software/sensors.html#installing
Then try following the Controlling manual section: https://turtlebot.github.io/turtlebot4-user-manual/software/sensors.html#controlling

Let me know,
Rusty

RustyCPR, 2024-05-22:
Hi @alexdo105 Any feedback to my comment? 
Thanks,
Rusty

alexdo105, 2024-05-23:
Hello!
Thanks so much for your comment, it worked when installing manually.
Thought I had answered.

Greetings,
Alejandro

closed 2024-09-09 by RustyCPR