turtlebot/turtlebot4 #406: PC not detecting ros2 topic list

Esme1352, 2024-04-30:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

ii  ros-humble-turtlebot4-desktop                      1.0.0-1jammy.20240327.181848            amd64        Turtlebot4 Desktop Metapackage

### Type of issue

Networking

### Expected behaviour

Hello!
I was configuring the discovery server, following the instructions in the turtlebot4 user manual (humble). The configuration part on the raspberry pi works, since all the topics appear. And I was hoping that the ros2 topic list on the PC would also happen.

### Actual behaviour

However, the PC does not receive the ros2 topic list. Only appears:
/parameter_events
/rosout

I have also verified that the ROS_DOMAIN_ID is 0 on the raspberry and PC.
To give my context further, until then I have only done the basic configuration of the robot and I had problems with the discovery server in relation to the PC. My PC and my raspberry ping, since they are connected to the same wifi network.
I created this wifi network using a usb wifi adapter that is connected to the raspberry, it is a hotspot created through the usb wifi adapter and using:
`nmcli of wifi hotspot ifname wlx347de44fbb54 ssid Turtlebot4AP password UPV12345`

### To Reproduce

Perform discovery server steps from the user manual

NiloofarZendehdel, 2024-05-02:
Hello,

I also have the same problem. I am using standard Turtlebot4, Ros2 humble, Creat3 H.2.6 and I tried both Simple Discovery and Discovery Server. I cannot get Create3 connected to WiFi in Simple Discovery and here is my log: 

May  2 03:46:21 iRobot-0369BA53009349F6985317C030A4024E daemon.err udhcpc[4408]: sending discover
May  2 03:46:21 iRobot-0369BA53009349F6985317C030A4024E daemon.err udhcpc[4408]: sending select for 192.168.0.33
May  2 03:46:21 iRobot-0369BA53009349F6985317C030A4024E daemon.err udhcpc[4408]: lease of 192.168.0.33 obtained, lease time 86400
May  2 03:46:21 iRobot-0369BA53009349F6985317C030A4024E daemon.info avahi-daemon[1223]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.0.33.
May  2 03:46:21 iRobot-0369BA53009349F6985317C030A4024E daemon.info avahi-daemon[1223]: New relevant interface wlan0.IPv4 for mDNS.
May  2 03:46:21 iRobot-0369BA53009349F6985317C030A4024E daemon.info avahi-daemon[1223]: Registering new address record for 192.168.0.33 on wlan0.IPv4.
May  2 03:46:23 iRobot-0369BA53009349F6985317C030A4024E daemon.notice network[1468]: [INFO] (src/provisioning_sm.c:542) provisioning_state_machine_run: rx event PROVISIONING_DHCP_COMPLETE in state WAITING_FOR_DHCP_COMPLETE
May  2 03:46:24 iRobot-0369BA53009349F6985317C030A4024E user.notice create-platform: [INFO] [1714621584.077892032] [ui_mgr]: Network Provision: Station is trying to connect to the AP
May  2 03:46:26 iRobot-0369BA53009349F6985317C030A4024E daemon.warn network[1468]: [WARN] (src/firr.c:535) Timed out waiting for a response
May  2 03:46:26 iRobot-0369BA53009349F6985317C030A4024E daemon.err network[1468]: [ERROR] (src/mqtt_client.c:785) firr_transfer failed
May  2 03:46:26 iRobot-0369BA53009349F6985317C030A4024E daemon.err network[1468]: [ERROR] (src/network_utils.c:224) network_send_firr_req() failed
May  2 03:46:27 iRobot-0369BA53009349F6985317C030A4024E daemon.notice network[1468]: [INFO] (src/wifi_state.c:579) sec type (WPA2-PSK )
May  2 03:46:27 iRobot-0369BA53009349F6985317C030A4024E user.notice create-platform: [INFO] [1714621587.477958328] [ui_mgr]: Network Provision: STA is connected to the AP and there is a valid IP address configured.
May  2 03:46:37 iRobot-0369BA53009349F6985317C030A4024E daemon.notice network[1468]: [INFO] (src/gateway_pinger.c:169) ready to ping using sudo arping -Iwlan0 -c1 192.168.0.1 -w 2 
May  2 03:46:56 iRobot-0369BA53009349F6985317C030A4024E user.notice create-platform: [INFO] [1714621616.387916870] [system_health]: CPU usage: max 82 [%] mean 51 [%] RAM usage: 33/59 [MB]
May  2 03:47:08 iRobot-0369BA53009349F6985317C030A4024E user.notice ntpd: ntpd: timed out waiting for 192.168.0.3, reach 0x00, next query in 32s

To get the create3 connected to network I also tried pressing two buttons and got connected to create3 access point. using 192.168.10.1 I tried to get connected to network, which I got the same error again.

In Discovery Server mode, I did as instruction and was able to have the "ip route" but I was not able to get ros2 topic list on the pc side.

your assistance is very much appreciated!

smatarCPR, 2024-05-13:
Hello @Esme1352 ,
Thank you for reaching out on the Turtlebot4 Github. I am very sorry to hear that you are having issues with one of our products. Some suggestions/questions I have regarding this are:
- If you run the `ros2 topic list` command a second time do the topics appear or does the output remain the same?
- When you run the command `ip route` on your computer do you see an output similar to 192.168.186.0/24 via <Raspberry Pi IP Address> dev <Computer Wireless Interface>
- You mentioned using a USB wifi hotspot for the network, is a different network available for you to use (such as a mobile hotspot) for the purposes of testing? If so, try reconfiguring the Turtlebot4 and your computer to use this seperate network rather than the USB hotspot. This way we can determine if the issue is due to underlying network issue.

Let me know the results of the checks. Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2024-06-24 by smatarCPR