turtlebot/turtlebot4 #407: Login failure on TB4

ValentinLeglise, 2024-05-02:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

*

### Type of issue

Networking

### Expected behaviour

Hello. This ticket is the continuation of this one (https://github.com/turtlebot/turtlebot4/issues/403).  The new problems specified are in the “Actual behavior” section.

### Actual behaviour

My customer is having trouble connecting his robot to a network. After 5-6 days of use and a first successful connection, the robot has disconnected and is now inaccessible to connection, whether from Ubuntu, Putty or cmd.

I asked them to reflash the image of the Raspberry Pi and to reset the battery by removing and reinserting it. He has done this several times and the problem persists. Here's the current state of his robot:

[image]

[image]

My customer's robot only displays 3 green LEDs out of 5. The Wifi and motor LEDs are off. My customer also says that when he runs the command “ros2 run teleop_twist_keyboard teleop_twist_keyboard”, it doesn't work, whereas with his second robot, it works perfectly.

Do you know what we can suggest ?

Best regards

### To Reproduce

*

RustyCPR, 2024-05-06:
Hi @ValentinLeglise 

Can you please ask the customer to ensure the USB connection from the Create 3 base to the RPi is securely connected? 
They may want to unplug the USB connection on the Create 3 Power distribution board and reconnect. 
Also on the other side of the connection with the RPi. 

[image]

RustyCPR, 2024-05-06:
Hi @ValentinLeglise 
Just to follow up on the above, if the USBs are connected can the customer check the below items: 

1. Can the customer access the Create 3 web server through the RPi?  [Details ](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#accessing-the-create-3-webserver )

2. Confirm the config setting on both the Create 3 and RPi - please send us screen shots.  
Create 3: https://iroboteducation.github.io/create3_docs/webserver/application/ 
Turtlebot 4: https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html#configuration-tools

Thanks,
Rusty

ValentinLeglise, 2024-05-13:
Hello,

Thank you for this information. I asked my customer and i come back as soon as i have his response.

Do you know if the Create3 factory reset might be useful? If so, how do you go about it?

Best regards, Valentin.

RustyCPR, 2024-05-13:
Hi @ValentinLeglise 
It shouldn't be required - but if the customer wants to try it - when you login to the Create3 it should be under the about page.
Details[ here](https://iroboteducation.github.io/create3_docs/webserver/about/#:~:text=set%20to%20USB.-,Factory%20Reset,-A%20hyperlink%20to)
Regards,
Rusty

RustyCPR, 2024-06-12:
Closing due to no activity / updates.

closed 2024-06-12 by RustyCPR