turtlebot/turtlebot4 #411: Can't launch SLAM, NAV2, or RVIZ2 for multiple Turtlebot4's

Robo52, 2024-05-15:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Version: 1.0.4-1jammy.20240407.042255

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I namespaced the two turtlebot4's I have according to the turtlebot4 user manual and the create3 documentation. I expect to be able to launch SLAM for both of these robots separately and then launch RVIZ2 and see both of the robots in the window.

### Actual behaviour

In reality, SLAM does not fully launch for either bot, even when I run them one at a time. The SLAM script makes it to the point where it displays (using JACOBI-PRECONDITIONER) in the terminal and then does not output anything else. When I launch RVIZ2 while the robots are in this state, I get the map error: "no map received".

### To Reproduce

1: Undock one robot with command: ros2 action send_goal /GIOVANNI/undock irobot_create_msgs/action/Undock "{}"
2: Undock second robot with command: ros2 action send_goal /HUMBOLDT/undock irobot_create_msgs/action/Undock "{}"
3 Launch slam for one robot: ros2 launch turtlebot4_navigation slam.launch.py namespace:=/GIOVANNI params:=/opt/ros/humble/share/turtlebot4_navigation/config/slam.yaml
4: Launch slam for other robot: ros2 launch turtlebot4_navigation slam.launch.py namespace:=/HUMBOLDT params:=/opt/ros/humble/share/turtlebot4_navigation/config/slam.yaml
5: Launch RVIZ2: ros2 launch turtlebot4_viz view_robot.launch.py

WallerJonathan, 2024-05-20:
I'm not too sure about the slam problem, but I will say that RVIZ2 (as well as any launch and run command in tb4) also needs namespacing. For example, you might want to run ```ros2 launch turtlebot4_viz view_robot.launch.py namespace:=/GIOVANNI```. Unfortunately, there seems to be some problems with turtlebot4 and some of the launching so it only works _sometimes_.

Also, if you are trying to run SLAM specifically, try with only one robot as that's all you really need for mapping!

smatarCPR, 2024-05-22:
Hello @Robo52 ,
Thank you for reaching out on the Turtlebot Github page. As @WallerJonathan mentioned, try launching RVIZ with the namespace specified, does SLAM appear to be functional when RVIZ is launched this way?

If not, would it be possible to provide the full terminal outputs when launching SLAM and NAV2 (particularly the ones where you mentioned it reached using JACOBI-PRECONDITIONER).

We'll take a look at the terminal output and see what we can determine. Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-05-22:
Hello @WallerJonathan ,
I had a question regarding your comment there seems to be some problems with turtlebot4 and some of the launching so it only works sometimes. Would it be possible to provide some details regarding this issue you are seeing.

Looking forward to hearing from you soon.
Best Regards,
Saif

WallerJonathan, 2024-05-22:
Hi @smatarCPR,

I have been working with 2 physical turtlebot4s for the past week or so, and I have experienced inconsistencies seemingly with the launch files. Sometimes, using the launch files works, but sometimes it doesn't.

For example, earlier today I typed the commands `ros2 launch turtlebot4_navigation localization.launch.py namespace:=/robot1 map:=maps/mymap.yaml` and `ros2 launch turtlebot4_viz view_robot.launch.py namespace:=/robot1`. This worked just fine and pulled up an rviz interface with the localization and correct map.

However, later in the day I typed in the exact same commands and it would not load the tfs. Even after rebooting my computer and the robot twice and trying again it did not work; this has happened to me before with other launches such as nav2.launch.py, and the same problem happens on my coworker's laptop.

It is not a problem with the computer as my CPU is only at about 5% load when I run rviz, so I am unsure what the problem could be.

Thanks for the help!

smatarCPR, 2024-06-12:
Hello @WallerJonathan ,
Apologies for the late reply on this. That is very unfortunate to hear. This is not an expected behaviour for the turtlebot4.

Would it be possible to perform the following for us so we can help determine what may be causing this occasional failure:
- Open I new Github ticket (different from #410 ) and provide the full system details for the two turtlebot units affected.
- If this issue is easily replicable, do so and after the failure collect the full system logs from the two turtlebots using the command: `journalctl -b > turtlebot4_log.txt`
- Do you find that the failure depends on whether both units are running or can it occur using only one unit?

Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2024-08-19 by smatarCPR