turtlebot/turtlebot4 #412: ROS2 jazzy Support

CycleMark, 2024-05-20:
### Describe the the feature you would like

Support for ROS2 jazzy - Being released in a few days

Thanks 

Mark

### Motivation and impact

Enables easy use of Raspberry Pi 5.
Up-to-date ROS distro (The existing ones are old now) 
Up-to-date OS Ubuntu 24.04

MikitaSaladukha, 2024-05-22:
Hello!
I'm really looking forward to this release of ROS2!
 Will ROS2 jazzy work with turtlebot3 or turtlebot4 or both? what versions of gazebo it will support?
 I really hope jazzy to work with turtlebot3, because I work with it a lot.
Best Regards,
Mikita Salaudukh

RustyCPR, 2024-05-22:
Hi @MikitaSaladukha  - current support for ROS2 Galactic and ROS2 Humble. With a strong future focus on ROS2 Humble.
I see Jazzy is still in development

CycleMark, 2024-05-22:
Hi @RustyCPR,

For me I was interested in TB4 looking for 24.04 and Jazzy. Which should be released tomorrow -https://docs.ros.org/en/jazzy/Releases.html

Just wanted to see when you'd be looking to support it - or possibly looking to support it. Even as a dev branch.

In the mean time I'm looking at compiling it myself. Any idea on how to install ignition-math6Config - which despite the various doc's I've found doesn't seem to want to play ball.

Thanks

Mark

RustyCPR, 2024-05-22:
Hi @CycleMark - I cannot comment on development and future release details. 
As stated the current team focus is ROS2 Humble. 

Regarding your last question: Any idea on how to install ignition-math6Config - try asking here: https://community.gazebosim.org/

Best regards,
Rusty

andreasBihlmaier, 2024-05-28:
Also wanting to run TurtleBot 4 _simulation_ with ROS Jazzy + Gazebo Harmonic on Ubuntu Noble, I started to patch some of the required packages. The following results in a _very basic_ working sim setup for me on top of `ros-jazzy-desktop-full`:

~/turtlebot4_ws/src
- create3_sim: https://github.com/andreasBihlmaier/create3_sim/tree/jazzy
- irobot_create_msgs: https://github.com/iRobotEducation/irobot_create_msgs (rolling)
- turtlebot4: https://github.com/turtlebot/turtlebot4 (humble)
- turtlebot4_desktop: https://github.com/turtlebot/turtlebot4_desktop (humble)
- turtlebot4_simulator: https://github.com/andreasBihlmaier/turtlebot4_simulator/tree/jazzy

No promises on my side to continue this and submit a PR. But the fixes in create3_sim and turtlebot4_simulator should at least be a good starting point for others.

@RustyCPR btw. It would be **great** to have working `ros-jazzy-turtlebot4-simulator` packages :)

CycleMark, 2024-05-28:
Hi @andreasBihlmaier - this looks good. How did  you manage to get  ignition-math6Config or  ignition-math7Config installed?

Aiming to get it installed on RPi with Ubuntu 24.04 LTS

`mark@mxnetPi5:~/turtlebot4_ws$ sudo apt-get install libignition-math6-dev
[sudo] password for mark: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libignition-math6-dev
mark@mxnetPi5:~/turtlebot4_ws$ sudo apt-get install libignition-math7-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libignition-math7-dev is already the newest version (7.4.0-3~noble).
0 upgraded, 0 newly installed, 0 to remove and 50 not upgraded.
mark@mxnetPi5:~/turtlebot4_ws$ `

Despite changing 6 for 7 etc in the package & makefiles still wind up with this error:

```
--- stderr: irobot_create_toolbox                                                                                                                    
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findignition-math7.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ignition-math7", but CMake did not find one.

  Could not find a package configuration file provided by "ignition-math7"
  with any of the following names:

    ignition-math7Config.cmake
    ignition-math7-config.cmake

  Add the installation prefix of "ignition-math7" to CMAKE_PREFIX_PATH or set
  "ignition-math7_DIR" to a directory containing one of the above files.  If
  "ignition-math7" provides a separate development package or SDK, be sure it
  has been installed.
```

Cheers

Mark

andreasBihlmaier, 2024-05-29:
I did not yet try on arm / Pi (but will soon-ish). On x86 I'm enjoying the new[ Gazebo "vendor" packages](https://docs.ros.org/en/jazzy/Releases/Release-Jazzy-Jalisco.html#changes-to-how-ros-2-and-gazebo-integrate), i.e. `ros-jazzy-gz-sim-vendor`.

CycleMark, 2024-05-29:
Okay cool. Will be interested to hear how you get on. In the meantime I'll continue to batter it this end and see if I can get things to play ball.

RustyCPR, 2024-09-09:
Closing due to no activity

closed 2024-09-09 by RustyCPR

patham9, 2024-10-21:
Any update on this? Still no support?
Here it claims there is a package existing:
https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html
however apt cannot find that package using Ubuntu 24.04.

avidbots-cameron, 2024-10-22:
There looks to be an issue with the build dependencies (specifically the irobot create package) that is blocking the turtlebot package from being released properly (hopefully someone is looking at it?).
At any rate, if you can build from source then it seems to work for me.

Details to build turtlebot4-simulator from source are available here: https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_common.html

I needed to also check out and build the irobot create content per the instructions here: https://github.com/iRobotEducation/create3_sim

patham9, 2024-10-24:
I see, so essentially it is not yet supported. Maybe I should wait a bit until ROS2 matures a bit, even though I got around some of the random segfaults in the meanwhile.

ahamboeck, 2024-10-26:
Is there any other robot simulation supported for ros2 jazzy?

patham9, 2024-11-01:
@ahamboeck Turtlebot3 also works, and Turtlebot 4 too, just not as in the tutorials.
It has become a standard simulation as a part of Nav2 so after installing that package you can run
`ros2 launch nav2_bringup tb4_simulation_launch.py slam:=True nav:=True headless:=False use_sim_time:=True`
and
`ros2 launch nav2_bringup tb3_simulation_launch.py slam:=True nav:=True headless:=False use_sim_time:=True`
I hope this is useful to you!

ahamboeck, 2024-11-04:
@patham9 Thank you! I didn't know about that.