turtlebot/turtlebot4 #413: turtlebot4 setup/networking

knickels, 2024-05-21:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

1.0.3-1jammy.20240118.171148 (used ros-humble-turtlebot4-setup)

### Type of issue

Networking

### Expected behaviour

ros2 topic list on user pc matches ros2 topic list on raspi

### Actual behaviour

```
ubuntu@turtlebot4:~$` ros2 topic list
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/diagnostics
/dock_status
/function_calls
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mobility_monitor/transition_event
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/static_transform/transition_event
/tf
/tf_static
/wheel_status
ubuntu@turtlebot4:~
```

but
```
$ ros2 topic list
/parameter_events
/rosout
$
```

### To Reproduce

Perform discovery server steps from the user manual

### Other notes

This is on a LattePanda, but I got same behavior with a standard PC, but that was on a different subnet.

`ip route` does show the expected route:
```
$ ip route
192.168.186.0/24 via 131.194.112.108 dev wlo1 
$
```

I tried simple discovery first, got this same result, then ran through the steps for discovery server.

Thanks for any help/guidance y'all can give!

smatarCPR, 2024-05-22:
Hello @knickels ,
Thank you for reaching out on the Turtlebot 4 Github page. Do the topics appear if you the list command a second time? 

If not, check that:
1. You are able to ping and SSH into the turtlebot from the computer
2. Double check that you have sourced your bashrc file `source ~/.bashrc`
3. Restart the ROS 2 Daemon - `ros2 daemon stop; ros2 daemon start`

Let me know how these suggestions work for you. Looking forward to hearing from you soon.
Best Regards,
Saif

knickels, 2024-05-22:
1. yes, yes
2. yes
3. Hm, the ros2 daemon wasn't running on my user PC - when I did this it told me `The daemon is not running`.  Then, the full topics list appeared after this step!

Do I need to set that to auto-start, or why isn't it running?

smatarCPR, 2024-06-12:
Hello @knickels ,
Apologies for the late reply on this. Typically when running commands such as `ros2 topic list` this will trigger the ros2 daemon to start. 

Most likely what occurred was after changing the settings to use discovery server from simple discovery, the terminal session did not restart, so it was still using the old settings hence why it was unable to view the topics.

Are you otherwise now able to view the topics?

Looking forward to hearing from you soon.
Best Regards,
Saif

knickels, 2024-06-13:
No, I went on to other stuff.  
I had it briefly working with Fast DDS with static discovery a few weeks ago, but the same setup doesn't seem to be working today.  Not sure what working state didn't make it into my setup script :(

I need to move my user pc back and forth between discovery server (for turtlebot stuff) and simple discovery (for other ros work).  I thought it was as simple as setting and clearing FASTRTPS_DEFAULT_PROFILES_FILE environment variable, which (when set) points FastRTPS to 

```
$ cat fastdds_discovery_static.xml 
<?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <participant profile_name="super_client_profile" is_default="true">
            <rtps>
                <builtin>
			<initialPeersList>
                                <metatrafficUnicastLocatorList>
                                    <locator>
                                        <udpv4>
						<address>131.xxx</address> <!-- ip of turtlebot -->
                                            <port>7400</port>
                                        </udpv4>
                                    </locator>
                                    <locator>
                                        <udpv4>
						<address>131.yyy</address> <!-- ip of user pc -->
                                            <port>7400</port>
                                        </udpv4>
                                    </locator>
                                </metatrafficUnicastLocatorList>
			<initialPeersList>
                </builtin>
            </rtps>
        </participant>
    </profiles>
</dds>
```

smatarCPR, 2024-06-25:
Hello @knickels ,
When switching from discovery server to simple discovery are resourcing your setup.bash file?

When setting up the two communication methods, two folders are created in /etc/ros/ one for simple discovery, and one for discovery server. 

In your ~/.bashrc have two lines:
One sourcing the simple discovery `source /etc/turtlebot4/setup.bash` and one sourcing discovery server `source /etc/turtlebot4_discovery/setup.bash`

Comment out the line you will not be using. When making the switch, edit your bashrc file, commenting out the version you will not use. Save, then restart the terminal you should make the full switch.

Let me know how this recommendation works for you.
Best Regards,
Saif

knickels, 2024-06-28:
Yes, I am.  Here are some logs.  ROS2 is my desktop.
First, on the TB I get as expected.
```
ubuntu@turtlebot4:~$ ros2 topic list
/battery_state
/cmd_vel
/diagnostics
/dock_status
/function_calls
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/parameter_events
/robot_description
/rosout
/tf
/tf_static
/wheel_status
ubuntu@turtlebot4:~$ env | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_LOCALHOST_ONLY=0
ROS_DISTRO=humble
ROS_DISCOVERY_SERVER=131.194.112.108:11811
ubuntu@turtlebot4:~$ 
```

Then on my desktop:
```
 knickels at ROS2 in ~  
$ source /opt/ros/humble/setup.bash 
 knickels at ROS2 in ~  
$ ros2 daemon start
The daemon has been started
 knickels at ROS2 in ~  
$ env | grep ROS
ROS_VERSION=2
SESSION_MANAGER=local/ROS2:@/tmp/.ICE-unix/2254,unix/ROS2:/tmp/.ICE-unix/2254
ROS_PYTHON_VERSION=3
ROS_LOCALHOST_ONLY=0
ROS_DISTRO=humble
 knickels at ROS2 in ~  
$ ros2 topic list
/parameter_events
/rosout
 knickels at ROS2 in ~  
$ ping turtlebot4
PING turtlebot4.trinity.local (131.194.112.108) 56(84) bytes of data.
64 bytes from turtlebot4.trinity.local (131.194.112.108): icmp_seq=1 ttl=63 time=2.07 ms
64 bytes from turtlebot4.trinity.local (131.194.112.108): icmp_seq=2 ttl=63 time=2.31 ms
^C
--- turtlebot4.trinity.local ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1002ms
rtt min/avg/max/mdev = 2.073/2.192/2.311/0.119 ms
 knickels at ROS2 in ~  
$ source /etc/turtlebot4_discovery/setup.bash 
 knickels at ROS2 in ~  
$ ros2 daemon stop
The daemon has been stopped
 knickels at ROS2 in ~  
$ ros2 daemon start
The daemon has been started
 knickels at ROS2 in ~  
$ ros2 topic list
/parameter_events
/rosout
 knickels at ROS2 in ~  
$ env | grep ROS
ROS_VERSION=2
SESSION_MANAGER=local/ROS2:@/tmp/.ICE-unix/2254,unix/ROS2:/tmp/.ICE-unix/2254
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_LOCALHOST_ONLY=0
ROS_DISTRO=humble
ROS_DISCOVERY_SERVER=131.194.112.108:11811
 knickels at ROS2 in ~  
$ 
```

knickels, 2024-06-28:
Hmm, I stumbled upon this useful example: https://docs.vulcanexus.org/en/latest/rst/tutorials/core/qos/initial_peers/initial_peers.html.

I think I'm on the right track here - by following this, I was able to get a basic talker/listener going, but `ros2 topic list` still doesn't come over.  Interestingly, I can listen to a talker demo but not list info on the /chatter topic.

```
$ ros2 run demo_nodes_py listener
[INFO] [1719608642.704603891] [listener]: I heard: [Hello World: 0]
[INFO] [1719608643.631987608] [listener]: I heard: [Hello World: 1]
[INFO] [1719608644.630042692] [listener]: I heard: [Hello World: 2]
^C knickels at ROS2 in ~  
$ ros2 topic list
/parameter_events
/rosout
 knickels at ROS2 in ~  
$ ros2 topic info /chatter
Unknown topic '/chatter'
 knickels at ROS2 in ~  
$ 
```

smatarCPR, 2024-07-04:
Hello @knickels ,
It is quite unfortunate hear that you are encountering issues. Looking at the details provided, there maybe some details missing in the setup.bash file. Would it be possible to:
1.  Recreate the discovery server setup.bash file using the command script outlined on our [online documentation](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html#user-pc). See if after this recreation you are able to get the topics on your computer when using discovery server.
2. If the above fails to resolve the issue, please attach copies of the two setup.bash files.

On another note, the topic list shown on the Turtlebot 4 seems a bit short, it is missing some topics from the Create3, was this list generated quickly after a restart?

Looking forward to hearing from you soon.
Best Regards,
Saif

knickels, 2024-07-18:
Yes, I'm noticing that too.  I don't think that it's communicating with
the create3 at all?  I can't send /cmd_vel messages (well, they don't do
anything), or e.g. 'ros2 echo topic /wheel_sensors'.  I seem to be
going backwards - I'm going to try starting over with a fresh image and at
least getting simple discovery working again.
--kn

knickels, 2024-07-18:
OK, so I'm back to trying to get Simple Discovery working, as it should
out-of-the-box.  At the moment, I'm not even able to drive it around with
the bluetooth controller :(

I re-flashed the pi, and seem to be getting about the same thing -
the create topics aren't showing up on the pi.  Our wifi is pretty
locked-down, how can I check to see if the create3 is able to access the
2.4GHz network?  Is getting an ip on http://131.194.112.37:8080/wifi enough?

--kn

smatarCPR, 2024-07-24:
Hello @knickels ,
From your description of the matter currently, I suspect that the Create3 is not connected to the wi-fi network. What type of network is this? The Create3 currently cannot support/connect to 5.0 GHz networks and must be connected to a 2.4 GHz network.

Is this a dual band network (i.e. one SSID for both 2.4 GHz and 5.0 GHz)? We've seen in the past some Create3's struggle with this kind of network, ideally you should have a dedicated 5.0 GHz network and a dedicated 2.4 GHz network.

Another note to add, when switch from simple discovery to discovery server, be sure to disconnect the Create3 from the wi-fi network via the webserver, and be sure to reconnect it when swapping from discovery server to simple discovery.

Let me know if you have any questions or if I can help with anything else.
Best Regards,
Saif

knickels, 2024-07-26:
That might be it, normally our campus has to add the MAC to a "whitelist" for it to get an IP from the WiFi.  Is there a way to check the MAC or assigned IP of the create3?

RustyCPR, 2024-08-26:
Hi @knickels 
Can you connect the TB4 to a home router or a smartphone WiFi Hotspot to detect the TB4 Mac ID? 
If using a Mobile phone Wifi hotspot in the Mobile hotspot menu> Connected Devices> TB4 ID> More information> MAC Address

knickels, 2024-09-13:
Hi @RustyCPR 
Yep, when I got it home and hooked into my wifi everything works normally.
So I guess I'll push on our campus IT a bit more.
Thanks for the extensive help!
--kn

BTW, I was able to find the TB4 MAC ID by rebooting it and looking at the logs on the TB4 through the web server interface:

Sep 13 13:39:28 iRobot-tb4-create3 daemon.notice network[1468]: [INFO] (src/wifi_control.c:925) found wlan0 iface: HWaddr 4C:B9:EA:2E:97:B7

smatarCPR, 2024-10-30:
Hello @knickels 
I will go ahead and close this issue. Should you encounter any follow up issues, do not hesitate to reach out by simply replying to this thread or opening up a new issue.
Best Regards,
Saif

closed 2024-10-30 by smatarCPR