turtlebot/turtlebot4 #414: RPI starts buffering randomly, Time sync does not worked out

Vasanth28897, 2024-05-22:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

```
anit@anit-GF63-Thin-9SCXR:~$ dpkg -s ros-$ROS_DISTRO-turtlebot4-desktop
Package: ros-humble-turtlebot4-desktop
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 41
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: amd64
Version: 1.0.0-1jammy.20240327.181848
Depends: ros-humble-turtlebot4-viz, ros-humble-ros-workspace
Description: Turtlebot4 Desktop Metapackage
```

### Type of issue

Other

### Expected behaviour

Raspberry pi should synchronize the time, and stops buffering randomly thru ssh.

### Actual behaviour

`sudo ntpdate` and other ways tried to get synchronize the date & time in the rpi, it stays the same like this
```
ubuntu@ubuntu:~$ date
Tue Apr 11 01:40:29 AM IST 2023
```
And the ros2 topic list is shown and available in the remote pc from RPI. Whenever we launch any file like(`view_model.launch.py` or `view_robot.launch.py` or `rplidar.launch.py` or `mobile_publisher.launch.py`) in the remote pc, the rpi(ssh terminal) starts buffering, and 
shows this error in the respective launch file launched
```
anit@anit-GF63-Thin-9SCXR:~$ ros2 launch turtlebot4_viz view_model.launch.py 
[INFO] [launch]: All log files can be found below /home/anit/.ros/log/2024-05-22-17-03-23-697699-anit-GF63-Thin-9SCXR-5469
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [5470]
[rviz2-1] [INFO] [1716377605.297677866] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1716377605.297813838] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1716377605.328423642] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1716377605.473630817] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1716377619.722805474] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1681156748.729 for reason 'discarding message because the queue is full'
```
we have to close the terminal forcefully, even we closed the launch file and restart the ssh. We are facing this issue from the start. 

We tried to test the sensors(rplidar, oakd-camera) in the remote pc, ended up this error.
```
anit@anit-GF63-Thin-9SCXR:~$ ros2 launch turtlebot4_bringup rplidar.launch.py
[INFO] [launch]: All log files can be found below /home/anit/.ros/log/2024-05-22-17-00-29-630837-anit-GF63-Thin-9SCXR-5253
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rplidar_composition-1]: process started with pid [5254]
[rplidar_composition-1] [INFO] [1716377429.719099727] [rplidar_composition]: RPLidar running on ROS2 package rplidar_ros. RPLIDAR SDK Version:2.0.0

[rplidar_composition-1] *** buffer overflow detected ***: terminated
[ERROR] [rplidar_composition-1]: process has died [pid 5254, exit code -6, cmd '/opt/ros/humble/lib/rplidar_ros/rplidar_composition --ros-args -r __node:=rplidar_composition --params-file /tmp/launch_params_3nv_ht4o'].
```

```
anit@anit-GF63-Thin-9SCXR:~$ ros2 launch depthai_examples mobile_publisher.launch.py
[INFO] [launch]: All log files can be found below /home/anit/.ros/log/2024-05-22-17-01-24-786394-anit-GF63-Thin-9SCXR-5354
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mobilenet_node-1]: process started with pid [5356]
[INFO] [robot_state_publisher-2]: process started with pid [5358]
[robot_state_publisher-2] [INFO] [1716377484.979541831] [oak_state_publisher]: got segment oak-d-base-frame
[robot_state_publisher-2] [INFO] [1716377484.979618816] [oak_state_publisher]: got segment oak-d_frame
[robot_state_publisher-2] [INFO] [1716377484.979627868] [oak_state_publisher]: got segment oak_imu_frame
[robot_state_publisher-2] [INFO] [1716377484.979632811] [oak_state_publisher]: got segment oak_left_camera_frame
[robot_state_publisher-2] [INFO] [1716377484.979637569] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[robot_state_publisher-2] [INFO] [1716377484.979642069] [oak_state_publisher]: got segment oak_model_origin
[robot_state_publisher-2] [INFO] [1716377484.979646561] [oak_state_publisher]: got segment oak_rgb_camera_frame
[robot_state_publisher-2] [INFO] [1716377484.979650901] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[robot_state_publisher-2] [INFO] [1716377484.979655330] [oak_state_publisher]: got segment oak_right_camera_frame
[robot_state_publisher-2] [INFO] [1716377484.979659645] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[mobilenet_node-1] terminate called after throwing an instance of 'std::runtime_error'
[mobilenet_node-1]   what():  No available devices
[ERROR] [mobilenet_node-1]: process has died [pid 5356, exit code -6, cmd '/opt/ros/humble/lib/depthai_examples/mobilenet_node --ros-args --params-file /tmp/launch_params_lcq72jz0 --params-file /tmp/launch_params_eg9k0mns --params-file /tmp/launch_params_mobof_jg --params-file /tmp/launch_params_oas3x3cv --params-file /tmp/launch_params_80pi6h62'].
```

### To Reproduce

I am not sure about how to reproduce these errors in your end. 

### Other notes

I reached the support team through mail, they asked me to add the response from this command here in this issue
```
ubuntu@ubuntu:~$ journalctl -b | grep networkd
Apr 11 00:58:36 ubuntu systemd-networkd[683]: lo: Link UP
Apr 11 00:58:36 ubuntu systemd-networkd[683]: lo: Gained carrier
Apr 11 00:58:36 ubuntu systemd-networkd[683]: Enumeration completed
Apr 11 00:58:36 ubuntu systemd-networkd[683]: usb0: Link UP
Apr 11 00:58:36 ubuntu systemd-networkd[683]: eth0: Link UP
Apr 11 00:58:37 ubuntu systemd[1]: Starting Dispatcher daemon for systemd-networkd...
Apr 11 00:58:38 ubuntu networkd-dispatcher[703]: No valid path found for iwconfig
Apr 11 00:58:38 ubuntu networkd-dispatcher[703]: No valid path found for iw
Apr 11 00:58:38 ubuntu systemd[1]: Started Dispatcher daemon for systemd-networkd.
Apr 11 00:58:39 ubuntu systemd-networkd[683]: wlan0: Link UP
Apr 11 00:58:42 ubuntu systemd-networkd[683]: usb0: Gained carrier
Apr 11 00:58:44 ubuntu systemd-networkd[683]: usb0: Gained IPv6LL
Apr 11 01:00:52 ubuntu systemd-networkd[683]: wlan0: Connected WiFi access point: LAB067 (6c:60:eb:cb:2f:cc)
Apr 11 01:00:52 ubuntu systemd-networkd[683]: wlan0: Gained carrier
Apr 11 01:00:54 ubuntu systemd-networkd[683]: wlan0: Gained IPv6LL
```

Awaiting your response on how to overcome this. Thanks in Advance.

Vasanth28897, 2024-05-31:
We are still encountering buffering sometimes. It happens at random time. We try to record the screen, and share here. 
below are the provision of regarding slam issues
full terminal output when launching slam
```
anit@anit-GF63-Thin-9SCXR:~$ ros2 launch turtlebot4_navigation slam.launch.py
[INFO] [launch]: All log files can be found below /home/anit/.ros/log/2024-05-31-11-52-06-710184-anit-GF63-Thin-9SCXR-5319
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [5320]
[sync_slam_toolbox_node-1] [INFO] [1717136526.833682049] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1717136526.868287314] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1717136526.868734644] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[sync_slam_toolbox_node-1] [INFO] [1717136539.473776002] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155992.744 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136542.296210180] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155992.867 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136542.297316715] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155993.002 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136542.298414786] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155993.125 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136542.301174461] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155993.248 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136542.388620360] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155995.667 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.360773406] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155995.790 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.360966018] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155995.925 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.361063392] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155996.048 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.361143590] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155996.171 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.361325988] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155996.306 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.361408111] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155996.429 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.361491588] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155996.551 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.445179543] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155998.713 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.564989328] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155998.836 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.693165539] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155998.971 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.815897192] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155999.094 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136545.939740350] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155999.217 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[sync_slam_toolbox_node-1] [INFO] [1717136546.075990363] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1681155999.352 for reason 'the timestamp on the message is earlier than all the data in [... cut ...]
Yes there are some timestamp related issues showing in the terminal while launching the slam.launch.py as you can see above.

this is for the status of timedatectl after restarting the ntpd/reboot the robot from terminal
```
ubuntu@ubuntu:~$ timedatectl status
               Local time: Tue 2023-04-11 01:01:59 IST
           Universal time: Mon 2023-04-10 19:31:59 UTC
                 RTC time: n/a
                Time zone: Asia/Kolkata (IST, +0530)
System clock synchronized: no
              NTP service: active
          RTC in local TZ: no
```

Regards
K.S.Vasanth

smatarCPR, 2024-06-12:
Hello K.S. Vasanth,
Apologies for the late reply on this. Thank you for providing the details regarding your setup. Let's try a tackle these issues one at a time, most likely they are all related to each other. 

The first issue we will addressing is the time syncing issues currently present. You mentioned that the Pi has access to the internet, are you able to perform the following:
- You mentioned that you were running a Discovery Server setup, but it appears that the Create3 has some wi-fi settings still saved from when running the simple discovery setup. Please ensure that these settings are cleared: https://iroboteducation.github.io/create3_docs/webserver/connect/
- Are you able to ping the Ubuntu NTP server `ping ntp.ubuntu.com` If you cannot, perhaps there are some network securities enabled by your organization which is blocking the Pi from connecting to these servers. If you cannot get around these securities/are unsure why this is occurring one potential solution is to have everything sync locally to a computer or server. 
- If you are able to successfully ping then please should provide the cat /etc/chrony/chrony.conf file. Additionally, please indicate if you have made any modifications to this file.

Looking forward to hearing from you soon.
Best Regards,
Saif

Vasanth28897, 2024-06-14:
Hello Saif,
Thanks for your reply. The NTP server is somehow blocking(by their server) the PI in our customer place because of the network securities. That's why we were not able to get the time sync done, and you also came up with the same suggestion. We did manage to work around on using the system clock as a NTP server and set the chrony service in the PI. Now the turtlebot4 is working fine, the slam and navigation also working fine. Thanks for your response.

Regards
K.S.Vasanth

closed 2024-06-17 by Vasanth28897