turtlebot/turtlebot4 #416: Help with error in navigation. (Smoother server and localization server sometimes giving trouble when trying to launch navigation)

alexdo105, 2024-05-26:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Newer one

### Type of issue

Networking

### Expected behaviour

I expect the robot to launch everytime without trouble localization and nav2 launch file...

### Actual behaviour

Sometimes, I can launch without trouble localization and nav2, but sometimes, I get an error showing timeouts in the smoother server (in the case of nav2), and in localization I get another server error. 

### Error messages

```bash
[smoother_server-2] [WARN] [1700600845.157126823] [smoother_server.rclcpp]: failed to send response to /smoother_server/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314
```

### To Reproduce

Run localization
Run Nav2

### Other notes

This kind of error did not happen when I had the turtlebot4 in simple discovery, but some other errors happened did happen really often, which involved network connection issues from the iCreate3 base. That is why I changed to discovery server. This setting helps the communication by making it faster and more reliable, but I often get the error shown above, and to fix it I just reboot the robot several times until it works again, sometimes I do not require to do this because it works seamlessly the first time. This behavior brings inconsistency and uncertainty to my project, and that is why I would appreciate a lot if you can show me how to fix it or explain to me why does this happen. An important note is that I always connect via ssh to the robot and launch everything here, which is why I think my computer has not anything to do with the error. I am not that much of a ROS2 expert, which is why I require your help. Thanks a lot btw.

RustyCPR, 2024-06-12:
Hi there @alexdo105 
Can you please provide full terminal outputs of what you are running. 
Also please provide journalctl -b > turtlebot4_log.txt 

Thanks,
Rusty

RustyCPR, 2024-07-08:
Closed due to no activity

closed 2024-07-08 by RustyCPR