turtlebot/turtlebot4 #417: I don't have the irobot_create_msgs package in my workspace

ValentinLeglise, 2024-05-29:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

Don't know

### Type of issue

Other

### Expected behaviour

While connected to the robot, i should be able to see the packages irobot_create_msgs using the command "ros2 pkg list"

### Actual behaviour

I am trying to do the tutorial "Creating my first node (Python)" in the user manual. I created my ws and my packages but when I started writing my python script, the irobot_create_msgs package is not recognized. I run the "ros2 pkg list" command and I only see the irobot_create_description package, whereas when I run the command after SSH in the robot, I see the irobot_create_msgs package.

Even using the code given in the github tutorials, my workspace can't find the package.

How can I get the package into my WS?

### To Reproduce

1. Follow tutorial to create my ws and my packages
2. Started to write my script and import irobot_create_msgs in my .py file and package.xml
3. "ros2 pkg list" to try to find the irobot_create_msgs package

CycleMark, 2024-05-30:
Hi @ValentinLeglise 

You should be able to download them from here:

https://github.com/iRobotEducation/irobot_create_msgs

Kind Regards

Mark

ValentinLeglise, 2024-05-30:
Hello @CycleMark,

Thank you. I downloaded in my ws and sourced but i still don't see the pkg when ros2 pkg list.

Best.

ValentinLeglise, 2024-06-05:
Hello,

Nevermind, i succeeded build the package correctly. Evering works now.

Best regards, Valentin.

closed 2024-06-05 by ValentinLeglise