turtlebot/turtlebot4 #418: No depth data obtained on oakd/stereo topics turtlebot4 lite

prachitgupta, 2024-05-31:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Version: 1.0.0-1jammy.20230228.211027 package turtlebot4_bringup installed from source
rest installed as usual 

### Type of issue

Camera

### Expected behaviour

get data on /oakd/stereo/image_raw/compressedDepth or any other oakd/stereo topics

### Actual behaviour

no stereo topics seen by default
on manually editing the configuration files of turtlebot4_bringup build from source getting oakd/stereo topics but no data is bieng published on them 

### Error messages

```bash
no error message , no data is being published
 
ubuntu@ubuntu:~$ ros2 topic list
/battery_state
/clicked_point
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/downsampled_costmap
/downsampled_costmap_updates
/function_calls
/global_costmap/costmap
/global_costmap/costmap_updates
/global_costmap/voxel_marked_cloud
/hazard_detection
/imu
/initialpose
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/local_costmap/costmap
/local_costmap/costmap_updates
/local_costmap/published_footprint
/local_costmap/voxel_marked_cloud
/local_plan
/map
/map_updates
/mobile_base/sensors/bumper_pointcloud
/mobility_monitor/transition_event
/mouse
/oakd/imu/data
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/oakd/stereo/camera_info
/oakd/stereo/image_raw
/oakd/stereo/image_raw/compressed
/oakd/stereo/image_raw/compressedDepth
/oakd/stereo/image_raw/theora
/odom
/parameter_events
/particle_cloud
/plan
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stop_status
/tf
/tf_static
/waypoints
/wheel_status
/wheel_ticks
/wheel_vels

ubuntu@ubuntu:~$ ros2 topic echo /oakd/stereo/image_raw/compressedDepth
```

### To Reproduce

using humble H.2.5 on create 3 , configured discovery server on rpi with turtlebot4 packages preinstalled with the flashed image
source installation and changing ros workspace settings to configure custom turtlebot4 package
    
Clone, source /opt/ros/humble/setup.bash and build turtlebot4_robot package
   
Modify the oakd_pro.yaml (or oakd_lite.yaml for Turtlebot 4 Lite) file to enable the stereo topic, see examples of different config files here: https://github.com/luxonis/depthai-ros/tree/humble/depthai_ros_driver/config
    
Link to your custom workspace using the turtlebot4-setup tool:
        run turtlebot4-setup
        navigate to ROS Setup -> Bash Setup
        select WORKSPACE_SETUP
        enter the path to your local setup.bash for your workspace. This will allow the turtlebot4 setup tool to locate and source these packages. e.g. /home/administrator/ros_ws/install/setup.bash
    
Reinstall the turtlebot4 service using the turtlebot4-setup tool under Robot Upstart -> Install.
        run turtlebot4-setup
        navigate to ROS Setup -> Robot Upstart
        select Install and agree as necessary

 oakd_lite.yaml file configurations

/oakd:
  ros__parameters:
    camera:
      i_enable_imu: true
      i_enable_ir: true
      i_nn_type: none
      i_pipeline_type: RGBD # Change to RGBD to enable depth
      i_usb_speed: SUPER_PLUS
    rgb:
      i_board_socket_id: 0
      i_fps: 30.0
      i_height: 720
      i_interleaved: false
      i_max_q_size: 10
      i_preview_size: 250
      i_enable_preview: true
      i_low_bandwidth: true
      i_keep_preview_aspect_ratio: true
      i_publish_topic: false
      i_resolution: '1080'
      i_width: 1280
    stereo:
      i_board_socket_id: 1
      i_fps: 30.0
      i_height: 480
      i_width: 640
      i_depth_align: true
      i_subpixel: true
      i_lr_check: true
      i_extended: false
      i_confidence_threshold: 200
      i_left_right_check_threshold: 5
      i_enable_preview: true
      i_publish_topic: true
    use_sim_time: false

also tried :
/oak:
  ros__parameters:
    camera:
      i_nn_type: none
      i_pipeline_type: Depth
    right:
      i_publish_topic: true
    stereo:
      i_extended_disp: true

### Other notes

ignoring the steps I have taken , please help me getting depth data from turtlebot4 lite in general 
completely new to working with this bot and urgent;y need depth data for my research project

RustyCPR, 2024-06-12:
Hi @prachitgupta 
I see you installed some of the initial packages from source.
1. Can you please detail what was modified and the hash version? 
2. Please share your manually edited files. 
3. Stop ros service by running - `sudo systemctl stop turtlebot4.service`
4. Manually run the [oak-d node](https://turtlebot.github.io/turtlebot4-user-manual/software/sensors.html#running-1) and share the output? 

Thanks,
Rusty

prachitgupta, 2024-06-17:
Thanks for the response
1) so I modified the config files turtlebot4_bringup package to enable the stereo node 
and changed the path of the workspace for the turtlebot4 service to build from while booting up to my local source installation. 
2) the files I edited were oakd_lite.yaml to enable stereo mode and oakd.launch (changed camera default model from pro to lite
posted as attachment

/oakd_lite.yaml
/oakd:
  ros__parameters:
    camera:
      i_enable_imu: true
      i_enable_ir: true
      i_nn_type: none
      i_pipeline_type: RGBD # Change to RGBD to enable depth
      i_usb_speed: SUPER_PLUS
    rgb:
      i_board_socket_id: 0
      i_fps: 30.0
      i_height: 720
      i_interleaved: false
      i_max_q_size: 10
      i_preview_size: 250
      i_enable_preview: true
      i_low_bandwidth: true
      i_keep_preview_aspect_ratio: true
      i_publish_topic: false
      i_resolution: '1080'
      i_width: 1280
    stereo:
      i_board_socket_id: 1
      i_fps: 30.0
      i_height: 480
      i_width: 640
      i_depth_align: true
      i_subpixel: true
      i_lr_check: true
      i_extended: false
      i_confidence_threshold: 200
      i_left_right_check_threshold: 5
      i_enable_preview: true
      i_publish_topic: true
    use_sim_time: false

/oakd.launch.py
#!/usr/bin/env python3
# Copyright 2021 Clearpath Robotics, Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#
# @author Roni Kreinin (rkreinin@clearpathrobotics.com)

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration, PathJoinSubstitution
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

from nav2_common.launch import RewrittenYaml

def generate_launch_description():
    pkg_turtlebot4_bringup = get_package_share_directory('turtlebot4_bringup')

    camera = LaunchConfiguration('camera')
    params_file = LaunchConfiguration('params_file')
    namespace = LaunchConfiguration('namespace')

    ARGUMENTS = [
        DeclareLaunchArgument('camera', default_value='oakd_lite'),
        DeclareLaunchArgument('params_file',
                              default_value=[PathJoinSubstitution(
                                [pkg_turtlebot4_bringup, 'config', camera]), '.yaml']),
        DeclareLaunchArgument('namespace', default_value='',
                              description='Robot namespace')
    ]

    namespaced_param_file = RewrittenYaml(
        source_file=params_file,
        root_key=namespace,
        param_rewrites={},
        convert_types=True)

    node = ComposableNodeContainer(
            name='oakd_container',
            namespace=namespace,
            package='rclcpp_components',
            executable='component_container',
            composable_node_descriptions=[
                    ComposableNode(
                        package='depthai_ros_driver',
                        plugin='depthai_ros_driver::Camera',
                        name='oakd',
                        parameters=[namespaced_param_file]
                    ),
            ],
            output='screen',
        )

    ld = LaunchDescription(ARGUMENTS)
    ld.add_action(node)
    return ld

here's the output of  ros2 launch turtlebot4_bringup oakd.launch.py

[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-06-17-13-24-40-594365-ubuntu-1681
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [1694]
[component_container-1] [INFO] [1718630682.886451880] [oakd_container]: Load Library: /opt/ros/humble/lib/libdepthai_ros_driver.so
[component_container-1] [INFO] [1718630684.381246145] [oakd_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1718630684.381470602] [oakd_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-1] [INFO] [1718630684.448653938] [oakd]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [INFO] [1718630687.168663093] [oakd]: Camera with MXID: 19443010C1F73C1200 and Name: 1.1.2 connected!
[component_container-1] [INFO] [1718630687.170477487] [oakd]: USB SPEED: SUPER
[component_container-1] [INFO] [1718630687.199171366] [oakd]: Device type: OAK-D-LITE
[component_container-1] [INFO] [1718630687.203198420] [oakd]: Pipeline type: RGBD
[component_container-1] [INFO] [1718630687.265120070] [oakd]: Finished setting up pipeline.
[component_container-1] [19443010C1F73C1200] [1.1.2] [0.969] [MonoCamera(6)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[component_container-1] [19443010C1F73C1200] [1.1.2] [0.971] [MonoCamera(4)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[component_container-1] [INFO] [1718630687.785046729] [oakd]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oakd' in container '/oakd_container'
[component_container-1] [19443010C1F73C1200] [1.1.2] [2.079] [IMU(9)] [critical] IMU driver failed with error code 1. Your board likely doesn't have IMU!

RustyCPR, 2024-06-24:
Hi @prachitgupta 
Can you update the two imu and ir parameters in the oakd_lite.yaml to false, save and retest? 

```
  ros__parameters:
    camera:
      i_enable_imu: false
      i_enable_ir: false
```
Regards,
Rusty

RustyCPR, 2024-07-08:
Closed due to no activity

closed 2024-07-08 by RustyCPR