turtlebot/turtlebot4 #419: Can't save a map. [ERROR] [robot2.map_saver]: Failed to spin map subscription

ValentinLeglise, 2024-06-03:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

*

### Type of issue

Select One

### Expected behaviour

Running the command 
_ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "name:
  data: 'map_name'"_
or the command
_ros2 run nav2_map_server map_saver_cli -f "map_name" --ros-args -p map_subscribe_transient_local:=true -r __ns:=/namespace_

and get a .pgm file to see my map.

### Actual behaviour

I used 3 terminals:

1. _ros2 launch turtlebot4_navigation slam.launch.py_ to start SLAM
2. _ros2 launch turtlebot4_viz view_robot.launch.py_ to see the SLAM on Rviz
3. _ros2 run teleop_twist_keyboard teleop_twist_keyboard_ to drive the robot while mapping.

After i finished my map, i Ctrl+C the terminal 3 to stop driving and then run _ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "name:
  data: 'map_name'"_. 

It gave me the result : 
_response:
slam_toolbox.srv.SaveMap_Response(result=0)_

I also tried : 
_ros2 run nav2_map_server map_saver_cli -f "map_name" --ros-args -p map_subscribe_transient_local:=true -r __ns:=/namespace_

and the result is : 
[INFO] [1717408486.515404440] [robot2.map_saver]: 
	map_saver lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1717408486.515544182] [robot2.map_saver]: Creating
[INFO] [1717408486.515585540] [robot2.map_saver]: Configuring
[INFO] [1717408486.516818275] [robot2.map_saver]: Saving map from 'map' topic to 'map_name' file
[WARN] [1717408486.516869291] [robot2.map_saver]: Free threshold unspecified. Setting it to default value: 0.250000
[WARN] [1717408486.516878688] [robot2.map_saver]: Occupied threshold unspecified. Setting it to default value: 0.650000
**[ERROR] [1717408488.518377685] [robot2.map_saver]: Failed to spin map subscription**
[INFO] [1717408488.519734102] [robot2.map_saver]: Destroying
[ros2run]: Process exited with failure 1

### To Reproduce

See actual behavior

WallerJonathan, 2024-06-04:
Hi Valentin,

The first command you shared was the old one on their user manual which I believe does not work anymore. In addition, I believe you had a typo in your other command. Note that in the command `map_name` refers to the name you want to give the map and the section `-r __ns:=/namespace` (I believe you were missing the double underscore) is only required if you namespaced your robot. If you did not namespace, you can just leave it out.

Here is the fixed command that you should add your substitutions:
`ros2 run nav2_map_server map_saver_cli -f "map_name" --ros-args -p map_subscribe_transient_local:=true -r __ns:=/namespace`

RustyCPR, 2024-06-24:
Hi @ValentinLeglise did @WallerJonathan feedback help here? 
Thanks,
Rusty

ValentinLeglise, 2024-06-25:
Hello,

I tried @WallerJonathan recommandation and it only worked after several attempts (I don't know why, but I managed to register my card).

Thanks for your help.

RustyCPR, 2024-07-22:
Thanks @ValentinLeglise

closed 2024-07-22 by RustyCPR