turtlebot/turtlebot4 #420: Generating a map not possible

Jan-HAP, 2024-06-04:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

-

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

With
ros2 launch turtlebot4_navigation slam.launch.py
and 
ros2 launch turtlebot4_viz view_robot.launch.py

I wanted to follow the "Generating a map" tutorial.

### Actual behaviour

Rviz states that "No map received" and "Frame [map] does not exist.
In the console I once saw the messages that the timestamps are wrong and mostly that the message is discarded because the queue is full. 
Using rviz with _ros2 launch turtlebot4_viz view_model.launch.py_ works without a problem.
I attached pictures of the messages and rviz. I can also see the complete topic list

[image]
[image]
[image]
[image]
[image]
[image]

### To Reproduce

-

smatarCPR, 2024-06-13:
Hello @Jan-HAP ,
Thank you for reaching out on the Turtlebot 4 Github page.

From the terminal outputs provided, it appears that there is a time sync issue, in figure 3the line stating "timestamp on the message is earlier than all the data in transform cache" is seen when a time sync issue is present.

Try the following suggestions and see if the time sync issue is resolved:
- Restart the ntpd from the create3 webserver (https://iroboteducation.github.io/create3_docs/webserver/restart-ntpd/)
- Similarly try forcing a time sync on your laptop. (More details on how to check last time sync etc on ubuntu [here](https://ubuntu.com/server/docs/use-timedatectl-and-timesyncd). 
- Run the command: systemctl status systemd-timesyncd and see when the timesync last occurred. 
- For the Raspberry Pi, you can use `systemctl status chrony` to see when the last sync was and you can force a sync by restarting the chrony service.
- Ensure that the Raspberry Pi can ping the Ubuntu NTP servers using `ping ntp.ubuntu.com`

Let me know how these suggestions work for you. Looking forward to hearing from you soon.
Best Regards,
Saif M.

Jan-HAP, 2024-06-17:
Hello,
thanks for the reply. As you suggested to ping "ntp.ubuntu.com" I guess I would need a setup where the devices have access to the internet. Would it be possible to have setup where they are just connected in a network without internet?

smatarCPR, 2024-07-08:
Hello @Jan-HAP ,
You do not necessarily need to have a system with an active internet connection. You can run a network without access to the internet.

If you do run this setup, you will need to setup up your user pc as an ntp server. The final setup should have:
- The Raspberry Pi connects and syncs with the user PC at its address.
- The Create3 will connect and sync with the Raspberry Pi.

Let me know if you have any questions or if I can help with anything else.
Best Regards,
Saif

closed 2024-08-19 by smatarCPR