turtlebot/turtlebot4 #421: Long delay to respond after publish a command on /cmd_vel

ValentinLeglise, 2024-06-05:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

*

### Type of issue

Networking

### Expected behaviour

Run the command:
 _ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.3" -r 30_

and the robot responds quickly by turning on itself.

### Actual behaviour

Publish the command and the robot wait more than 10 sec before responding. You can see it with the video attached.

https://github.com/turtlebot/turtlebot4/assets/148329757/fa7aab1d-9a9a-4990-bccc-48ea8be2ded8

The robot user has already tried to reflash the Rpi card and run the command while connected to the robot via SSH. 

The Rpi is connected to a 5GHz network while the Create3 base is connected to a 2.4GHz network.

Interestingly, when the robot is controlled with the joystick, it responds directly to the joystick commands.

### To Reproduce

run  _ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.3" -r 30_

RustyCPR, 2024-06-12:
Hi @ValentinLeglise 
Can you please make sure the packages are all updated.  `sudo apt update && sudo apt upgrade `
Make sure the Create3 is also updated: https://iroboteducation.github.io/create3_docs/webserver/update/
Turn off diagnostics in the [setup.bash file](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html?highlight=diagnostics#configuration-tools)
Please try to replicate the issue and then provide the[ Create3 logs. ](https://iroboteducation.github.io/create3_docs/webserver/logs/)
Thanks,
Rusty

RustyCPR, 2024-07-08:
Hi @ValentinLeglise - closing this due to no-activity - feel free to re-open.

closed 2024-07-08 by RustyCPR