turtlebot/turtlebot4 #422: Turtlebot4 SLAM do not work

PierreSteve1, 2024-06-06:
### Robot Model

Turtlebot4 Standard

### ROS distro

Galactic

### Networking Configuration

Simple Discovery

### OS

Ubuntu 20.04

### Built from source or installed?

Installed

### Package version

Debian package

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I want to be able to correctly launch the synchronous, asynchronous and Rviz slam graphical interface

### Actual behaviour

Hello, I am currently working on turtlebot4 standard. I am in the turtlebot4 common part of the turtlebot4 user manual. As mentioned in the guide, I started by installing the individual packages and launch files. You can see this in my screenshots screen. Once everything is installed, when I now enter this command on my Linux terminal: ros2 launch turtlebot4_navigation [nav_bringup.launch.py](http://nav_bringup.launch.py/) slam:=sync , I receive error messages that I highlighted in white finally from you show what it is. The command does not launch the GUI. Also the other command: ros2 launch turtlebot4_navigation [nav_bringup.launch.py](http://nav_bringup.launch.py/) slam:=async does not work too. That's not all, I I then tried to launch rviz with the command: ros2 launch turtlebot4_viz [view_robot.launch.py](http://view_robot.launch.py/), but I cannot view the map. I just see a graphical interface without a map when launching rviz. Can you help me please would you like to resolve this?
[image]
[image]
[image]
[image]

### Error messages

```bash
Could not find a connection beetween Odom and base_link.Because they are not part or the same tree.Tf has two or more unconnected trees.
rplidar_link error.
```

### To Reproduce

Installation

GALACTIC
HUMBLE
Debian package

Individual packages can be installed through apt:

sudo apt update
sudo apt install ros-galactic-turtlebot4-description \
ros-galactic-turtlebot4-msgs \
ros-galactic-turtlebot4-navigation \
ros-galactic-turtlebot4-node

Navigation

The turtlebot4_navigation packages contains launch and configuration files for using SLAM and navigation on the TurtleBot 4. It also contains the TurtleBot 4 Navigator Python node.

GALACTIC
HUMBLE
Launch files:

[Nav Bringup](https://github.com/turtlebot/turtlebot4/blob/galactic/turtlebot4_navigation/launch/nav_bringup.launch.py): Launches navigation. Allows for launch configurations to use SLAM, Nav2, and localization.
[SLAM Sync](https://github.com/turtlebot/turtlebot4/blob/galactic/turtlebot4_navigation/launch/slam_sync.launch.py): Launches slam_toolbox with online synchronous mapping. Recommended for use on a PC.
[SLAM Async](https://github.com/turtlebot/turtlebot4/blob/galactic/turtlebot4_navigation/launch/slam_async.launch.py): Launches slam_toolbox with online asynchronous mapping. Recommended for use on the Raspberry Pi 4.
Running synchronous SLAM:

ros2 launch turtlebot4_navigation nav_bringup.launch.py slam:=sync
Running asynchronous SLAM with Nav2:

ros2 launch turtlebot4_navigation nav_bringup.launch.py slam:=async
Running Nav2 with localization and existing map:

ros2 launch turtlebot4_navigation nav_bringup.launch

smatarCPR, 2024-06-13:
Hello @PierreSteve1 ,
Thank you for reaching out on the Turtlebot 4 Github page.

Based on the the terminal outputs provided, it seems that the underlying issue is most likely a breakdown of communication between the Raspberry Pi, Create3 and your computer or a time sync issue.

Let's tackle the communication issues first. On your computer, can you see all topics from the Turtlebot 4 and Create3 (for example, we can see the rplidar exists which suggests the Raspberry Pi is communicating with your computer fine, but see if you can see and echo topics such as battery and wheel status). Is the complete list of topics that appear when ssh'ed to the Turtlebot 4 matched by the list of topics available on your offboard computer?

Next let's see how communication is between the Create3 and Raspberry Pi. First would it be possible to provide a copy of the latest Create3 logs (https://iroboteducation.github.io/create3_docs/webserver/logs/)? From those we'll see if the Create3 is connected to the wireless network. Do the ROS 2 domain settings in the Create3 match the ones being used by the Turtlebot 4 and your computer?

Should the above two checks pass, we'll begin looking into any potential time sync issues.

Lastly, I want to clear up some potential misconceptions, when launching the navigation stack `ros2 launch turtlebot4_navigation nav_bringup.launch.py slam:=sync` no GUI will be launched be default. After launching this command, in a separate terminal you can launch RVIZ (`ros2 launch turtlebot4_viz view_robot.launch.py`).

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif

PierreSteve1, 2024-06-14:
hello Saif,
Regarding the list of topics, the command `ros2 topic list` executed on PC returns the following topics:
```
/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels
```
which is exactly the same as in the terminal when ssh'ed to the RPi. The command `ros2 topic echo /topic_name` seems to work fine for all topics, but there is a delay of about 3 seconds before receiving anything on the terminal.

Here you can fine the latest Create3 [logs](https://github.com/user-attachments/files/15842751/messages.txt). Please, note that the date given in the logs `Jun 28` doesn't correspond to today's date, which is weird.

The ROS_DOMAIN_ID is the same for Create3, Turtlebot4 and the PC.

PierreSteve1, 2024-06-19:
I would like to add this remark regarding the connections between the frames. I used this command `ros2 run tf2_tools view_frames` to view the tree, which shows two disconnected trees. This is visualised in the following [file.pdf](https://github.com/user-attachments/files/15903016/frames1.pdf):
[image]
This shows that there is a connection between 'odom' and 'map' but they are isolated from the rest of the tree.

pedrocgr, 2024-06-19:
I'm having the same problem. Echoing the topics shows that both outputs(SSH Terminal and Client side) are the same. Some topics like /scan or /stereo are not showing up.

I hope you can solve your problem and maybe give me some light. I tried running:

`turtlebot4-ntpd-sync`
in the SSH terminal but the error is still there.

My topic list:
[image]

smatarCPR, 2024-06-24:
Hello @PierreSteve1 ,
Thank you for sending the requested details and apologies for the late reply on this.

In terms of what we could find:
- You mentioned that the Create3 topics (/odom) was seen on your computer, were you able to echo the topic on your computer (**not while** SSH'ed into the Pi)?
- From the logs provided, it appears that the Create3 is not connected to Wi-Fi. If it was it would be able to time sync. In the logs we can see that the Create3 attempts to sync with an NTP server at 10.0.0.1, but it never appears to connect. This raises two questions:
- Is the Create3 connected to a wi-fi network which has access to internet? You should be able to view an IP address in the connect tab in the webserver (https://iroboteducation.github.io/create3_docs/webserver/connect/)
- The second question, have you made any changes to the ntp.conf for the Create3? The address 10.0.0.1 is not a standard NTP server address that we set as standard. We do not have a record of that address which leads us to suspect that a change has been made to ntp.conf for the Create3 updating it to use that address.

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif

PierreSteve1, 2024-07-03:
hello Saif,
Thank you for the feedback. You are right, it was a connection to internet issue. We have now correctly configured our rooter and the mapping seems to work fine.
Even though the problem is now solved, I have some remarks:

1. `daemon` doesn't start automatically at the RPi, and we always need to start it manually each time we ssh to the robot.
2. The date given in the logs is now correct, but the time is still off.  [logs](https://github.com/user-attachments/files/16088116/messages.9.txt)
3.   Almost all errors regarding the RobotModel are now gone, but there still two errors. See attached figures.
[image]
[image]

Do you have any recommendations to solve those issues.

smatarCPR, 2024-07-04:
Hello @PierreSteve1 ,
Glad to hear that progress was made A couple of notes regarding your observations:
1. The Daemon does not start automatically on boot, it starts after the first ROS 2 command is given (e.g. ros2 topic list). Calling this command should auto start the Daemon.
2. The errors we see in the logs provided, seem to resolve itself after some time. The error itself is not a timesync error, more a communication error with the specified address, but after switching to the new address the issue seems to resolve itself. Keep a close eye on it but for now, it should be fine
3. For these errors, the bugfix for these was released in the Humble version of the packages. If you are able to switch to Humble, you'll find that these errors no longer appear. If you must stay in Galactic, no worries, those errors should have no affect on functionality.

Let me know if you have any questions or if I can help with anything else.
Best Regards,
Saif