turtlebot/turtlebot4 #423: Turtlebot4's unable to communicate with remote pc

Nickelleee, 2024-06-07:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

1.0.0-1jammy.20240517.212705 (used sudo apt install ros-humble-turtlebot4-desktop)

### Type of issue

Networking

### Expected behaviour

- ```ip route``` returns an IP from the turtlebot4 on remote pc
- ```ros2 topic list``` lists topics from turtlebot4 on remote pc
- ```ros2 multicast receive``` and ```ros2 multicast send``` communicates between remote pc and turtlebot4

### Actual behaviour

- ```ros2 topic list``` only outputs ```/parameterevents``` and ```/rosout``` on remote pc
- Remote pc can ping turtlebot4, but when ssh'd into turtlebot4 it cannot ping remote pc (can ping other turtlebots)
- ros2 multicast does not work between remote pc VM and turtlebot4
- ```ip route``` does not give any route according to turtlebot's ip

### To Reproduce

1. Follow steps outlined at: https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html inside a virtual machine
2. Attempt to ping or multicast between virtual machine and turtlebot4

### Other notes

My Ubuntu environment is ran through VirtualBox. The turtlebot4 is able to ping my ip from windows when the firewall is disabled, but not from the virtual machine, even with anything firewall related disabled. I am unable to get any world data in Rviz using SLAM but I am assuming that is caused by the networking issues described.

Nickelleee, 2024-06-07:
Update: I have the bot able to ping back now that I have bridged adapter for the virtual machines network settings. I now also get the topics from the bot.

However, when I SSH into the bot I can't get anything from ```ros2 multicast receive``` when I ```ros2 multicast send``` from my non-SSH'd terminal even though the reverse works for some reason.

RustyCPR, 2024-06-12:
Hi @Nickelleee 
Thanks for your question but your request is outside of the scope of our support. We don't support VM setups. 
You can try posting the question here: https://robotics.stackexchange.com/

Best regards,
Rusty

RustyCPR, 2024-07-08:
Closed due to non-activity

closed 2024-07-08 by RustyCPR