turtlebot/turtlebot4 #424: Unable to read partition as FAT

codingBigBird, 2024-06-08:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

 turtlebot4_standard_humble_1.0.0.zip 2023-03-03 23:51 1.9G

### Type of issue

Other

### Expected behaviour

I can follow the robot setup after installing latest Raspberry Pi image

### Actual behaviour

[image]

### To Reproduce

Find the latest TurtleBot 4 Raspberry Pi images at http://download.ros.org/downloads/turtlebot4/.

    Download the latest image and extract it.
    Power off your robot and then remove the microSD card from the Raspberry Pi.
    Insert the microSD card into your PC. You may need an adapter.
    Install the imaging tool dcfldd

sudo apt install dcfldd

    Identify your SD card

sudo fdisk -l

wget https://raw.githubusercontent.com/turtlebot/turtlebot4_setup/humble/scripts/sd_flash.sh
bash sd_flash.sh /path/to/downloaded/image

### Other notes

I just follow the steps of installing a new image on the Raspberry Pi. But always meet the error of 'Unable to read partition as FAT'. I tried to format the SD card as FAT 32 partition and change the SD cards, but it always shows the problem like this. I also flash the Raspberry Pi OS to a new SD card and I can use it. What might be the issue here?

codingBigBird, 2024-06-08:
And if I use this command it shows :
sudo dosfsck -w -r -l -v -t /dev/sdb1
fsck.fat 4.2 (2021-01-31)
Currently, only 1 or 2 FATs are supported, not 251.

smatarCPR, 2024-06-13:
Hello @codingBigBird ,
Thank you for reaching out on the Turtlebot 4 Github page. 

When identifying the SD card what is the output?

When trying to flash the SD card, what are the exact commands you are using (i.e. do you specify the path to the extracted file?)

What are you inputting as the SD card device? Often times, the sd card identifying command will list the partitions available on the drive, for example, when I identify the SD card I get /dev/sda0 and /dev/sda1, but they are both listed under /dev/sda. 

When flashing the device, you should specify to use the entire device, not a partition of the device.

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif M.

closed 2024-08-19 by smatarCPR