turtlebot/turtlebot4 #425: Discovery server and Remote PC connection to RPi4 through Ethernet

lasuomela, 2024-06-13:
### Robot Model

Turtlebot4 Standard

### ROS distro

Galactic

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

ros-humble-turtlebot4-bringup/jammy,now 1.0.2-1jammy.20240606.001820 arm64 [installed,automatic]

### Type of issue

Networking

### Expected behaviour

Hi,

I'm trying to configure the Turtlebot4 to communicate with a remote PC through a wired Ethernet connection, using the Discovery server. On the remote PC, `ros2 topic list` shows the topics running both on the RPi4 and the Create3, but `ros2 topic echo` only produces output for the nodes running on RPi4. If I run `ros2 topic echo` on the RPi4, I get output also from the nodes running on Create3.

Setup:
Jetson AGX Orin --Ethernet--> Router
RPi4                    --Ethernet--> Router
Create3               --Usb--> RPi4

I have a new Turtlebot4. I started by flashing the RPi4 SD card to Ubuntu 22.04 and set up Turtlebot software using the setup scripts (https://github.com/turtlebot/turtlebot4_setup/tree/humble). I then updated Create3 firmware to H2.6 through the web portal. On Orin, I'm running ROS2 Humble inside docker, image is  `dustynv/ros:humble-desktop-l4t-r36.2.0` from https://hub.docker.com/r/dustynv/ros/tags.

Next, I followed the instructions on
https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html
except replacing the wifi interface with eth0 interface of my AGX Orin. On Orin, `ip route` prints
 `192.168.186.0/24 via 192.168.1.77 dev eth0`, where `192.168.1.77` is the ip my router assigns to the RPi4. I assume there still must be some problem with forwarding the traffic from Create3 to the Orin. I can ping Create3 at `192.168.186.2` both from inside and outside the docker container on Orin.
 
Are there any modifications I should add to the setup process when I'm using Ethernet instead of WIFI? If not, do you have ideas about possible solutions?

Best regards,
Lauri Suomela

### Actual behaviour

-

### To Reproduce

-

lasuomela, 2024-06-14:
Update:

I can send ROS messages to Create3 from Orin / Remote PC. Publishing to `/cmd_vel` makes the Turtlebot move. However, I'm still unable to listen to Create3 topics on Orin.

Also, to be able to ping Create3 from the Orin, I have to set
`sudo iptables -A FORWARD -o usb0 -j ACCEPT`
`sudo iptables -t nat  -A POSTROUTING -o usb0 -j MASQUERADE`
on the RPi4.

lasuomela, 2024-06-14:
Solved:

For the traffic from Create3 to be forwarded to the `eth0` interface, the WiFi setting on RPi4 has to be in the 'Client' mode with the SSID and password empty. For me, RPi4 was in the 'Access point' mode, so I guess the traffic from Create3 was sent there.

Closing the issue.

lasuomela, 2024-06-19:
The issue still persists, switching wifi from Access Point to Client mode wasn't the fix after all. The way to get the connection between Create3 and Remote PC to work is to boot up both Create3 and RPi4, then ssh into RPi4, and issue sudo reboot. After the reboot, messages come through from Create3 to Remote PC.

Now, this is very tedious to do every time the robot is turned on, since rebooting the RPi4 takes several minutes. Any ideas how to fix this?

smatarCPR, 2024-06-26:
Hello @lasuomela ,
Thank you for reaching out on the Turtlebot 4 github. I am very sorry to hear that you are encountering issues with your Turtlebot 4.

Due to the way that the Raspberry Pi is setup, it has a fixed IP address of 192.168.185.3 on the ethernet interface. Based on your setup goals our recommendation is to:
Set the network for your router to utilise the 192.168.185.X subnet, and reserve/assign the 192.168.185.3 address to the Raspberry Pi for the Turtlebot 4.

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif

lasuomela, 2024-06-27:
Hello,

Thanks for the suggestion. Even if I set the router to use the same subnet (192.168.185.X) with DHCP turned off, the problem persists. I need to ssh into the RPi4 and reboot it once after the initial startup of Create3 and RPi4 to get the ROS messages from Create3 to Remote PC.

Let me know if you have other ideas to try.

Br,
Lauri

Ragnar7982, 2024-06-30:
Hello, i'm using turtlebot4 standard(galactic) and ROS2 galactic, i followed the manual steps and tried to use discovery server,but it didn't work. Can you explain how exactly change to discovery server? 
TYVM

smatarCPR, 2024-07-04:
Hello @lasuomela ,
Would it be possible to share the journalctl logs from the Raspberry Pi? SSH into the pi and run the command: `journalctl -b > turtlebot4_log.txt`. Please send the generated .txt file. We'll review the details and let you know what we find.

@Ragnar7982 , would it be possible to open a new Github ticket and provide some details about what specifically did not work? We will assist you in that new thread so there is no cross wiring of information between yourself and @lasuomela.

Looking forward to hearing from you soon!
Best Regards,
Saif

lasuomela, 2024-07-15:
Hi,

Sorry for the delay. Here are the journalctl logs both before and after rebooting the RPi.

[turtlebot4_log.txt](https://github.com/user-attachments/files/16234209/turtlebot4_log.txt)
[turtlebot4_log_after_reboot.txt](https://github.com/user-attachments/files/16234208/turtlebot4_log_after_reboot.txt)

Br,
Lauri

smatarCPR, 2024-07-24:
Hello @lasuomela ,
Thank you for sending the log files requested. From the details we could find we have the following proposed fix for this matter:

SSH into the turtlebot and navigate to the /etc/netplan folder. Within this folder there should be a YAML file named 40-ethernets.yaml. Use sudo alongside your preferred text editor (for example nano) and modify this file removing the two lines that say: `optional: true` This should fix the network wait online function and it should now take into account the eth0 and usb0 networks rather that focus solely on the wireless interface.

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif

lasuomela, 2024-08-08:
Hello,

Thanks for looking into it. I edited netplan/40-ethernets.yaml as suggested. However, the issue still doesn't go away. Let me know if you have any other ideas.

Thanks!
-Lauri

smatarCPR, 2024-08-09:
Hello Lauri,
Would it be possible to provide the following:
- Please provide the latest logs using the command: `journalctl -b > turtlebot4_log.txt`
- Please provide the contents of the /etc/netplan folder, we want to review all the netplan yaml files that are currently active and in use. You may edit the files beforehand, removing any network passwords that may be present inside these files.

Looking forward to hearing from you soon.
Best Regards,
Saif

lasuomela, 2024-08-12:
Hi,

I think I finally solved the issue. In short, 

1. The robot namespace specification for the create3_republisher node in https://github.com/turtlebot/turtlebot4_robot/blob/humble/turtlebot4_bringup/launch/robot.launch.py#L80 (line 80) is malformed. I had to change the argument `_do_not_use` to `/_do_not_use` because of how 
https://github.com/iRobotEducation/create3_examples/blob/humble/create3_republisher/republisher.cpp#L373 
handles topic name parsing. The original approach caused the create3_republisher to crash with

```
Aug 12 11:44:19 turtlebot4 turtlebot4-start[1114]: [create3_republisher-3] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidTopicNameError'
Aug 12 11:44:19 turtlebot4 turtlebot4-start[1114]: [create3_republisher-3]   what():  Invalid topic name: topic name must not contain repeated '/':
Aug 12 11:44:19 turtlebot4 turtlebot4-start[1114]: [create3_republisher-3]   '//_do_not_use/battery_state'
```

2. I'm using the turtlebot4 driver from before the multi-device discovery server support. Before this, the turtlebot setup script didn't set Create3 namespace to `_do_not_use`. I had to set it manually. Seems like the newest version of turtlebot4 setup handles this automatically.

I don't know why the republisher namespace issue didn't show up in the journalctl logs before. However, now everything seems to work. I'm not sure if the namespace issue is something specific to the package versions I have on the turtlebot, but I suppose this problem would come up with wifi connections also. Hope this helps you continue improving the turtlebot drivers. 

Thanks for the help!
-Lauri

lasuomela, 2024-08-22:
Hi,

I have version create3_republisher version 0.0.5 on the RPi, so it doesn't include https://github.com/iRobotEducation/create3_examples/commit/37311feca69bd81e92e0123a4b853b340bab92c7. I suppose that commit will fix the issue.

Thanks for the support!
-Lauri

closed 2024-08-22 by lasuomela