turtlebot/turtlebot4 #426: Map while using SLAM is getting Distorted traversing some distance

Rahul007RR, 2024-06-14:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

humble

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Create a map in a proper way as shown in tutorials

### Actual behaviour

While performing the SLAM and generating a map, the map is getting distorted as show below.
[image]

### Error messages

```bash
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [32607]
[rviz2-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[rviz2-1] [INFO] [1718362029.594707625] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1718362029.594778064] [rviz2]: OpenGl version: 4.3 (GLSL 4.3)
[rviz2-1] [INFO] [1718362029.601496305] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1718362030.865796886] [rviz2]: Trying to create a map of size 155 x 164 using 1 swatches
[rviz2-1] [ERROR] [1718362030.897419377] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
[rviz2-1] active samplers with a different type refer to the same texture image unit
[rviz2-1] [ERROR] [1718362031.091367957] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1700600512.330883 but the latest data is at time 1700600512.286958, when looking up transform from frame [rplidar_link] to frame [map]
[rviz2-1] [INFO] [1718362031.635140799] [rviz2]: Trying to create a map of size 156 x 164 using 1 swatches
```

### To Reproduce

ros2 launch turtlebot4_navigation slam.launch.py
ros2 launch turtlebot4_viz view_robot.launch.py

control the turtlebot4 with keyboard using the following command
ros2 run teleop_twist_keyboard teleop_twist_keyboard

RustyCPR, 2024-07-04:
Hi @Rahul007RR 
Thanks for your patience we looked into this. 

Can you please: 
1. Set up time sync / NTP to ensure timesync is setup properly between RPi and C3. [Detail here](https://github.com/turtlebot/turtlebot4/issues/216)
2.  Please repeat the above again and provide the full terminal output of both RVIZ and full terminal output from the SLAM Node terminal.

Thank you,
Rusty

RustyCPR, 2024-09-09:
Closing due to no activity

closed 2024-09-09 by RustyCPR