turtlebot/turtlebot4 #427: After Setting the Initial pose on the map , it is similar to when generating a map and not navigation, localization is not happening

Rahul007RR, 2024-06-14:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

ros-humble-turtlebot4

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

localization and navigation after setting the initial pose

### Actual behaviour

No localisation is seen and the map is same as while generating a map
[image]
[image]

### To Reproduce

ros2 launch turtlebot4_navigation localization.launch.py map:=office.yaml
ros2 launch turtlebot4_navigation nav2.launch.py
ros2 launch turtlebot4_viz view_robot.launch.py

RustyCPR, 2024-07-04:
Hi @Rahul007RR 
Thanks for your patience. 
Can you please expand on what the issue you are seeing is? 

[Localization](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html#localization)
Localization uses an existing map along with live odometry and laserscan data to figure out the position of the robot on the given map. It does not update the map if any changes have been made to the environment, but we can still avoid new obstacles when navigating. Because the map doesn't change, we can get more repeatable navigation results.

Thank you,
Rusty

taara67, 2024-08-28:
I have a similar problem.

I was successful in generating the map and nav_to_pose.py once,
but I can't reproduce it.

The execution procedure is the same:
```
ros2 launch turtlebot4_navigation localization.launch.py ​​map:=office.yaml

ros2 launch turtlebot4_navigation nav2.launch.py

ros2 launch turtlebot4_viz view_robot.launch.py
```
----
nav2 log
```
[controller_server-1] [ERROR] [1724825409.267286839] [local_costmap.local_costmap]: StaticLayer: "map" passed to lookupTransform argument target_frame does not exist. 
[planner_server-3] [INFO] [1724825409.580182719] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[controller_server-1] [ERROR] [1724825409.666773656] [local_costmap.local_costmap]: StaticLayer: "map" passed to lookupTransform argument target_frame does not exist. 
[controller_server-1] [ERROR] [1724825409.974961593] [local_costmap.local_costmap]: StaticLayer: "map" passed to lookupTransform argument target_frame does not exist. 
[planner_server-3] [INFO] [1724825410.080182260] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[controller_server-1] [ERROR] [1724825410.282637001] [local_costmap.local_costmap]: StaticLayer: "map" passed to lookupTransform argument target_frame does not exist. 
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)

```

----
rviz log

```
s full'
[rviz2-1] [INFO] [1724823856.778712920] [rviz2]: Message Filter dropping message: frame 'odom' at time 1724823853.038 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1724823857.163267390] [rviz2]: Message Filter dropping message: frame 'odom' at time 1724823853.277 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1724823857.163353939] [rviz2]: Message Filter dropping message: frame 'odom' at time 1724823850.405 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1724823857.484356087] [rviz2]: Message Filter dropping message: frame 'odom' at time 1724823853.743 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1724823857.771204074] [rviz2]: Message Filter dropping message: frame 'odom' at time 1724823854.030 for reason 'discarding message because the queue is full'

```

[image]

RustyCPR, 2024-09-13:
Hi @taara67 
Can you please open your own ticket and please share the ros2 launch turtlebot4_navigation localization.launch.py full terminal output? 
I'm closing this ticket because user @Rahul007RR has not replied in some time. 
Thank you,
Rusty

closed 2024-09-13 by RustyCPR