turtlebot/turtlebot4 #428: Turtlebot Charging Issues

dartik24, 2024-06-19:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Ros2 Humble

### Type of issue

Battery

### Expected behaviour

Robot charges over 2 hours to full battery which is then displayed. While charging the dock light glows green showing connectivity.

### Actual behaviour

When put on the dock the dock light flashes green for about 5 seconds then turns off. The robot batter never displays above 0 however at times the robot does seem to hold some charge.

### Error messages

```bash
N/A
```

### To Reproduce

Place robot on the dock, wait.

### Other notes

The system has been reimaged that did not fix the issues.

smatarCPR, 2024-06-26:
Hello @dartik24 ,
Thank you for reaching out on the Turtlebot4 Github. I am very sorry to hear that you are encountering issue with your Turtlebot.

Would it be possible to try and reset the battery? To do so: remove the 5 screws at the bottom of the TB4 and remove the battery for 30 minutes to reset it, then reinstall and try charging - see images.
[image]

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif

dartik24, 2024-06-26:
Understood, we will attempt this fix and let you know if there are further issues.

tharukakk1, 2024-07-15:
I am also having this same issue with both of my Turtlebots. I tried this suggestion of removing the battery for 30 mins and replacing it. Both have been charging on their docks for 2 hours now after that and still show 0% on their onboard screens. One of them is showing ~1/3 charge on the light ring now (it was showing 0 charge on the light ring at the beginning). The other bot has a dim solid white light. @dartik24 did you manage to fix your issue, and if so, how did you?

dartik24, 2024-07-16:
@tharukakk1 For one of the turtle bots the issue has gone away after restarting it a few time after the battery trick. For the other the issue persists and I have been relying on the power button or the battery topic to tell the battery.

dgel27, 2024-07-17:
Hi, **don't leave robot uncharged for a long time**. If voltage in cells inside battery pack fall less then 3v, the battery controller of the battery pack stop charging the cells. In this case, the battery will newer charged. Only way to fix it - disasaembly pack,  charge each cell with external charger and assembly it back. It not simply task, and need experiences and tools for this.

tharukakk1, 2024-07-23:
Sorry for the late reply. I appreciate the comments from both of you. Seems like I'm in the same situation as @dartik24 now. The create3 webserver shows proper charging and connection, but somehow that doesn't get transmitted to the onboard screen. One of my robots started working very briefly yesterday, showing 100% on the screen. When it started working, all of the topics also showed up in $ros2 topic list, but they went away again at the same time the screen stopped working again. It stopped working when I rebooted the robot. Rebooting again multiple times didn't fix the problem. This is a completely different problem all of its own so I'm going to start a new issue for it.

tharukakk1, 2024-09-18:
Hi @smatarCPR, my turtlebots are both working great now. Reinstalling the Raspberry Pi image as described in [Basic Setup](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html), then charging it for a day did the trick. As the owner of two Turtlebots, removing the battery from the other robot was generally helpful when troubleshooting the first robot, and vice versa.

smatarCPR, 2024-10-30:
Hello all,
I will go ahead and close this issue. Should you encounter any follow up issues, do not hesitate to reach out by simply replying to this thread or opening up a new issue.
Best Regards,
Saif

closed 2024-10-30 by smatarCPR