turtlebot/turtlebot4 #430: Turtlebot 4 Mapping error - URGENT

dipeshmenon, 2024-06-21:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

I believe latest version

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I have 2 turtlebot 4 standards with me. The first was able to map after the tutorial from the docs, the second one doesnt map

### Actual behaviour

I ran the ros2 launch turtlebot4_navigation slam.launch.py 
and then rviz, ros2 launch turtlebot4_viz view_robot.launch.py

[image]
[image]
[image]
[image]

### To Reproduce

1. Run the slam launch file
2. Run the rviz launch file

### Other notes

itS VERY URGENT, any help is apprecaited

smatarCPR, 2024-06-26:
Hello @dipeshmenon ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are encountering issues with your Turtlebot 4.

With regards to this matter I have the following recommendations/requests:
- The dpkg command you are attempting to run should be: `dpkg -s ros-humble-turtlebot4-navigation`
- Are you able to view the second Turtlebot 4 model in RVIZ without running SLAM?
- Are you able to view and echo topics from the second Turtlebot 4 on your remote computer?
- Are the two Turtlebots operating on the same ROS_DOMAIN_ID?
- Have you provided unique namespaces for your ROS environment for the two Turtlebots?

Looking forward to hearing from you soon.
Best Regards,
Saif

dipeshmenon, 2024-06-27:
Hi Saif, thank you for replying, I fixed this issue somehow but I do have an issue with the navigation part. It shows client server didn't respond or sometimes it just does abort bringup. It is rare that the navigation works. Any idea on how to fix this?

Ragnar7982, 2024-07-01:
Hello check ur rpi time is correct or no

smatarCPR, 2024-07-15:
Hello @dipeshmenon ,
Apologies for the late reply on this. A couple questions regarding your current setup:
-  Are you using the same remote computer or different computers between the two turtlebots?
- Double check that both Turtlebots are running the most up to date/same version of Turtlebot Navigation
- Are the configurations between the two turtlebots the same (i.e. diagnostics enabled/disabled)
- If all else fails, please provide the journalctl logs `journalctl -b > turtlebot4_log.txt`
Looking forward to hearing from you soon.
Best Regards,
Saif

RustyCPR, 2024-09-09:
Closing due to no activity

closed 2024-09-09 by RustyCPR